wolfSSL/wolfssl #5908: compiling for SHAKE128 fails

devendranaga, 2022-12-18:
### Version

91f8b5e58

### Description

Hi,

I am trying to use latest wolfssl (91f8b5e58)  for SHAKE128 to compile for my application.

I have enabled SHAKE support from the `configure` with `--enable-shake128` and it shows up in the final options.

```
* SHAKE128:                   yes
```

Performed `make clean`, `make` and `sudo make install` to install the latest library.

I define `WOLFSSL_SHAKE128` right above the `sha3.h` header definition in my code and use Shake128 API.

```c
#define WOLFSSL_SHA3
#define WOLFSSL_SHAK128
#include <wolfssl/wolfcrypt/sha3.h>
```

But fails compiling with the following error.

Any hits to what i am doing wrong?

```bash
n file included from /usr/local/include/wolfssl/wolfcrypt/types.h:34,
                 from /usr/local/include/wolfssl/wolfcrypt/sha.h:30,
                 from /home/devnaga/work/auto_os/auto_crypto/src/wolfssl/wolfssl_hash_function.cc:7:
/usr/local/include/wolfssl/wolfcrypt/settings.h:2506:14: warning: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Wcpp]
 2506 |             #warning "For timing resistance / side-channel attack prevention consider using harden options"
      |              ^~~~~~~
In file included from /home/devnaga/work/auto_os/auto_crypto/src/wolfssl/wolfssl_hash_function.cc:23:
/usr/local/include/wolfssl/wolfcrypt/sha3.h:65:32: error: ‘WC_HASH_TYPE_SHAKE128’ was not declared in this scope; did you mean ‘WC_HASH_TYPE_SHA512’?
   65 |         WC_SHAKE128          = WC_HASH_TYPE_SHAKE128,
      |                                ^~~~~~~~~~~~~~~~~~~~~
      |                                WC_HASH_TYPE_SHA512
/home/devnaga/work/auto_os/auto_crypto/src/wolfssl/wolfssl_hash_function.cc: In member function ‘virtual int auto_os::crypto::wolfssl_hash_function::hash(uint8_t*, size_t, uint8_t*, size_t&)’:
/home/devnaga/work/auto_os/auto_crypto/src/wolfssl/wolfssl_hash_function.cc:278:36: error: too few arguments to function ‘int wc_Shake128_Final(wc_Shake*, byte*, word32)’
  278 |             ret = wc_Shake128_Final(&shake, hash);
      |                   ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/usr/local/include/wolfssl/wolfcrypt/sha3.h:170:17: note: declared here
  170 | WOLFSSL_API int wc_Shake128_Final(wc_Shake* shake, byte* hash, word32 hashLen);
      |                 ^~~~~~~~~~~~~~~~~
make[2]: *** [auto_crypto/CMakeFiles/auto_crypto.dir/build.make:384: auto_crypto/CMakeFiles/auto_crypto.dir/src/wolfssl/wolfssl_hash_function.cc.o] Error 1
```

dgarske, 2022-12-18:
Hi @devendranaga ,

You must include wolfssl/options.h which contains the build options generated by ./configure. Your application must have the same build macros defined to enable features and ensure our structures match. 

See FAQ item 1 here:

https://www.wolfssl.com/docs/frequently-asked-questions-faq/#How_do_I_manage_the_build_configuration_for_wolfSSL?

Thanks,
David Garske, wolfSSL

devendranaga, 2022-12-19:
Hello @dgarske ,

The below change worked. Thanks for the FAQ, i have been using the header and macros wrong all the time :-).

```c
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>
```

Thanks very much!

closed 2022-12-19 by devendranaga