wolfSSL/wolfssl #5935: Add context arg to mbedtls_ssl_set_export_keys_cb()

kevinpt, 2022-12-27:
### Version

5.5.4

### Description

mbedtls_ssl_set_export_keys_cb() doesn't take a caller specified context argument. This significantly limits what you can do in the callback function without resorting to global state. It doesn't appear that struct WOLFSSL and struct WOLFSSL_CTX have a generic void* akin to the OpenSSL app data so that isn't an option either.

It would be nice to have a ctx arg on mbedtls_ssl_set_export_keys_cb() similar to many of the other callback registration functions.

embhorn, 2022-12-27:
Hello @kevinpt 

Thanks for contacting wolfSSL Support. We do not maintain the `mbedtls_ssl_set_export_keys_cb` API as part of the wolfSSL library. 

Are you using a compatibility layer to access wolfSSL? Could you please share a link in GutHub to the code you are referring to?

Thanks,
@embhorn - wolfSSL Support

kevinpt, 2023-01-04:
Sorry. I had a brainfart. I meant the API provided by wolfSSL_CTX_set_keylog_callback(). I lacks a way to provide any user context to the callback.

embhorn, 2023-01-04:
Hi @kevinpt 

That makes sense. I can open a feature request if you'd like. Could you tell us a bit about your project using the key log callback? Feel free to email support@wolfssl.com if you'd prefer to use a more private format.

Thanks,
@embhorn - wolfSSL Support

embhorn, 2025-12-10:
Closing this older feature request. There is a user context pointer in the wolfSSL_CTX structure