wolfSSL/wolfssl #5961: Failure in few test cases while executing make test command

sandilya761, 2023-01-07:
### Version

wolfssl-5.5.1

### Description

Some of the test cases are failing while executing **make test** command. Following image contains the failed test cases:

[image]

[image]

Following are my environment details:
**Wolfssl version:** 5.5.1
**make version:** GNU Make 4.3
**Kernel version:** 5.7.0-kali1-amd64.

ejohnstown, 2023-01-09:
Can you tell me how you are configuring wolfSSL? I tried:

    ./configure --enable-all && make check

and didn't have any failures. I'm testing this in a VM of the Kali Linux 2022.4 image.

kareem-wolfssl, 2023-01-09:
Hi @sandilya761,

Please retry this with our latest release, 5.5.4.  As John stated please share your configure line if you continue to have issues.

sandilya761, 2023-01-11:
I tried the following command with **5.5.4 version**:
`./configure --enable-all && make check`

It is getting stuck after scripts/ocsp.test as shown in the image below:

[image]

kareem-wolfssl, 2023-01-11:
The next test should be our unit test, please try running it standalone with `./scripts/unit.test` and let me know if it still freezes.  If it does, please attach the full output here.
Note that this test can take a long time.

sandilya761, 2023-01-12:
`./scripts/unit.test` command is working fine.

[image]

kareem-wolfssl, 2023-01-16:
Hi @sandilya761,

Running `make test` is equivalent to `make check`.
Please try rebuilding with `--disable-jobserver`, with this enabled the tests run in a somewhat random order, disabling it will cause the tests to run in a consistent order.

Thanks,
Kareem

sandilya761, 2023-01-17:
It's working fine. 

Thank You!!

closed 2023-01-17 by sandilya761