wolfSSL/wolfssl #5965: Support for hardware crypto acceleration using ATMEL ATECC608

AScholan, 2023-01-09:
I am attempting to use the ATMEL ATECC608 TnG device for hardware acceleration of WolfSSL on an embedded device without an in-built crypto-engine.

I have found some support for private keys and certificates using this device which is supported through the PK callbacks. However, it looks to me that although there may have been an intent to support the hardware acceleration in this device (perhaps through the crypto callback mechanism in WolfCrypt) if appears that this was never implemented. For example the macro `WOLFSSL_ATECC_SHA256` mentioned in the `README.md` in the `src/port/atmel` folder does not appear anywhere in the code.

My questions are:
1. Do you have an intention to include this support in the future?
2. If not, what steps would you suggest to me if I were to implement this through the crypto callback mechanism?

Thank you,
Andrew Scholan

dgarske, 2023-01-10:
Hi @AScholan ,

Thanks for your interest and excellent question. When the ATECC support was added back with the original ATECC508 we only supported the TLS layer PK callbacks. Later we added the crypto callbacks, which are very handy. We did implement crypto callbacks for the STSAFE here (https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/port/st/stsafe.c#L359). You could use this as a template. We do not have an immediate plan to add ATECC crypto callback support, but it would be very nice to have. You are welcome to submit a PR with your code contribution.

It looks like the ATECC SHA256 support is not in upstream master, but was added for one of the demos using the ATECC here: https://github.com/wolfSSL/microchip-atecc-demos/blob/48f4dc4c6f548ce1c7d9f9b9974bc1bbb1d6c2d1/wolfssl_pic32mz_curiosity/wolfssl_server/firmware/src/third_party/wolfssl/wolfssl/wolfcrypt/src/sha256.c

Can you tell us more about your project or use-case with the ATECC?

Thanks,
David Garske, wolfSSL

AScholan, 2023-01-17:
Hi @dgarske,

Thanks for the reply. Based on what you provided I was able to hook the SHA256 methods through the Crypto Callback
system but actually the performance of using the ATECC608B across the I2C bus was significantly worse (~ x10 slower)
than using the software crypto on the PIC32MZ@200MHz.

The reason for doing this is that the PIC32MZ we are using has a variant with a hardware crypto accelerator but because of
supply chain issues it's not possible to readily source these at the moment so we wanted to explore using the ATECC608
that we already have on-board but only using for credential management (certificates and private keys).
 
The product is an IoT bridge for industrial sensors to a bespoke cloud infrastructure, not using MQTT.

Some observations regarding the CryptoCB interface would be:

1. It's really not clear what you should use as a device ID; in the end it looks fairly arbitrary to me.
2. Hooking the callback function is not sufficient, and it's not clear what other functions you should also hook. 
   I discovered that to hook the SHA256 functionality it was necessary to also hook the following methods:
   -  `int wc_InitSha256_ex(wc_Sha256* sha, void* heap, int devId)`
   -  `void wc_Sha256Free(wc_Sha256* sha)` 
   -  `int wc_Sha256Copy(wc_Sha256* src, wc_Sha256* dst)`
   I did this by declaring the WolfCrypt methods in `sha256.c` as weakly linked using the `__WEAK` macro decorator rather
   than add another set of `#ifdef` .. `#endif` sections to the wolf code.
3. The life-cycle for the wc_Sha256 objects is not well defined and in multiple instances the `wc_Sha256Free` method is never
   invoked on an object that has been initialised with `wc_InitSha256_ex`. This means that you have to manage all allocations
   of dynamic objects through the callback function - in the case of SHA256 creating it when the first 'update' is invoked and
   freeing it either on error or at the end of 'finalise'.

I don't actually think that, as it stands at the moment, that the CryptoCB functionality is enough to fully implement a "plug-in"
hardware accelerator, I think you would need to register a structure of function calls rather than a single callback function (to handle the init, free, copy) or provide a hinting system to the callback method so that it knows whether to init, free, copy, or action. It would have been less code to simply hook the SHA256 update and finalise methods with weak linkage, as with the others.

Anyway, I've attached my solution as it may be helpful to others seeing this issue: [wolf_atecc_glue.zip](https://github.com/wolfSSL/wolfssl/files/10434630/wolf_atecc_glue.zip)

In regard to the original enquiry, I going to close this issue (Not Planned) as the performance of using the ATECC608 over the I2C bus at 400kHz is really not adequate enough to provide a useful solution.

Thank you once again for your help.
Andrew.

closed 2023-01-17 by AScholan

dgarske, 2023-01-31:
Hi @AScholan ,

To close on this topic... you would use the `wc_Sha256` -> `devCtx` member to provide the callback context you need for this. https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/sha256.h#L224

You can find an example in wolfTPM here: https://github.com/wolfSSL/wolfTPM/blob/master/src/tpm2_wrap.c#L5926

Thanks,
David Garske, wolfSSL