wolfSSL/wolfssl #5966: [Bug]: --enable-kapi incompatable with --enable-opensslextra

duplexsystem, 2023-01-11:
### Contact Details

duplexsys@protonmail.com

### Version

Latest from Github

### Description

the options  ` --enable-kcapi` and ` --enable-opensslextra` are incompatible with each other and result in a build failure 

configure command: `./configure --enable-opensslextra --enable-kcapi`

Error when running `make`

Errors with both clang and gcc

### Reproduction steps

1. `./configure --enable-opensslextra --enable-kcapi`
2. `make`

### Relevant log output

```shell
make -j9  all-recursive
make[1]: Entering directory '/home/duplexsystem/wolfssl'
make[2]: Entering directory '/home/duplexsystem/wolfssl'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-kdf.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/port/kcapi/src_libwolfssl_la-kcapi_hash.lo
  CC       wolfcrypt/src/port/kcapi/src_libwolfssl_la-kcapi_aes.lo
  CC       wolfcrypt/src/port/kcapi/src_libwolfssl_la-kcapi_hmac.lo
  CC       wolfcrypt/src/port/kcapi/src_libwolfssl_la-kcapi_ecc.lo
  CC       wolfcrypt/src/port/kcapi/src_libwolfssl_la-kcapi_rsa.lo
  CC       wolfcrypt/src/port/kcapi/src_libwolfssl_la-kcapi_dh.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sp_int.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
  CC       wolfcrypt/src/src_libwolfssl_la-error.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo
wolfcrypt/src/sha512.c:1225:11: error: call to undeclared function 'Transform_Sha512'; ISO C99 and later do not support implicit function declarations [-Werror,-Wimplicit-function-declaration]
    ret = Transform_Sha512(sha);
          ^
1 error generated.
make[2]: *** [Makefile:5870: wolfcrypt/src/src_libwolfssl_la-sha512.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/duplexsystem/wolfssl'
make[1]: *** [Makefile:7239: all-recursive] Error 1
make[1]: Leaving directory '/home/duplexsystem/wolfssl'
make: *** [Makefile:4274: all] Error 2
```

dgarske, 2023-01-13:
Hi @duplexsystem ,

I can reproduce and will put up a fix shortly. Seems to be some recent code added for compatibility layer that is not setup correctly for the KCAPI use-case.

Can you tell us more about your project? Is this hobby, academic, commercial, etc..?

Thanks,
David Garske, wolfSSL

dgarske, 2023-01-13:
Hi @duplexsystem ,

The fixes have been pushed to PR https://github.com/wolfSSL/wolfssl/pull/5977

Let me know if you notice anything else. We designed the KCAPI later to be used with FIPS, but it should work in non-FIPS mode also.

Thanks,
David Garske, wolfSSL

closed 2023-01-16 by SparkiDev

dgarske, 2023-01-17:
Hi Zoe,

Thank you for the details on your project. You might find our Open Source Project (OSP) repo interesting. https://github.com/wolfSSL/osp

We are happy to support your efforts.

Thanks,
David Garske, wolfSSL

duplexsystem, 2023-01-17:
Thank you for pointing me to that repo! It does seem to be very useful for my project. And thank you for your support!

PR #5977 (2023-01-16): Fixes for building KCAPI with opensslextra enabled
# Description

Fixes for building KCAPI with opensslextra enabled. The raw hash transform and ECC Shamir math options are not available with KCAPI.

Cleanup of the compatibility layer hash functions for whitespace, indentation and consistency.

Fixes #5966

# Testing

```
./configure --enable-opensslextra --enable-kcapi && make
```

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation