wolfSSL/wolfssl #5968: How to get the gpg publickey to verify the souce code ?

LubinLew, 2023-01-12:
### Version

5.5.4

### Description

```bash
$ gpg --verify wolfssl-5.5.4-stable.tar.gz.asc wolfssl-5.5.4-stable.tar.gz

gpg: Can't check signature: No public key
```

JacobBarthelmeh, 2023-01-12:
Try with the secure (@) wolfssl key up on opengpg.org.

https://keys.openpgp.org/search?q=A2A48E7BCB96C5BECB987314EBC80E415CA29677

LubinLew, 2023-01-13:
@JacobBarthelmeh 

It works, Thank you very much.

```bash
$ wget https://keys.openpgp.org/vks/v1/by-fingerprint/A2A48E7BCB96C5BECB987314EBC80E415CA29677 -O wolfssl.asc
$ gpg --show-keys wolfssl.asc|grep '^ '|tr -d ' '|awk '{print $0":6:"}' > /tmp/ownertrust.txt
$ gpg --import-ownertrust < /tmp/ownertrust.txt
$ gpg --import wolfssl.asc
$ gpg --verify wolfssl-5.5.4-stable.tar.gz.asc wolfssl-5.5.4-stable.tar.gz
gpg: Signature made Wed Dec 21 18:06:08 2022 UTC
gpg:                using RSA key A2A48E7BCB96C5BECB987314EBC80E415CA29677
gpg: Good signature from "wolfSSL <secure@wolfssl.com>" [ultimate]
gpg:                 aka "Jacob Barthelmeh (wolfSSL) <secure@wolfssl.com>" [ultimate]

closed 2023-01-13 by LubinLew