wolfSSL/wolfssl #5971: [Bug]: Recent PR breaks Clang build with SP_WORD_SIZE=8

guidovranken, 2023-01-13:
### Version

Git repository checkout

### Description

https://github.com/wolfSSL/wolfssl/pull/5969

This PR introduced the line

```c
d >>= 8;
```

I had checked that PR for correctness in the main configuration, but I hadn't tested all configurations.
It turns out it can cause compilation failures.

### Reproduction steps

```sh
export CC=clang
export CFLAGS="-DSP_WORD_SIZE=8"
./configure --enable-sp-math-all
make
```

### Relevant log output

```shell
wolfcrypt/src/sp_int.c:16885:23: error: shift count >= width of type [-Werror,-Wshift-count-overflow]
                    d >>= 8;
                      ^   ~
1 error generated.
```

SparkiDev, 2023-01-13:
Hi @guidovranken

I have a fix for this: #5972

Thanks,
Sean :-)

guidovranken, 2023-01-14:
Confirmed fixed.

closed 2023-01-14 by guidovranken