wolfSSL/wolfssl #5981: wc_SignatureVerifyHash isn't working with this hashed message + key + sig, what am I missing?

petertorelli, 2023-01-17:
### Version

5.5.4

### Description

I'm was using wc_SignatureVerify, and it worked fine, so I switched to wc_SignatureVerifyHash, but now it is failing. I've included debug info further down.

Here is the code call, the Rng and Key are already properly loaded (I know this because the wc_SignatureVerify() is working):

    ret = wc_SignatureVerifyHash(WC_HASH_TYPE_SHA256,
                                 WC_SIGNATURE_TYPE_RSA_W_ENC,
                                 p_hash,
                                 hashlen,
                                 p_sig,
                                 siglen,
                                 ctx->pubkey,
                                 sizeof(RsaKey));

p_hash = 

b703b9c084bee8a278a956534844d7fa4010a5868c0c82a4506bf81c36218de1

p_pub (ASN.1) = 

30820122300d06092a864886f70d01010105000382010f003082010a0282010100d084161e7f069fefee250a9e9b8079cd8b747175e5576ab483b37e01bf1de263ecf3962881ddaba4feb8614465838eb21947dc6e4dc996121d13d789ec671a8b032b9da94cec6917f37056fabe22143950af93d226f478222f22abdf6e437cfed45a7936b1fee4cf6006b3282f21afbbeb95a7cd87a89d68358ad17f13de4c292aa2342714cdb42e81eaeb32812167211bd2029c214c9be230039cf8eb997f403674d742715a3fd3a29a0621bb5e7dd90c88ec808fe45e310410940f96c8c772f4c6cd988f301f792d9b43af8cd655b7d89ad3a46c70d8bcc66360f4beb4735693e9742fd5363691701cd38dc5202051abbb46f4c53a59775568836cdb4a21470203010001

p_pub parsed from ASN.1 above = 

D084161E7F069FEFEE250A9E9B8079CD8B747175E5576AB483B37E01BF1DE263ECF3962881DDABA4FEB8614465838EB21947DC6E4DC996121D13D789EC671A8B032B9DA94CEC6917F37056FABE22143950AF93D226F478222F22ABDF6E437CFED45A7936B1FEE4CF6006B3282F21AFBBEB95A7CD87A89D68358AD17F13DE4C292AA2342714CDB42E81EAEB32812167211BD2029C214C9BE230039CF8EB997F403674D742715A3FD3A29A0621BB5E7DD90C88EC808FE45E310410940F96C8C772F4C6CD988F301F792D9B43AF8CD655B7D89AD3A46C70D8BCC66360F4BEB4735693E9742FD5363691701CD38DC5202051ABBB46F4C53A59775568836CDB4A2147

p_sig = 

07078211c72dc4841d32c57883dc500297f758a5de02035fb8ba354da1b5cea38215b22cd612324327ec86257cc30d048cb20c13645d1164366a620a88e79f983a563d79ba4a69e035f4c901ffecd35b8e830d7b139e56d4bc0cee559442aedaf64ad4a60cd20b0c69b97a2a7d0286773ba44d2b745ab8447d206b0756547313d8a1a69a5f1d819f69e0491b034d7231ab045b70b5df3e80465639f1d344caaea53d10717f58b7aa583cc1430aa628c118ee87dfa942a11b60a645234a47c6df4a487fce99b7f7446429c7a7e04d3646a862e192ee68565af0b3d047101b327c4f02279c4a8b8682b8884ac5edff3d4b40dae0e4d3f8f5fe05fed17f7deeeaef

If I decrypt the SIG I get this:

0x1ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff003031300d060960864801650304020105000420b703b9c084bee8a278a956534844d7fa4010a5868c0c82a4506bf81c36218de1

We can see the hash in ASN.1 of OID|hash. Why is VerifyHash failing with SIG_VERIFY_E? I'm using WC_SIGNATURE_TYPE_RSA_W_ENC which should decode ASN.1, correct?

Thanks,
Peter

embhorn, 2023-01-17:
Hi @petertorelli 

We have a signature verify example that is very similar to what you are testing:
https://github.com/wolfSSL/wolfssl-examples/tree/master/signature/sigtest

You should uncomment this line in `wolfsigtest.c`
`#define DEMO_RSA_VERIFY_ONLY`
...to see `wc_SignatureVerifyHash` being used.
https://github.com/wolfSSL/wolfssl-examples/blob/a363898fbc2f6a16240b81da39b00e6b5064101c/signature/sigtest/wolfsigtest.c#L13-L14

Could you construct a modified version of the example that reproduces the issue you are seeing?

petertorelli, 2023-01-18:
Hello @embhorn,

It appears that SignatureGenerateHash, even with WC_SIGNATURE_TYPE_RSA_W_ENC only pads, but does not perform ASN.1(oid|hash) encoding. When I decrypt the signature with the public key (python `hex(pow(s, e, m)`) I get the padded raw hash in the signature and not the encoded hash which is what I thought I was supposed to get.

Is there a way to enable PKCS1v15 for RSA sign/verify and avoid having to manually do the encoding? I was using a different library that automatically unwrapped the signature during the comparison.

Here is my code that, note that the expected signature is what I included in the original issue, which is encoded as well as padded.

```c
#include <stdio.h>
#include <stdlib.h>

#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/signature.h>
#include <wolfssl/wolfcrypt/rsa.h>

const byte private_key_der[] = {
0x30,0x82,0x04,0xbc,0x02,0x01,0x00,0x30,0x0d,0x06,0x09,0x2a,0x86,0x48,0x86,0xf7,
0x0d,0x01,0x01,0x01,0x05,0x00,0x04,0x82,0x04,0xa6,0x30,0x82,0x04,0xa2,0x02,0x01,
0x00,0x02,0x82,0x01,0x01,0x00,0xd0,0x84,0x16,0x1e,0x7f,0x06,0x9f,0xef,0xee,0x25,
0x0a,0x9e,0x9b,0x80,0x79,0xcd,0x8b,0x74,0x71,0x75,0xe5,0x57,0x6a,0xb4,0x83,0xb3,
0x7e,0x01,0xbf,0x1d,0xe2,0x63,0xec,0xf3,0x96,0x28,0x81,0xdd,0xab,0xa4,0xfe,0xb8,
0x61,0x44,0x65,0x83,0x8e,0xb2,0x19,0x47,0xdc,0x6e,0x4d,0xc9,0x96,0x12,0x1d,0x13,
0xd7,0x89,0xec,0x67,0x1a,0x8b,0x03,0x2b,0x9d,0xa9,0x4c,0xec,0x69,0x17,0xf3,0x70,
0x56,0xfa,0xbe,0x22,0x14,0x39,0x50,0xaf,0x93,0xd2,0x26,0xf4,0x78,0x22,0x2f,0x22,
0xab,0xdf,0x6e,0x43,0x7c,0xfe,0xd4,0x5a,0x79,0x36,0xb1,0xfe,0xe4,0xcf,0x60,0x06,
0xb3,0x28,0x2f,0x21,0xaf,0xbb,0xeb,0x95,0xa7,0xcd,0x87,0xa8,0x9d,0x68,0x35,0x8a,
0xd1,0x7f,0x13,0xde,0x4c,0x29,0x2a,0xa2,0x34,0x27,0x14,0xcd,0xb4,0x2e,0x81,0xea,
0xeb,0x32,0x81,0x21,0x67,0x21,0x1b,0xd2,0x02,0x9c,0x21,0x4c,0x9b,0xe2,0x30,0x03,
0x9c,0xf8,0xeb,0x99,0x7f,0x40,0x36,0x74,0xd7,0x42,0x71,0x5a,0x3f,0xd3,0xa2,0x9a,
0x06,0x21,0xbb,0x5e,0x7d,0xd9,0x0c,0x88,0xec,0x80,0x8f,0xe4,0x5e,0x31,0x04,0x10,
0x94,0x0f,0x96,0xc8,0xc7,0x72,0xf4,0xc6,0xcd,0x98,0x8f,0x30,0x1f,0x79,0x2d,0x9b,
0x43,0xaf,0x8c,0xd6,0x55,0xb7,0xd8,0x9a,0xd3,0xa4,0x6c,0x70,0xd8,0xbc,0xc6,0x63,
0x60,0xf4,0xbe,0xb4,0x73,0x56,0x93,0xe9,0x74,0x2f,0xd5,0x36,0x36,0x91,0x70,0x1c,
0xd3,0x8d,0xc5,0x20,0x20,0x51,0xab,0xbb,0x46,0xf4,0xc5,0x3a,0x59,0x77,0x55,0x68,
0x83,0x6c,0xdb,0x4a,0x21,0x47,0x02,0x03,0x01,0x00,0x01,0x02,0x82,0x01,0x00,0x03,
0xe0,0x78,0xd2,0xcf,0x54,0xcd,0xb3,0x75,0xcf,0x8d,0x7f,0x7e,0x6f,0xa8,0x94,0x18,
0x44,0x2d,0x79,0xde,0x2e,0x3e,0x67,0x25,0xcd,0xb8,0x09,0x8b,0x46,0xa0,0x77,0x70,
0x27,0xe5,0x42,0x52,0x8b,0xcf,0x0d,0xad,0xc8,0xba,0xd2,0x3d,0x3a,0xf1,0x73,0xc1,
0x41,0x93,0x49,0x1f,0xbc,0x20,0x44,0xac,0xf2,0xb9,0xfe,0xaf,0xfd,0x08,0xc2,0xcd,
0xe7,0xc9,0x32,0x25,0x22,0x7e,0x1f,0xba,0x60,0x7b,0xd5,0x84,0xa4,0x78,0x90,0xca,
0x3d,0xc1,0x77,0x56,0x4f,0x14,0x4e,0xba,0x66,0xa0,0x20,0x72,0xd9,0xe1,0x60,0x91,
0x8d,0x2f,0x14,0xc3,0xc1,0x73,0xfa,0xa1,0x13,0xee,0xcf,0xcf,0xf4,0x85,0xe2,0x6e,
0x12,0x65,0xc0,0xce,0x05,0x66,0x0b,0xd2,0x77,0x06,0xc7,0x20,0x94,0xf1,0x00,0x88,
0x57,0xb7,0xa9,0xd4,0x1c,0x78,0x0b,0xb4,0xda,0x09,0x27,0xa2,0x77,0x4f,0x7e,0x69,
0xb1,0xd9,0x6c,0xd5,0x70,0x7e,0xd7,0x24,0x5f,0x4e,0x38,0x4f,0xf9,0xc0,0x87,0xc3,
0x31,0x8e,0x8d,0x99,0x2c,0x0c,0xd9,0x4b,0xa5,0x71,0xec,0x9e,0x4f,0x24,0x92,0xf5,
0x7d,0x30,0x27,0x33,0x17,0xb5,0xf1,0xf4,0xd5,0x38,0x3a,0xa0,0x75,0xf9,0x7a,0x9d,
0x0e,0x5c,0x81,0x4e,0x21,0x6d,0xcc,0x55,0xac,0x3d,0x78,0x18,0x6c,0x34,0x42,0x9d,
0x9b,0xd5,0x44,0x3f,0x95,0x5f,0xbd,0xef,0x18,0x0d,0xef,0xca,0xc3,0x9a,0x25,0x82,
0x39,0xc0,0x70,0x33,0x54,0xcd,0xe2,0x70,0x04,0xfc,0x8b,0x24,0x81,0x9c,0xf5,0x9f,
0xd1,0x71,0xe8,0xdc,0x17,0x0a,0x8d,0x2e,0xdf,0xee,0xf7,0x95,0x9e,0x4c,0x35,0x02,
0x81,0x81,0x00,0xdd,0xf4,0x28,0x51,0x05,0x12,0xed,0xa4,0xd6,0x16,0x73,0x3b,0xd9,
0x67,0x44,0x4d,0x26,0x97,0x76,0xb3,0xa7,0x6e,0x45,0x50,0xe0,0x8b,0xe7,0x71,0xe1,
0x89,0x04,0xa9,0xcf,0x6e,0x23,0x78,0x82,0xa6,0xed,0x45,0x9d,0xb0,0xd0,0x0a,0x01,
0xd8,0x15,0x0b,0xc7,0xb0,0x89,0xdc,0xd2,0xb1,0xd5,0x21,0x96,0x4f,0x51,0x66,0x0f,
0xde,0xa1,0x5c,0xd8,0x4e,0x9a,0xdf,0x85,0xb3,0x15,0x6e,0x2e,0xb4,0x4f,0xe1,0x5b,
0x86,0xad,0x79,0x63,0xbe,0x4c,0x80,0xf7,0xd9,0x5a,0x0d,0xdd,0xd2,0x4f,0x14,0xaa,
0x50,0x32,0xd4,0x95,0x9d,0x8b,0x03,0xea,0x57,0xc3,0x2b,0x3c,0x25,0x7f,0x60,0x87,
0x7d,0xfd,0x3f,0xad,0x14,0x06,0x20,0x94,0x07,0x9e,0x92,0xf7,0xf8,0x2b,0x0a,0x80,
0x60,0xed,0x95,0x02,0x81,0x81,0x00,0xf0,0x80,0x3e,0x90,0x76,0x67,0x3a,0xe4,0xa9,
0xe3,0x62,0x49,0xb8,0xdc,0x7e,0xb2,0x83,0x68,0x52,0x68,0xa9,0x7d,0xa2,0xcd,0x90,
0x15,0x27,0xcb,0x47,0x65,0xfc,0x6c,0x3b,0x42,0x6c,0x1d,0x30,0x02,0x35,0x62,0x52,
0x11,0xf4,0x6c,0xc3,0x8d,0xff,0x50,0x24,0xfc,0x2d,0x31,0xb6,0x12,0xb2,0xba,0x98,
0xe4,0x70,0x2e,0x0e,0x82,0x5f,0xf2,0xe9,0xbc,0xad,0x11,0x5a,0x6e,0x3e,0x17,0x4a,
0xd5,0xe7,0xa7,0x8e,0xd8,0x63,0xe5,0x67,0x6d,0x19,0x58,0xd9,0xd4,0xe1,0x0c,0x39,
0x8f,0x7c,0xd8,0x79,0x5d,0x3b,0x8c,0x1a,0xd0,0x8d,0x0e,0x21,0xe9,0xa5,0x3e,0xef,
0x7d,0xc6,0xa6,0xaa,0x32,0xcb,0x4e,0x56,0xab,0x17,0x96,0x1b,0xff,0xdb,0x83,0xbc,
0xb1,0x31,0x4d,0x89,0xae,0x84,0x6b,0x02,0x81,0x80,0x1a,0xde,0xba,0x1a,0xc4,0xe0,
0x7b,0xc0,0x9e,0xe2,0x6c,0xe4,0xeb,0xf6,0xe2,0x7f,0xec,0x92,0x1d,0x9e,0xb6,0x40,
[... 130 lines cut ...]
        sizeof(private_key),
        &rng);
    printf("wc_SignatureGenerateHash: %d\n", ret);
    print_buf("generated :", signature, signature_length);
    print_buf("expected  :", expected_signed_results, sizeof(expected_signed_results));

    ret = wc_SignatureVerifyHash(
        WC_HASH_TYPE_SHA256,
        WC_SIGNATURE_TYPE_RSA_W_ENC, 
        hashed_message,
        sizeof(hashed_message),
        signature,
        signature_length,
        &public_key,
        sizeof(public_key)
    );
    printf("wc_SignatureVerifyHash: %d\n", ret);

    wc_FreeRsaKey(&private_key);
    wc_FreeRsaKey(&public_key);
    wc_FreeRng(&rng);
    return 0;
}

```

dgarske, 2023-01-19:
Thanks @petertorelli , we can reproduce. @philljj is going to debug and fix.

philljj, 2023-01-26:
Hi @petertorelli,

I apologize for the slow reply.

The issue I think is this:  in the `wc_SignatureGenerate()` call chain (in `wolfcrypt/src/signature.c`), `wc_SignatureDerEncode()` is called to put the hash in ASN1/Der encoding before calling `wc_SignatureGenerateHash()`. The same is true with `wc_SignatureVerify()` and `wc_SignatureVerifyHash()`.

So if `wc_SignatureGenerateHash()` is called directly without first Der encoding the input hash, then the results will not be what you expect.

I modified your example by adding the call to `wc_SignatureDerEncode()` before `wc_SignatureGenerateHash()`, and after that generated and expected do match (note: `wc_SignatureDerEncode` is static and had to be locally implemented in the example).

```
    memcpy(hash_enc_data, hashed_message, hash_len);
    ret = wc_SignatureDerEncode(WC_HASH_TYPE_SHA256, hash_enc_data, hash_len,
        &hash_enc_len);

    printf("wc_SignatureDerEncode: %d\n", ret);

    ret = wc_SignatureGenerateHash(
        WC_HASH_TYPE_SHA256,
        WC_SIGNATURE_TYPE_RSA_W_ENC,
        hash_enc_data,
        hash_enc_len,
        signature,
        &signature_length,
        &private_key,
        sizeof(private_key),
        &rng);
    printf("wc_SignatureGenerateHash: %d\n", ret);
    print_buf("generated :", signature, signature_length);
    print_buf("expected  :", expected_signed_results, sizeof(expected_signed_results));
```

Output:
 
```
wc_RsaPrivateKeyDecode: 0
wc_RsaPublicKeyDecode: 0
wc_SignatureDerEncode: 0
wc_SignatureGenerateHash: 0
generated :
07 07 82 11 c7 2d c4 84 1d 32 c5 78 83 dc 50 02 
97 f7 58 a5 de 02 03 5f b8 ba 35 4d a1 b5 ce a3 
82 15 b2 2c d6 12 32 43 27 ec 86 25 7c c3 0d 04 
8c b2 0c 13 64 5d 11 64 36 6a 62 0a 88 e7 9f 98 
3a 56 3d 79 ba 4a 69 e0 35 f4 c9 01 ff ec d3 5b 
8e 83 0d 7b 13 9e 56 d4 bc 0c ee 55 94 42 ae da 
f6 4a d4 a6 0c d2 0b 0c 69 b9 7a 2a 7d 02 86 77 
3b a4 4d 2b 74 5a b8 44 7d 20 6b 07 56 54 73 13 
d8 a1 a6 9a 5f 1d 81 9f 69 e0 49 1b 03 4d 72 31 
ab 04 5b 70 b5 df 3e 80 46 56 39 f1 d3 44 ca ae 
a5 3d 10 71 7f 58 b7 aa 58 3c c1 43 0a a6 28 c1 
18 ee 87 df a9 42 a1 1b 60 a6 45 23 4a 47 c6 df 
4a 48 7f ce 99 b7 f7 44 64 29 c7 a7 e0 4d 36 46 
a8 62 e1 92 ee 68 56 5a f0 b3 d0 47 10 1b 32 7c 
4f 02 27 9c 4a 8b 86 82 b8 88 4a c5 ed ff 3d 4b 
40 da e0 e4 d3 f8 f5 fe 05 fe d1 7f 7d ee ea ef 

expected  :
07 07 82 11 c7 2d c4 84 1d 32 c5 78 83 dc 50 02 
97 f7 58 a5 de 02 03 5f b8 ba 35 4d a1 b5 ce a3 
82 15 b2 2c d6 12 32 43 27 ec 86 25 7c c3 0d 04 
8c b2 0c 13 64 5d 11 64 36 6a 62 0a 88 e7 9f 98 
3a 56 3d 79 ba 4a 69 e0 35 f4 c9 01 ff ec d3 5b 
8e 83 0d 7b 13 9e 56 d4 bc 0c ee 55 94 42 ae da 
f6 4a d4 a6 0c d2 0b 0c 69 b9 7a 2a 7d 02 86 77 
3b a4 4d 2b 74 5a b8 44 7d 20 6b 07 56 54 73 13 
d8 a1 a6 9a 5f 1d 81 9f 69 e0 49 1b 03 4d 72 31 
ab 04 5b 70 b5 df 3e 80 46 56 39 f1 d3 44 ca ae 
a5 3d 10 71 7f 58 b7 aa 58 3c c1 43 0a a6 28 c1 
18 ee 87 df a9 42 a1 1b 60 a6 45 23 4a 47 c6 df 
4a 48 7f ce 99 b7 f7 44 64 29 c7 a7 e0 4d 36 46 
a8 62 e1 92 ee 68 56 5a f0 b3 d0 47 10 1b 32 7c 
4f 02 27 9c 4a 8b 86 82 b8 88 4a c5 ed ff 3d 4b 
40 da e0 e4 d3 f8 f5 fe 05 fe d1 7f 7d ee ea ef 

wc_SignatureVerifyHash: 0
```

I agree the behavior of `wc_SignatureGenerateHash()` with input parm `WC_SIGNATURE_TYPE_RSA_W_ENC` is confusing here, and either should be documented better or modified.

Thanks,
Jordan

eembc, 2023-01-26:
Thanks @philljj,

I had actually implemented that in the referring issue, except I used `wc_EncodeSignature` rather than `wc_SignatureDerEconde`... same effect. But you're right, the `WC_SIGNATURE_TYPE_RSA_W_ENC` needs clarificaiton.

I'll close it.

What I implemented as a workaround, but apparently this isn't a workaround but the actual mode of operation:

```c
ee_status_t
th_rsa_verify(void    *p_context,
              uint8_t *p_hash,
              uint32_t hashlen,
              uint8_t *p_sig,
              uint32_t siglen,
              bool    *p_pass)
{
    rsa_context_t *ctx = (rsa_context_t *)p_context;
    int            ret = 0;
    uint8_t        encoded_hash[100]; /* should be around 50 bytes */
    uint32_t       encoded_size;

    *p_pass = false;

    /* Add ASN digest info header */
    ret = wc_EncodeSignature(
        encoded_hash,
        p_hash,
        hashlen,
        SHA256h
    );

    if (ret < 0)
    {
        th_printf("e-[wc_EncodeSignature: %d]\r\n", ret);
        return EE_STATUS_ERROR;
    }

    encoded_size = (uint32_t)ret;

    ret = wc_SignatureVerifyHash(WC_HASH_TYPE_SHA256,
                                 WC_SIGNATURE_TYPE_RSA_W_ENC,
                                 encoded_hash,
                                 encoded_size,
                                 p_sig,
                                 siglen,
                                 ctx->pubkey,
                                 sizeof(RsaKey));

    if (ret != 0 && ret != SIG_VERIFY_E)
    {
        th_printf("e-[wc_SignatureVerifyHash: %d]\r\n", ret);
        return EE_STATUS_ERROR;
    }

    if (ret == 0)
    {
        *p_pass = true;
    }

    return EE_STATUS_OK;
}
```