wolfSSL/wolfssl #5982: [Bug]: Build fails with --enable-aesgcm-stream --enable-kcapi

guidovranken, 2023-01-17:
### Version

Git repository checkout

### Description

When compiling with streaming AES GCM (--enable-aesgcm-stream) and libkcapi support (--enable-kcapi), the build fails because although the streaming AES GCM functions `wc_AesGcm...` are declared in the headers, they are not compiled and linked into the library. This makes it unable to build the test suite, and impossible for programs to use streaming AES GCM functions.

### Reproduction steps

./configure --enable-aesgcm-stream --enable-kcapi && make

### Relevant log output

```shell
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0xe5f): undefined reference to `wc_AesGcmEncryptUpdate'
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0xe97): undefined reference to `wc_AesGcmEncryptFinal'
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0xf49): undefined reference to `wc_AesGcmDecryptInit'
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0xf92): undefined reference to `wc_AesGcmDecryptUpdate'
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0xfca): undefined reference to `wc_AesGcmDecryptFinal'
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0x1055): undefined reference to `wc_AesGcmEncryptInit'
/usr/bin/ld: test.c:(.text.aesgcm_test[aesgcm_test]+0x10c2): undefined reference to `wc_AesGcmEncryptUpdate'

and so forth
```

guidovranken, 2023-01-17:
That fix does not resolve my issue. It is present in the current latest repository checkout.

philljj, 2023-01-17:
Thank you Guido, I can reproduce. I'll open a PR for this.

philljj, 2023-01-18:
To give a bit more info: When building with `--enable-kcapi`, the Streaming AES-GCM functions are not intended to be available.

So this build combination is not intended to work. We should either add a helpful error message to configure, or let `--enable-kcapi` override and disable ` --enable-aesgcm-stream`.

guidovranken, 2023-01-19:
> We should either add a helpful error message to configure, or let `--enable-kcapi` override and disable ` --enable-aesgcm-stream`.

I agree.

PR #5990 (2023-01-19): Don't allow aesgcm-stream option with kcapi
# Description

The build options `--enable-aesgcm-stream` and `--enable-kcapi` were being allowed in `configure.ac`, but they are not compatible and will not build successfully.

This commit adds a check to `configure.ac` that will error out with a helpful error message:

```
configure: error: --enable-aesgcm-stream is incompatible with --enable-kcapi.
```

Fixes #5982.

# Testing

```
./configure --enable-aesgcm-stream --enable-kcapi && make
```