wolfSSL/wolfssl #5999: [Bug]: DTLS using IPv6 fails with error -2

Flole998, 2023-01-22:
### Version

Git master

### Description

I tried running the examples to see if they reproduce a certain issue. And indeed they do not work properly:

```
-> ./server-dtls-ipv6
Socket allocated
Awaiting client connection on port 11111
Connected!
```

```
-> ./client-dtls-ipv6 ::1
err = 2, non-blocking socket wants data to be read
SSL_connect failed
```
So somehow the client fails with a "EWOULDBLOCK" and interprets that as an error. Oddly enough this only seems to be the case on IPv6, when using IPv4 it does work properly. As I am seeing this when trying to use the library "normally" I believe it is not an issue in the examples but in the library itself.

Information about my system:
```
-> uname -a
Linux x01 5.15.0-58-generic #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
-> cat /etc/issue
Ubuntu 22.04.1 LTS \n \l
-> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/12/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 12.1.0-2ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-12/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-12 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-12-sZcx2y/gcc-12-12.1.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-12-sZcx2y/gcc-12-12.1.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.1.0 (Ubuntu 12.1.0-2ubuntu1~22.04)
```

embhorn, 2023-01-23:
Hello @Flole998 

Thanks for sharing this report. I was able to reproduce with the examples. I have requested a review by our engineers.

Kind regards,
@embhorn - wolfSSL Support

rizlik, 2023-01-23:
Hi @Flole998 ,

Can you double-check that wolfSSL is built with IPv6 support? 
If you are using configure you need to use the configure option `--enable-ipv6` .

embhorn, 2023-01-23:
> Hi @Flole998 ,
> 
> Can you double-check that wolfSSL is built with IPv6 support? If you are using configure you need to use the configure option `--enable-ipv6` .

That worked for me! Thanks, @rizlik

Flole998, 2023-01-23:
I'll double-check, thanks.

Would it be possible to add a meaningful error message to indicate that something that's not compiled in is attempted? It should be possible to try to check the socket family.

rizlik, 2023-01-23:
This is a good point, thanks @Flole998 . Let's first be sure that this was indeed the problem, then I'll try to make wolfSSL return a more helpful error.

closed 2023-05-24 by JacobBarthelmeh