wolfSSL/wolfssl #6008: Functions missing documentation

PaulMartinsen, 2023-01-24:
### Version

latest

### Description

The following functions (to support loading certificates with passwords) are not documented, though getters are (e.g., [`wolfSSL_CTX_get_default_passwd_cb`](https://www.wolfssl.com/documentation/manuals/wolfssl/group__Setup.html#function-wolfssl_ctx_get_default_passwd_cb))
* `wolfSSL_CTX_set_default_passwd_cb` (see `ssl.c`, line 16932)
* `wolfSSL_CTX_set_default_passwd_cb_userdata` (see `ssl.c`, line 16923)

Also, the documented functions don't mention that defining `WOLFSSL_ENCRYPTED_KEYS` may be required to use them. 

These functions support loading of certificates that use passwords. Is this a supported feature or will it be deprecated?

kareem-wolfssl, 2023-01-24:
Thanks for the report, I will look into these functions' documentation.
Yes, that is a supported feature and should be working, are you having any issues loading a cert with a password?

PaulMartinsen, 2023-01-25:
Thanks @kareem-wolfssl . 

I've been able to load a certificate without difficulty, but a partner working on a different architecture (maybe Linux) is getting missing symbol link errors. So we're trying to figure out what's missing and worried the feature might be unsupported. So glad it is!

BTW: I just noticed there seems no documentation for the preprocessor symbol WOLFSSL_POST_HANDSHAKE_AUTH, which is required for: 
https://www.wolfssl.com/documentation/manuals/wolfssl/group__Setup.html#function-wolfssl_ctx_allow_post_handshake_auth

closed 2025-12-26 by dgarske