wolfSSL/wolfssl #6014: [Bug]: libwolfssl5.5.4-stable will crash if the ssl certificate contains Chinese characters

osnosn, 2023-01-26:
### Version

5.5.4-stable

### Description

```
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            a0:d0:20:30:90:b0:00:00:00:15
        Signature Algorithm: ecdsa-with-SHA512
        Issuer: C=CN, ST=\xE5\x8C\x97\xE4\xBA\xAC, L=bj, O=Home, CN=Wifi \xE6\x88\x91\xE5\xAE\xB6 Root CA
        Validity
            Not Before: Jan 26 06:56:09 2023 GMT
            Not After : Jan 23 06:56:09 2033 GMT
        Subject: C=CN, ST=\xE5\x8C\x97\xE4\xBA\xAC, L=bj, O=Home, CN=WiFi \xE6\x88\x91\xE5\xAE\xB6 Radius Server
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
```
ca_key : ec256 ; server_key : RSA2048

Both ST and CN in the certificate contain Chinese characters.
I use "hostapd" as radius authentication server.
"hostapd" using wolfssl library cannot pass TLS authentication, and "hostapd" will crash out.

I use "hostapd" of openssl library instead, no problem.

If the certificate does not contain Chinese characters, "hostapd" using wolfssl library works normally.

dgarske, 2023-01-26:
Hi @osnosn ,

We will attend to reproduce. Can you share your build settings for wolfSSL?

Also, can you try with our `--enable-asn=template` build option? We are planning to make this the default soon and it has some advanced parsing that may help.

Thanks,
David Garske, wolfSSL

anhu, 2023-01-27:
Hi @osnosn , 

Unfortunately, I was not able to reproduce what you were seeing. I made a leaf and root certificate chain where many of the fields are "中国" (China). The PEM of the certs are: 

```
-----BEGIN CERTIFICATE-----
MIIEHzCCAwegAwIBAgIUaduygVWzTPmVF0kqLYGC+mKQIWAwDQYJKoZIhvcNAQEL
BQAwgZ4xCzAJBgNVBAYTAkNOMRUwEwYDVQQIDAzDpMK4wq3DpcKbwr0xFTATBgNV
BAcMDMOkwrjCrcOlwpvCvTEVMBMGA1UECgwMw6TCuMKtw6XCm8K9MRUwEwYDVQQL
DAzDpMK4wq3DpcKbwr0xFTATBgNVBAMMDMOkwrjCrcOlwpvCvTEcMBoGCSqGSIb3
DQEJARYNcm9vdEBjaGluYS5jbjAeFw0yMzAxMjcxMzEyNDBaFw0yMzAyMjYxMzEy
NDBaMIGeMQswCQYDVQQGEwJDTjEVMBMGA1UECAwMw6TCuMKtw6XCm8K9MRUwEwYD
VQQHDAzDpMK4wq3DpcKbwr0xFTATBgNVBAoMDMOkwrjCrcOlwpvCvTEVMBMGA1UE
CwwMw6TCuMKtw6XCm8K9MRUwEwYDVQQDDAzDpMK4wq3DpcKbwr0xHDAaBgkqhkiG
9w0BCQEWDXJvb3RAY2hpbmEuY24wggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEK
AoIBAQDDvIZIj4aAboaXg3m9H5RwAyunW8/7ErQn+K6ALSo39L5yamWkmGMgQyYg
9Lb+Z9Aan3fzdCj31s0AlR7c2w5QwSpWmb4I12sEUhW/G47k7VQfI0JJ7x9aciQr
SSKEJOzGodGVwZ9Y/RDFvgQx5hKs0QYhg5DJUecz6YnVqfTTVM88dIVGl4uUeANw
HZACafd6hD6lxIe8XMXRJEA5Yo+/TH3Gvzh/Ro9XuBlAlhjbejv1fcpyB5QvA64x
HLm7cm9qCjxU0p075PGZvzrAVAbG20zQ224sbVoE/kUZrMMHPHWkTPho/Hwnqlcc
aSY5D+Gkza3K3w5ctyU9FoRv+PdLAgMBAAGjUzBRMB0GA1UdDgQWBBSyigBtWkHA
2sZQPAagUqTfF+h1YDAfBgNVHSMEGDAWgBSyigBtWkHA2sZQPAagUqTfF+h1YDAP
BgNVHRMBAf8EBTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQBtTbRmunCrTF4xemD7
NoNzkVaz26L8FOh0DibyYD0hwo7udT+Iyiz1hURIpbMunzRztJKsOdTO/OGJFEMl
KNjVkBIQJ76bKHEQnAHttntqiIZepzIHKcjeiJrtxHsRgGBW2a7p+HuC6IBFeddJ
CHR1JN5HsMgTlsAnVD4Dda7ZmKrqxg7Qw7bt61vsK04zV8fh4E822Y1T/iVcPF86
A/7srSBmhNBOArX2J8OoE6En1V9Uko4NmrJzGw3bNuObd2xLhHQvSE7/ozcrFHdb
vH4VlnFUmNdMQ3K/CLUAtx6xmvK6LMZQVsFJnq8Dn+69Oq/T0kxaj7J6YQStjO8L
WOaJ
-----END CERTIFICATE-----
```

Leaf Certificate: 

```
-----BEGIN CERTIFICATE-----
MIIDrzCCApcCAQYwDQYJKoZIhvcNAQELBQAwgZ4xCzAJBgNVBAYTAkNOMRUwEwYD
VQQIDAzDpMK4wq3DpcKbwr0xFTATBgNVBAcMDMOkwrjCrcOlwpvCvTEVMBMGA1UE
CgwMw6TCuMKtw6XCm8K9MRUwEwYDVQQLDAzDpMK4wq3DpcKbwr0xFTATBgNVBAMM
DMOkwrjCrcOlwpvCvTEcMBoGCSqGSIb3DQEJARYNcm9vdEBjaGluYS5jbjAeFw0y
MzAxMjcxMzE3MjlaFw0yMzAyMjYxMzE3MjlaMIGbMQswCQYDVQQGEwJDTjEVMBMG
A1UECAwMw6TCuMKtw6XCm8K9MRUwEwYDVQQHDAzDpMK4wq3DpcKbwr0xFTATBgNV
BAoMDMOkwrjCrcOlwpvCvTEVMBMGA1UECwwMw6TCuMKtw6XCm8K9MRIwEAYDVQQD
DAkxMjcuMC4wLjExHDAaBgkqhkiG9w0BCQEWDWxlYWZAY2hpbmEuY24wggEiMA0G
CSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQC9Rzu1ri9k70B4uvzmEdg3PHqKo2B6
DELqL+UKbEXOSRMfPUzLumoUxIrxwFIQF6fAX1AxjKmJpVmrSF2cfnUALy/bWFiA
KslugdXmp+h+xXUnIQhQIsbuaJpc/asbz6u6RnznS7JHp2hY4moRBVF0hKYUZkdo
C1YqZHTVVIibEsEE7uuZaxGDgMnkZmLtAPD8t8zhVU8KTj2ziRuBqJWlBcSEGCi0
ZalabukLCx1GI5IjkTOt0M4w2YtDEdNdu7EzYRdv70qNP5nAv/mygCPC1oNYnlvA
gGBIJvno+LTRAvuM6er2fSdeUQ9lIUyFqvKfXd7mxaazBr6J5iHco+OFAgMBAAEw
DQYJKoZIhvcNAQELBQADggEBAG1pbB5BL6LJk3D4l5fGoLv5pha5PTg81BgSfuNB
eTmpRagXvGawbG9ve5xZl3R1Dgy4IV3q5jVBLVkXWZy9hQgtffwuJHmaTWHx2BVY
nJaVUL89y5I8rGLGxtL/SFGw7lwD6T0lojreot5tkU8+qRB1qO2PdbnAqmhDmd6G
jGlyvSpwY+5el6t5E0moJRvBf7fvTcWmyedbE4slAz2+LfXPksXtcCGkePX8ElQQ
yRJRjtInGLlZZue/72VHK+5EUegJlJE28VpmhuNekALafjUkm+c4h8ozU3/z2M53
DUT2SLoQsTkKOCbEm/BQS7ZlsX4Hy4NQhmyMAKYK49f5P0E=
-----END CERTIFICATE-----
```
I executed our example server and client and they connected without any error. 

```
./examples/server/server -c leafchina.pem  -k leafprivkey.pem
```

```
/examples/client/client -A rootchina.pem
```

I tried with `./configure` as well as `./configure --enable-asn=template` and in both cases, did not have any problems. Can you please share your build settings as well as the certificates you used? If these are non-production certificates used for testing, could you also share the keys? 

Warm regards, Anthony

anhu, 2023-01-27:
Quick update: was able to reproduce with `--enable-all` .  Both  `./configure --enable-all` as well as `./configure --enable-all --enable-asn=template` have the certificate verification failure. I will investigate further.

anhu, 2023-01-27:
After further investigation, I realized that the failure I was seeing was due to failure during CRL check. Once I disable CRL checking on the example client, the connection works fine. 

Server side: 
```
./examples/server/server -c leafchina.pem -k leafprivkey.pem -d 
peer has no cert!
SSL version is TLSv1.2
SSL cipher suite is ECDHE-RSA-AES256-GCM-SHA384
SSL signature algorithm is (null)
SSL curve name is SECP256R1
Server Random : 23EE0790BC4D1B62EDE3FA038A3BCBB24A4A7824997D14957EC316ED5AFD5ADF
Client message: hello wolfssl!
```

Client Side:
```
$ examples/client/client -A rootchina.pem -C
Session Ticket CB: ticketSz = 178, ctx = initial session
Alternate cert chain used
 issuer : /C=CN/ST=ä¸­å=ä¸­å=ä¸­åU=ä¸­åN=ä¸­åmailAddress=root@china.cn
 subject: /C=CN/ST=ä¸­å=ä¸­å=ä¸­åU=ä¸­åN=127.0.0.1/emailAddress=leaf@china.cn
 serial number:06 
SSL version is TLSv1.2
SSL cipher suite is ECDHE-RSA-AES256-GCM-SHA384
SSL curve name is SECP256R1
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIDrzCCApcCAQYwDQYJKoZIhvcNAQELBQAwgZ4xCzAJBgNVBAYTAkNOMRUwEwYD
VQQIDAzDpMK4wq3DpcKbwr0xFTATBgNVBAcMDMOkwrjCrcOlwpvCvTEVMBMGA1UE
CgwMw6TCuMKtw6XCm8K9MRUwEwYDVQQLDAzDpMK4wq3DpcKbwr0xFTATBgNVBAMM
DMOkwrjCrcOlwpvCvTEcMBoGCSqGSIb3DQEJARYNcm9vdEBjaGluYS5jbjAeFw0y
MzAxMjcxMzE3MjlaFw0yMzAyMjYxMzE3MjlaMIGbMQswCQYDVQQGEwJDTjEVMBMG
A1UECAwMw6TCuMKtw6XCm8K9MRUwEwYDVQQHDAzDpMK4wq3DpcKbwr0xFTATBgNV
BAoMDMOkwrjCrcOlwpvCvTEVMBMGA1UECwwMw6TCuMKtw6XCm8K9MRIwEAYDVQQD
DAkxMjcuMC4wLjExHDAaBgkqhkiG9w0BCQEWDWxlYWZAY2hpbmEuY24wggEiMA0G
CSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQC9Rzu1ri9k70B4uvzmEdg3PHqKo2B6
DELqL+UKbEXOSRMfPUzLumoUxIrxwFIQF6fAX1AxjKmJpVmrSF2cfnUALy/bWFiA
KslugdXmp+h+xXUnIQhQIsbuaJpc/asbz6u6RnznS7JHp2hY4moRBVF0hKYUZkdo
C1YqZHTVVIibEsEE7uuZaxGDgMnkZmLtAPD8t8zhVU8KTj2ziRuBqJWlBcSEGCi0
ZalabukLCx1GI5IjkTOt0M4w2YtDEdNdu7EzYRdv70qNP5nAv/mygCPC1oNYnlvA
gGBIJvno+LTRAvuM6er2fSdeUQ9lIUyFqvKfXd7mxaazBr6J5iHco+OFAgMBAAEw
DQYJKoZIhvcNAQELBQADggEBAG1pbB5BL6LJk3D4l5fGoLv5pha5PTg81BgSfuNB
eTmpRagXvGawbG9ve5xZl3R1Dgy4IV3q5jVBLVkXWZy9hQgtffwuJHmaTWHx2BVY
nJaVUL89y5I8rGLGxtL/SFGw7lwD6T0lojreot5tkU8+qRB1qO2PdbnAqmhDmd6G
jGlyvSpwY+5el6t5E0moJRvBf7fvTcWmyedbE4slAz2+LfXPksXtcCGkePX8ElQQ
yRJRjtInGLlZZue/72VHK+5EUegJlJE28VpmhuNekALafjUkm+c4h8ozU3/z2M53
DUT2SLoQsTkKOCbEm/BQS7ZlsX4Hy4NQhmyMAKYK49f5P0E=
-----END CERTIFICATE-----
Session timeout set to 300 seconds
Client Random : 294E3982F483BCB5A4DEF50573836142079D5F36AD279BCF44A1BC8F6AEBFA9C
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
    Session-ID: FCD51CD53CD29CCE0A59D657004B1C7B00000000000000000000000000000000
    Session-ID-ctx: 
    Master-Key: AB84857A1661FC5EE733E0770689AB99F11379B8143AAAEE3C9561FC7A1F7F54085C27456E94B5C391AFA07A0096C567
    TLS session ticket:
    0000 - 06 2E 49 FC DA CB C0 E4-51 05 EB 8F C3 26 19 9C   ..IlJK.dA.k.C...
    0010 - 2C FE 0B 98 3E 81 88 F2-73 BB 5D 3A 4C A3 30 CE   .n.....bc.M.L..N
    0020 - 00 70 81 49 61 B8 6B F9-F8 43 1A C5 79 30 1E 6A   .`.Ia.kihC.Ei..j
    0030 - 82 E8 EB 1A AE 05 92 A1-C7 2A F9 ED 14 E3 9F AE   .hk.....G.im.c..
    0040 - 37 43 DE B2 F8 2A 4E D5-E2 85 21 3B 3E E9 34 14   .CN.h.NEb....i..
    0050 - EB F8 A9 AB 36 71 38 9A-76 C6 44 29 4B EC B7 28   kh...a..fFD.Kl..
    0060 - 12 B0 49 A1 4C C9 42 3D-E2 46 80 26 91 21 2D F4   ..I.LIB.bF.....d
    0070 - AA 1C 9A 11 02 64 C2 04-D6 85 30 20 1E FA 47 5F   .....dB.F....jGO
    0080 - 4F AF B6 55 B2 BE 0D 11-2C 19 2F 16 67 4A 09 51   O..E........gJ.A
    0090 - 67 9F FC D5 1C D5 3C D2-9C CE 0A 59 D6 57 00 4B   g.lE.E.B.N.IFG.K
    00A0 - 1C 7B 00 00 00 00 00 00-00 00 00 00 00 00 00 00   .k..............
    00B2 - 00 00                                             ..
    Start Time: 1674849727
    Timeout   : 300 (sec)
    Extended master secret: yes
I hear you fa shizzle!
```
The `-C` turns off CRL checks.  So, it turns out that I cannot reproduce what you are seeing.  @osnosn , Can you please let me know your build configuration flags and your certificates and keys? 

Warm regards, Anthony

Spudz76, 2023-01-27:
Build environment can be gleaned from the OpenWRT build tree [wolfssl/Makefile](https://github.com/openwrt/openwrt/blob/master/package/libs/wolfssl/Makefile#L118);
defaults are summarized here for reference:
```
CFLAGS = -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_NAMES
LDFLAGS = -flto

CONFIGURE_ARGS += \
        --enable-reproducible-build \
        --enable-lighty \
        --enable-opensslall \
        --enable-opensslextra \
        --enable-sni \
        --enable-stunnel \
        --enable-altcertchains \
        --enable-crypttests \
        --disable-examples \
        --disable-jobserver \
        --enable-ipv6 \
        --enable-aesccm \
        --enable-certgen \
        --enable-chacha \
        --enable-poly1305 \
        --enable-dh \
        --enable-arc4 \
        --enable-tlsv10 \
        --enable-tls13 \
        --enable-session-ticket \
        --disable-dtls \
        --enable-curve25519 \
        --enable-curve448 \
        --enable-openvpn \
	--enable-ocsp \
	--enable-ocspstapling \
	--enable-ocspstapling2 \
	--enable-wpas \
	--enable-fortress \
	--enable-fastmath
        
# --enable-stunnel needed for OpenSSL API compatibility bits
```

CFLAGS may also include `-fPIC` depending on platform, CONFIGURE_ARGS may include `--enable-intelasm` if x86_64, or `--enable-armasm` if aarch64.  There is also [one patch](https://github.com/openwrt/openwrt/blob/master/package/libs/wolfssl/patches/100-disable-hardening-check.patch) which forces `WC_NO_HARDEN` logic to be ignored for some unknown reason.

anhu, 2023-01-27:
I have applied the patch, and use the `LDFLAGS` and `CFLAGS` and all the flags above except `--disable-examples`, but I still cannot reproduce an authentication failure. 

Would it be possible to add `--enable-debug` and calling `wolfSSL_Debugging_ON()` ?  This will give us much more information and perhaps give a hint as to where the problem lies. 

Warm regards, Anthony

Spudz76, 2023-01-27:
I am not the OP however I do have the same platform (ramips/mt7621) and could attempt to reproduce the configuration.  I think my usual build is against mbedtls so it will take some recompile time as every SSL consumer package must be switched to wolfssl.  The default libc for OpenWRT and this ramips platform is musl (`mipsel_24kc_musl`) which could have something to do with it.

It could very well be some implementation bug in [hostapd](https://github.com/openwrt/openwrt/blob/master/package/network/services/hostapd/Makefile#L106), or one of the [numerous patches](https://github.com/openwrt/openwrt/tree/master/package/network/services/hostapd/patches) for that.

OP was also using RADIUS with hostapd so there could be a whole extra usually-unused-usage path there.  I am not sure I can fully duplicate the environment to reproduce the bug (stub out a test RADIUS server seems not-fun).

Spudz76, 2023-01-27:
I will look into a patch file so that OP could more easily rebuild with those debug enablements.

Spudz76, 2023-01-27:
I am already using wolfssl, not sure where the RADIUS option goes.

anhu, 2023-01-30:
Hello, 
Is there anything I can do to expedite getting the logs? 
Warm regards, Anthony

osnosn, 2023-02-02:
非常感谢。

anhu, 2023-02-02:
不客氣. 

However, I have not reproduced this yet. @osnosn , can you please enable debugging and attach some logs? 

Warm regards, Anthony

osnosn, 2023-02-02:
> 不客氣.
> 
> However, I have not reproduced this yet. @osnosn , can you please enable debugging and attach some logs?
> 
> Warm regards, Anthony

好的，我找时间，把 hostapd 的 log 截取出来。

osnosn, 2023-02-03:
```
root@OpenWrt:~# logread -e hostapd
Fri Feb  3 13:53:11 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: authenticated
Fri Feb  3 13:53:11 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: associated (aid 1)
Fri Feb  3 13:53:11 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:11 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:53:11 2023 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 5c:e0:c5:5c:ac:e2 off
Fri Feb  3 13:53:11 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:11 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:53:14 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-RETRANSMIT 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:34 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: authenticated
Fri Feb  3 13:53:34 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: associated (aid 1)
Fri Feb  3 13:53:34 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:34 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:53:34 2023 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 5c:e0:c5:5c:ac:e2 off
Fri Feb  3 13:53:34 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:34 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:53:37 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-RETRANSMIT 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:43 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-RETRANSMIT 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:51 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-STARTED 00:00:00:00:00:00
Fri Feb  3 13:53:51 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-PROPOSED-METHOD method=1
Fri Feb  3 13:53:51 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=25
Fri Feb  3 13:53:52 2023 daemon.notice hostapd: EAP-PEAP: TLV Result - Failure - requested Failure
Fri Feb  3 13:53:52 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-FAILURE 00:00:00:00:00:00
Fri Feb  3 13:53:52 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-FAILURE2 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:53:52 2023 daemon.warn hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.1X: authentication failed - EAP type: 25 (PEAP)
Fri Feb  3 13:54:02 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: authenticated
Fri Feb  3 13:54:02 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: associated (aid 1)
Fri Feb  3 13:54:02 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:02 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:54:02 2023 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 5c:e0:c5:5c:ac:e2 off
Fri Feb  3 13:54:02 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:02 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:54:05 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-RETRANSMIT 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:11 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-STARTED 00:00:00:00:00:00
Fri Feb  3 13:54:11 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-PROPOSED-METHOD method=1
Fri Feb  3 13:54:11 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=25
Fri Feb  3 13:54:12 2023 daemon.notice hostapd: EAP-PEAP: TLV Result - Success - requested Success
Fri Feb  3 13:54:12 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-SUCCESS 00:00:00:00:00:00
Fri Feb  3 13:54:12 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-SUCCESS2 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:12 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 WPA: pairwise key handshake completed (RSN)
Fri Feb  3 13:54:12 2023 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:12 2023 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:12 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 RADIUS: starting accounting session 8127CFA01E459060
Fri Feb  3 13:54:12 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
Fri Feb  3 13:54:21 2023 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:27 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: authenticated
Fri Feb  3 13:54:27 2023 daemon.info hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.11: associated (aid 1)
Fri Feb  3 13:54:27 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:27 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:54:27 2023 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 5c:e0:c5:5c:ac:e2 off
Fri Feb  3 13:54:27 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:27 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Feb  3 13:54:30 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-RETRANSMIT 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:31 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-STARTED 00:00:00:00:00:00
Fri Feb  3 13:54:31 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-PROPOSED-METHOD method=1
[... cut ...]
Fri Feb  3 13:54:43 2023 daemon.notice hostapd: : CTRL-EVENT-EAP-FAILURE 00:00:00:00:00:00
Fri Feb  3 13:54:43 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-FAILURE2 5c:e0:c5:5c:ac:e2
Fri Feb  3 13:54:43 2023 daemon.warn hostapd: wlan1: STA 5c:e0:c5:5c:ac:e2 IEEE 802.1X: authentication failed - EAP type: 13 (TLS)
root@OpenWrt:~#
```
* use `wpad-wolfssl`.
* 4 test,
  1. PEAP, fail.  i type a wrong pwd.
  2. PEAP, ok. 
  3. TLS, use correct user certification. but fail.
  4. TLS, use correct user certification. but fail.

anhu, 2023-02-03:
It is unfortunate that I cannot see any indications of what has gone wrong.  I cannot reproduce this with wolfssl's tools using the same build configurations you use. I'm guessing you are not running a debug build version.  Can you perhaps re-build a test system that reproduces this issue?  Or can someone give me instructions on how to build the system?

osnosn, 2023-02-03:
Sorry.
I don't know how to build wolfssl in debug mode in OpenWRT-22.03.3.
I only know how to opening the debug information of hostapd.

But this BUG can be reproduced.

osnosn, 2023-02-04:
@anhu 
I have an idea.

Try [a similar way](https://github.com/wolfSSL/wolfssl/issues/6014#issuecomment-1413946211) on the x86 virtual machine.
* Use [x86_64 SDK](https://downloads.openwrt.org/releases/22.03.2/targets/x86/64/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64.tar.xz), re-build wpad-wolfssl with RADIUS option.
* Maybe can rebuild "libwolfssl" in debug mode.
* Install [openwt-22.03.3 x86_64](https://downloads.openwrt.org/releases/22.03.2/targets/x86/64/openwrt-22.03.2-x86-64-generic-ext4-combined-efi.img.gz) in kvm virtual machine.
* SSH log into openwrt, replace the custom builded package.
* Run hostapd as a RADIUS server. 
  * Both wpad and hostapd in openwrt are linked to the same program.
* Use tool "eapol_test" in Debian system to test the RADIUS server, TLS certification test.

Spudz76, 2023-02-04:
I made a test branch of openwrt-master with all the mods provided and including the two debug enablements: [test-wpad-wolfssl-enable-debug](https://github.com/openwrt/openwrt/compare/master...Spudz76:openwrt:test-wpad-wolfssl-enable-debug)

The [raw patch](https://github.com/openwrt/openwrt/compare/master...Spudz76:openwrt:test-wpad-wolfssl-enable-debug.patch) should apply on top of your 22.03.3 SDK tree even though I based it on master.

Spudz76, 2023-02-04:
Also the openwrt-22.03 tag has a different `PKG_RELEASE` but can be adjusted by hand if it rejects.

Although your patch above shows the `.2` just like master.

```
 PKG_NAME:=hostapd
-PKG_RELEASE:=$(AUTORELEASE).1
+#PKG_RELEASE:=$(AUTORELEASE).1
+PKG_RELEASE:=15.3
```

osnosn, 2023-02-05:
I use op-22.03.3 SDK,  NOT source code. 
It shows hostapd version `PKG_RELEASE:=$(AUTORELEASE).2` .
You can change to:
```
-PKG_RELEASE:=$(AUTORELEASE).2
+PKG_RELEASE:=$(AUTORELEASE).3
```
It is same. In op-22.03.3, `AUTORELEASE=15`.

Spudz76, 2023-02-05:
The patches are for you to test again but now it should provide the wolfSSL debug messages.

anhu, 2023-02-08:
Hi @osnosn ,

Have you been able to apply the patches and follow @Spudz76 's instructions?  I'd love to see the logs.

osnosn, 2023-02-12:
I'm really sorry.
I have no time for the moment.
In addition, the mt7621 device is installed with wpad-openssl and is in use. It cannot be removed for testing.
I will find another time to test in the kvm of x86.

osnosn, 2023-02-27:
Copy it into a file. then RUN `base64 -d filename > logs.tgz`
```
H4sIAAAAAAAAA+yaSW+cR5KGdfavqFurDzRyidwMeACOrMEIsGVBSx/mlkukJUAteiiqgfn380SR
1kJJ3WrKHmPcol0iq/gtmZFvvEt+fHr28qL/vL5+fvbTrd/qy/GVRY7f+br2Pbrk4y0vsRQnOZZ0
y3lfJN46uN9sRG99vWL+54fDrfOzs4u/d9w/+v3/069vDs9eXOj57lMPVOJCD0/u37t/7/G90+/v
/dfd707+7e7903///u53X31zOH1wcvXm8NXD0+/uPXl0+Ku+fNl/0m8O82zpt/5w+3ROPjp5qP/9
Sl9e/PnwbOmLi2f7mZ5/6w7P9cVPF0+/9SF8dTgcTi8uzp+NV9yR85681POT+/2v+udfjsp2DF9/
6c9fcYM/XXC9P717nhxu3z99dHLvwcnpWufc92Mn+1C+dvzn3z0/cuM7/fnzZy9+OnnE3J+dvTi5
t15fxLdrQ3DhxLm3/vfXxuPD4fZ/nDOJdfLD4ycfHYw49+552V9O5MHZ+cXJ4//5+aNFuBzRWyce
bj/S8789m/p3T7tW7lKY9tmLFzovTu692GevTwtybcJ3frx//+6dxwfvfxg/vzxUF772flybdWmH
23fBxg+XWHhTPv/u1VxwnXZvjubOqcQi716nusPtq2ucnL66eGrAmf3i7PzNFeu7V0y7SPGlVt2q
1fUcp2+hq2zn1lgKZu88fvj9yd2/3L3/+MTG+Ojx6cPH4Ne5b979//1DHzz88cGPj+iAH+4+/s8f
vwPqF0/P1rf+Hx/5N32xzgzvV6eE9JF2edMvd54CQyapH+6YdG0FA8g3wH50xa8Y1n3aSl0rq/Nu
GGNTS/+5S9RHEu2SxvIi0mSXVnrrbSQnI+V9EyLxb4gkfSGSPz6RXIdnuIQnxmH9Km1xA+KpIw3N
vk83SpMQapp+zbmr3z7q/uc5wscbcoT/3ThiHhdhhc+upnO0VWq9pJT2xASOqDFQ16y9rxhuwhHh
l5vJ5d2+cMQfmyNCSu+xxARXzS3gZVwRSRUKbHnru71vbrQxY+OQoEt9QqRG6LGEXLaARxxF7GmW
FEd3c42WU6LpXZo7o2bWO6Lerhtd4BV5Ca80XZi8xuQ2fMeNNHspWYezepmzN16VFwwSCq/Iq/IS
uorPXOPF547PPJ95fu+9TWjya17Ki2tmmDAXZ5NxMdpM5+T4yfHTp+NsmYRzYR2DVrDZXx4pnCWJ
t/bdO5nUKumWUWLyLeuUqq0OcdxV5hBdMUuMszQrSi7aVyth7ThkjVpGGTKXDOkzDG5WjQnHFGXq
/NDK6tWX0FupqUsPY6y8JUuh10ef2AHqdXSHlXHKcb1svPE4By54HLoeWT9dHWG/rXxS0YPKp6wX
L/k0wMh1NqzcO3M3qxqzPt45HO9u47r8yR0dLOto5Oepua+XFfSGsqt6g7bjl31+vfn+zxQKG8w4
wPfMtqTN55qalD1I2fysN1Sb17TqaZDfVG/eb2gPDFIZbs3L6yQdV43t+NTw4Uqx99q3rFVHH33T
pyPUrIbGwD/Z5RiW1B1Cz2AystCSl3S6KwZxYNGa4hJ2zoLEsVneapQw26hrp64gu+havWe6bs26
6YcU24pcNtnlSxkZzA/ZdcaZ5spT3So7RU+7sDoryNC0XA1hFfBfR2utBy2eaIOp2LQgiKv05mw7
01nqrxrFwC++XM7fQ1hcycApfQX4rG0YYww/fJkpdm68Z+z0T56puF/O45iS615+MfNc8ti1cevu
8gxlRXXoL93WwuyrFt9T2as7kUKV1pgr+r2T717LGGVGNfascSyCA0gb1kDTexlzQCgRzlpLVqfa
5WoEHMPYdmFlEqPubs+9m3aKuZKnTgrF3hAwJc2YW5kSMBW5zg3dY9M8w9hQ1ZoCrZcRZoitCnwU
V8tNG60jkxH41rbWQoU6lMWih816jjyb1ATpAbTZmbPiBHdIqwhrtLvfACymGcQ3DU1Qm54oRQo5
derpG0Kya6y1Z8pHHkpbqiuykaI6S9wpNOq1qTWUDJxmQ1Ka6wkhkh5lt45keYFY655jxOXrUFYp
SQNTXWPqa81RtlGUuD5QOmmltdVL9RP27fBs0ZRoqcVHadXNe9AcRijDc4+Vc0xJCG9Lcp8trQaA
V0EBuZ8fTCGM7KkMMHAljdB8p1qZ+3V+3AmRS9K7QTqHunuYYE+RkNRZ/Vy8xzmDrjJHSIihW9E6
y/RZNWvY4jYrF8B7cxullCQhZqDhKPiaXdHmKhom8E+e7rB1AJ3jhoDZo46pJbTdk6MOVszakOTa
x6g1x7EBvmGKTgqeFmkctSTZmoZBWyc6vlAXO1YTjLvHFsxGCT75rXP7Yg01oqSgvkoflVrxTo/4
yfAVt4ji19guESV6rTFTm55CpVPgjd4W2lwgt0ANgXEEQYIuxxQrLGH6H0TBZ1thq/ayZhrTkMIF
p9PUUkQSaHQfNA8QwRy65EADANlA0yWH//YqRgD8SO/72pquhDarYEsoM4QlPmfEmzJNoLoSRDZC
ZEgDcsREFIwB0OCkAbEErSnr8rinbRwxi4ez4Q8H5fSBKm2JeXAUDEghjk6mM5MEQ1JJiKwanEFV
8GtPicGok/aLu5QZbHmSn3CQo3MqyI5QNWCBO+lhv2gkbybFGkWHIAo7mxzEHW8IGJYee1QwYoFy
NUyctaLWUOukjlAL1cFOViQmCk1RmjULEJ+lpIzYVGXEyvRpsx1mwXjA2WBYWDwcUsqwAlJyXF2f
gQ2dwvJBZBH9cNIUl1VZw6wl9QK9wzLYP4+ocYkS4OG9RZupI61JQp21psj4gE83Jdq9OFCGAGAM
KtgqnM/oqF7AzQUaAOB5jZkPaU07p6y2hpo1jWllpS9dTSOyBlKbcmsyHAEu1pKl17Gj27GXSGAu
GrKRvkHBmL+43iCFwKJhuCEsjoY6F+ZF6QZXgsG5NlrQ0YR0PMMtWxdtwsKbTQNWxnXK5XNQBQWI
euoRlu5IFZZ8MYu6s5sK8CBMSHUxMgM8tmCZUo4J5rRJ8prBLk0BtxRMJZIRM3zXCxHB1FSkfqLH
fA8wSkGKurjzmh4oYN8hG1IASr3SaKnb+IsUlpiRrQq1KB5lwQwamC+9Q9oYqk2RfA4LBhAViD/l
HmJqHl6h20gTJgSFZYYSe7Id0Z7dxveb6IwhwIzzoJkcPJwSoONIQ8PXGBe/aoxgsGibkBtWY0fr
qo0LaTiYwNrMCbRUwEOe0ihRovP85CadNQOOfsEd3SbUlQvSipm2ZmLOogalxWKsNgqyhUUKBhLp
CjTCNkiosh6jZ8Vso1e0f8UN4YEaVIjCTQtNVz6CwavOgcLIyowaoQ7FYXxh4gKxcZlCyGvUTAYA
BY6JPAb3rFSAtBdcQRIKwwtfgCOCcMEmzGMCW+CvBrzJR6wG3EPrIKh+dDGc9pkbPYrUV8EhDMtY
+ElohtmQllz6NMCQ1q8xDMDEGUxMU8dT0WyNeVERfBYaEVi7UnUq/I3gKKuZsCALlwq3wesrrZ4U
v0VB0+ias6yNZa3mOgLTg1JLbgkSgcO4LPkS0s8YmJTNkSDIrCZVQrzWRpZJoClizqLLjWUSuD1R
H06TTR1DhQXgJ0wTgIWw6F1vRhvUuMzJwA5dK6mg3wMWJFfWMnPhFvTd9G5Omt4joqY0EauFkJpL
bbJkqtTxudFoRDoKFoqpegDYUwVHERPmMB2U5ybbTfHLlvS/1HbT+1vS63I39HcL/FEh8oHJJAig
vRPfh/HzrcP15D65YeB/A+zrj+t+5bxP5H4v7yvR97j5I5jBjmyjEROWgfzVglSMEQetc1lGa80v
R06xvSRCie3ZmVdCRVF0pLRv8hZqJS3kib02cw3XzGFGtWDE0cIhGEXdguhBq/hlDO2Aa1/n5UQ1
J5G0EtcJ9XiElcIgDlTkkZiNNOOKycu6N5Kb4JNaYc2Brq9qSYEUig+sueN5MHx17QKXhrQx8t1s
PCfCqv6zH47ipwguOhpOfZCOXMUbY1axMiy4rpvQnLy+mQvlC8/98XnuQ9vq+Nd6ZLqrrsA/j5Dx
STjjLBUBDxXPjT1f9ODIdSzsIpmY4/B/Qt9pK1gny3kxtEUxV12r0N40Eu4Fm1VpZCwWVt72FIgI
k8/NOPIiareWCQOaQl+kpBxioMNCjYuRFWJ2O1rpHHPOxW6YPJGz0Q8ZHlE4wtG+Ds9NCObUzC3d
wGeIKOPFY++6B5G87ST1uMFne+AL16wdvw4b4ceb00r4wsxjZ3A25LRKkNVjCs84FiqyLZdY45NI
61zURFIkqLdCVMaTidUqEEbnZP44LOiBAraGz+sRW0dYDhixaVZSMUl16MAOhuXJ8Bg+ezBh22Uw
HUPFJ5ERJ0xlO1XDpSauVoLGxCvComlW26hwO3UcK2bdch9rUfIQIiUVIol1bPZY0qJTC5UVp73K
TTNyZEiWW0h7MiEiEhR2HROPa0wEgGi0vkrxltInRF3wuQoL5227tM1ShyW5AvA28Y2cXAq5FK5U
1hbAsJAoRMCzGjoQvlAx434zXaLObpGboBZTC+5VCTUzVRWWGr+7K+6eyCEkhj0m0XHltDvLxSoS
OsImgncvnXiI0DQlpxUUoqYigVnYjh+wJRKDbYmZCW2Li5jezS81N6JXWxhtcqvrJAJ6xxsnu0a8
7D5h6VkrRbC4FLa7V3C582gMpGD/lVS8ZkbM6ggioN82zcgcXABwFaCTWkfnwQTCQSgjwDhhuN7b
0xkLWagApUQw0UaalCKihcOLbX1ulsTKuaOlzRasLxq467Y7R+OMXkqJKdAnm+TFne0Jlw6KROhK
6PBwZTUsBiEuRKIG/5DY/bxpRg5q2wqbOLtRRbSWSBqxMZPS0/iMtFaPJWiJ5JEkimDkPXlUiNCl
2Tra4qDinWgJxFOfIaO4vU2CdKaMBMM1J3HUutSlyPBZwhA8WMmhg3oygu0+VCFPQl0kymhJr9kT
MHxFJGJ7E3K4hKOpgm0B5UYrR2VwYntyhaRGWG8AhbnQpA4qYR1qBk1aCwEzKYMI8GBj/HtsnAm5
rSnOpGTnJTO0mvImnVmqA9wVmkghlla5L81MWp9k+eJqKKQ621hOKbUIvxCTQ4NRgm1KQ3nk/+GO
e4eN6YDGNY97BCTzDRHDm5saKvxufRSaNNfJ8NX2Oxd5qYYOkc1A2e1hFxyxdVtf+CV0Adk8u5Yp
[... 36 lines cut ...]
7RBIOtAIjPbydBb5Jd1vzNRZq6IL5mXLaetOdYRvLOfabl0pEb9t3UiuYmibFTe10Nq+CLxic3Dd
STY6Hw1vVUHUZq1Vmg67BXJalaUKsXYhjS31/YD+1Jp/t+T5nz50PGUbH8j/vCiJ7+X/UpHl/+co
[... cut ...]
* I installed `wpad-wolfssl_2022-01-16-cff80b4f-13.1` and `libwolfssl5.5.1.ee39414e (with debug mode)`, which provide command "hostapd".
* then RUN: `hostapd -dd hostapd.conf`
  * use `eapol-test-openssl - 2022-01-16-cff80b4f-15.2` , test OK, Authentication succeeded。
  * use `eapol-test-wolfssl - 2022-01-16-cff80b4f-15.2`+`libwolfssl5.5.4.ee39414e`, Authentication failed.
    The above base64 file is the result of this failed test.

It seems that 5.5.1 is normal. This bug was introduced from 5.5.4.

closed 2025-11-17 by anhu