wolfSSL/wolfssl #6023: huge size compare to openssl 1.1

calvin2021y, 2023-01-28:
### Version

v5.5.4-stable

### Description

```sh
   Features 
   * FIPS:                       no
   * Single threaded:            yes
   * Filesystem:                 no
   * OpenSSH Build:              no
   * OpenSSL Extra API:          yes
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * Reproducible Build:         no
   * Side-channel Hardening:     no
   * Single Precision Math:      yes
   * SP implementation:          all
   * Fast Math:                  no
   * Heap Math:                  no
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AES-CBC:                    yes
   * AES-CBC length checks:      no
   * AES-GCM:                    yes
   * AES-GCM streaming:          no
   * AES-CCM:                    no
   * AES-CTR:                    no
   * AES-CFB:                    no
   * AES-OFB:                    no
   * AES-SIV:                    no
   * DES3:                       no
   * Camellia:                   no
   * NULL Cipher:                no
   * MD2:                        no
   * MD4:                        no
   * MD5:                        yes
   * RIPEMD:                     no
   * SHA:                        yes
   * SHA-224:                    yes
   * SHA-384:                    yes
   * SHA-512:                    yes
   * SHA3:                       yes
   * SHAKE128:                   no
   * SHAKE256:                   no
   * BLAKE2:                     no
   * BLAKE2S:                    no
   * SipHash:                    no
   * CMAC:                       no
   * keygen:                     no
   * certgen:                    no
   * certreq:                    no
   * certext:                    no
   * certgencache:               no
   * CHACHA:                     yes
   * XCHACHA:                    no
   * Hash DRBG:                  yes
   * MmemUse Entropy:            no
   * PWDBASED:                   yes
   * scrypt:                     no
   * wolfCrypt Only:             no
   * HKDF:                       yes
   * X9.63 KDF:                  no
   * MD4:                        no
   * PSK:                        yes
   * Poly1305:                   yes
   * LEANPSK:                    no
   * LEANTLS:                    no
   * RSA:                        yes
   * RSA-PSS:                    yes
   * DSA:                        no
   * DH:                         yes
   * DH Default Parameters:      yes
   * ECC:                        yes
   * ECC Custom Curves:          no
   * ECC Minimum Bits:           224
   * FPECC:                      no
   * ECC_ENCRYPT:                no
   * Brainpool:                  no
   * CURVE25519:                 yes
   * ED25519:                    yes
   * ED25519 streaming:          no
   * CURVE448:                   no
   * ED448:                      no
   * ED448 streaming:            no
   * KYBER:                      no
   * KYBER wolfSSL impl:         no
   * ECCSI                       no
   * SAKKE                       no
   * ASN:                        yes
   * Anonymous cipher:           no
   * CODING:                     yes
   * MEMORY:                     yes
   * I/O POOL:                   no
   * wolfSentry:                 no
   * LIGHTY:                     no
   * WPA Supplicant:             no
   * HAPROXY:                    no
   * STUNNEL:                    no
   * tcpdump:                    no
   * libssh2:                    no
   * ntp:                        no
   * rsyslog:                    no
   * Apache httpd:               no
   * NGINX:                      no
   * OpenResty:                  no
   * ASIO:                       no
   * LIBWEBSOCKETS:              no
   * Qt:                         no
   * Qt Unit Testing:            no
   * SIGNAL:                     no
   * chrony:                     no
   * strongSwan:                 no
   * OpenLDAP:                   no
   * ERROR_STRINGS:              yes
   * DTLS:                       no
   * DTLS v1.3:                  no
   * SCTP:                       no
   * SRTP:                       no
   * Indefinite Length:          no
   * Multicast:                  no
   * SSL v3.0 (Old):             no
   * TLS v1.0 (Old):             no
   * TLS v1.1 (Old):             yes
   * TLS v1.2:                   yes
   * TLS v1.3:                   yes
   * Post-handshake Auth:        no
   * Early Data:                 yes
   * QUIC:                       no
   * Send State in HRR Cookie:   no
   * OCSP:                       yes
   * OCSP Stapling:              yes
   * OCSP Stapling v2:           yes
   * CRL:                        no
   * CRL-MONITOR:                no
   * Persistent session cache:   yes
   * Persistent cert    cache:   no
   * Atomic User Record Layer:   no
   * Public Key Callbacks:       no
   * liboqs:                     no
   * Whitewood netRandom:        no
   * Server Name Indication:     yes
   * ALPN:                       yes
   * Maximum Fragment Length:    yes
   * Trusted CA Indication:      no
   * Truncated HMAC:             yes
   * Supported Elliptic Curves:  yes
   * FFDHE only in client:       no
   * Session Ticket:             yes
   * Extended Master Secret:     yes
   * Renegotiation Indication:   no
   * Secure Renegotiation:       no
   * Fallback SCSV:              no
   * Keying Material Exporter:   no
   * All TLS Extensions:         no
   * PKCS#7:                     no
   * S/MIME:                     no
   * wolfSSH:                    no
   * wolfEngine:                 no
   * wolfTPM:                    no
   * wolfCLU:                    no
   * wolfSCEP:                   no
   * Secure Remote Password:     no
   * Small Stack:                no
   * Linux Kernel Module:        no
   * valgrind unit tests:        no
   * LIBZ:                       no
   * Examples:                   no
   * Crypt tests:                no
   * Stack sizes in tests:       no
   * Heap stats in tests:        no
   * User Crypto:                no
   * Fast RSA:                   no
   * Async Crypto:               no
   * PKCS#8:                     yes
   * PKCS#11:                    no
   * PKCS#12:                    yes
   * Cavium Nitrox:              no
   * Cavium Octeon (Sync):       no
   * Intel Quick Assist:         no
   * ARM ASM:                    no
   * ARM ASM SHA512/SHA3 Crypto  no
   * AES Key Wrap:               no
   * Write duplicate:            no
   * Xilinx Hardware Acc.:       no
   * Inline Code:                yes
   * Linux AF_ALG:               no
   * Linux KCAPI:                no
   * Linux devcrypto:            no
   * Crypto callbacks:           no
   * i.MX CAAM:                  no
   * IoT-Safe:                   no
   * IoT-Safe HWRNG:             no
   * NXP SE050:                  no
   * Maxim Integrated MAXQ10XX:  no
   * PSA:                        no
   * System CA certs:            no
   * ERR Queues per Thread:      yes
```

Try build with CURL static link,  after strip openssl 3.1M, wolfssl 3.3M.

I see the document said wolfssl is much smaller like 1/10 compare to openssl.   how can I get this small size ?

calvin2021y, 2023-01-28:
openssl enabled features: 
```sh
    afalgeng
    asm
    autoalginit
    autoload-config
    bf
    blake2
    chacha
    cmac
    cms
    deprecated
    dgram
    dh
    dsa
    dtls
    ec
    ecdh
    ecdsa
    engine
    hw(-.+)?
    makedepend
    multiblock
    nextprotoneg
    ocsp
    poly1305
    posix-io
    psk
    rdrand
    seed
    sock
    srtp
    sse2
    ssl
    static-engine
    stdio
    threads
    tls
    weak-ssl-ciphers
    tls1
    tls1-method
    tls1_1
    tls1_1-method
    tls1_2
    tls1_2-method
    tls1_3
    dtls1
    dtls1-method
    dtls1_2
    dtls1_2-method
```

kareem-wolfssl, 2023-01-30:
Hi @calvin2021y ,

Please share your full ./configure line for wolfSSL.  Which platform are you building for?

calvin2021y, 2023-02-02:
Thanks for the tips:

```sh
./configure -enable-shared=no --enable-harden=no --enable-filesystem=no --enable-pwdbased=no --enable-sni --enable-alpn --enable-truncatedhmac --enable-earlydata --enable-tlsv10=no --enable-oldtls=yes --enable-tlsv12=yes --enable-tls13 --enable-rsa --enable-psk-one-id --enable-session-ticket --enable-savesession --enable-sessioncerts --enable-rng --enable-aescbc=yes --enable-aescfb=no --enable-aesccm=no --enable-aesctr=no --enable-aesctr=no --enable-maxfragment=yes --enable-blake2=no --enable-blake2s=no --enable-hkdf=no --enable-sys-ca-certs=no --enable-examples=no --enable-crypttests=no --enable-singlethreaded --enable-asynccrypt=no --enable-asyncthreads=no --enable-sha384 --enable-asm --enable-sp --enable-sp-asm --enable-hugecache --enable-curve25519=yes --enable-ed25519=yes --enable-crl=no --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-opensslall --disable-opensslextra --enable-hrrcookie=no
```

The size is compare with static linked curl binary, with LTO and strip all symbols.

kareem-wolfssl, 2023-02-02:
Our configure has an --enable-curl option which I'd recommend you use.  SP math assembly is faster, but also larger, so I'd recommend disabling this if size is the priority.  Note that most options which reduce size in wolfSSL are a tradeoff against performance.
Make sure you are passing -Os or your compiler's equivalent via CFLAGS during the configure step, for both wolfSSL and curl.  By default we use -O2 which optimizes for performance, not size.
`CFLAGS='-Os' ./configure --enable-curl --enable-sp=small --disable-sp-asm`

On a separate note, we strongly recommend against the use of `--enable-harden=no`, may I ask where you found this setting?

calvin2021y, 2023-02-06:
Thanks for the very helpful explain.   now I an able to build android with smaller size.  but I get problem when build with linux or osx:

calvin2021y, 2023-02-06:
> On a separate note, we strongly recommend against the use of --enable-harden=no, may I ask how you came up with your configure line?

I remember last time I turn it on there is a error, now I can enable it.

calvin2021y, 2023-02-06:
The `lld: error: relocation` error only exits for x86_64, not arm or x86.   

Is there a word around for this ?

calvin2021y, 2023-02-06:
I test on android, and there is problem with curl DOH, not sure cause by wolfssl.

```sh
* DoH request SSL peer certificate or SSH remote key was not OK
* DoH: Too small type A for
* Couldn't resolve host name
```
openssl work ok.

kareem-wolfssl, 2023-02-06:
What compiler/toolchain are you using and what version is it?  Does updating your toolchain help?
I would recommend trying to use the configure line I posted above: `CFLAGS='-Os' ./configure --enable-curl --enable-sp=small,asm --enable-intelasm`

What host are you trying to connect to?  Does your system have CA certificates installed?

calvin2021y, 2023-02-07:
> Thanks for the very helpful explain. now I an able to build android with smaller size. but I get problem when build with linux or osx:
> 
> ```shell
> --enable-shared=no --enable-harden --enable-filesystem=no --enable-pwdbased=no --enable-sni --enable-alpn --enable-truncatedhmac --enable-earlydata --enable-tlsv10=no --enable-oldtls=yes --enable-tlsv12=yes --enable-tls13 --enable-rsa --enable-psk-one-id --enable-session-ticket --enable-savesession --enable-sessioncerts --enable-rng --enable-aescbc=yes --enable-aescfb=no --enable-aesccm=no --enable-aesctr=no --enable-aesctr=no --enable-maxfragment=yes --enable-blake2=no --enable-blake2s=no --enable-hkdf=no --enable-hc128=no --enable-sys-ca-certs=no --enable-examples=no --enable-crypttests=no --enable-singlethreaded --enable-asynccrypt=no --enable-asyncthreads=no --enable-sha384 --enable-asm --enable-sp=small,asm --enable-aesni --enable-intelasm --enable-intelrand --enable-sp=small --enable-hugecache --enable-curl --enable-curve25519=yes --enable-ed25519=yes --enable-crl=no --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-hrrcookie=no
> ```
> 
> ```shell
> lld: error: relocation SIGNED_1 is out of range: 2461107603 is not in [-2147483648, 2147483647]; references _gRandMethodsInit
> lld: error: relocation SIGNED_1 is out of range: 2461107612 is not in [-2147483648, 2147483647]; references _gRandMethodsInit
> lld: error: relocation SIGNED_1 is out of range: 2461107385 is not in [-2147483648, 2147483647]; references _gRandMethodsInit
> lld: error: relocation SIGNED is out of range: 2461106051 is not in [-2147483648, 2147483647]; references _wolfSSL_BIO_s_mem.meth
> lld: error: relocation SIGNED is out of range: 2461105794 is not in [-2147483648, 2147483647]; references _wolfSSL_BIO_s_mem.meth
> lld: error: relocation SIGNED_1 is out of range: 2461105834 is not in [-2147483648, 2147483647]; references _wolfSSL_BIO_s_mem.meth
> lld: error: relocation SIGNED is out of range: 2460890541 is not in [-2147483648, 2147483647]; references _implementation.5132
> lld: error: relocation SIGNED_1 is out of range: 2460890559 is not in [-2147483648, 2147483647]; references _implementation.5132
> lld: error: relocation SIGNED_4 is out of range: 2460889997 is not in [-2147483648, 2147483647]; references __sodium_lock
> lld: error: relocation SIGNED_4 is out of range: 2460890005 is not in [-2147483648, 2147483647]; references _initialized.5139
> lld: error: relocation SIGNED_1 is out of range: 2460890053 is not in [-2147483648, 2147483647]; references __cpu_features.7
> lld: error: relocation SIGNED_1 is out of range: 2460890094 is not in [-2147483648, 2147483647]; references __cpu_features.3
> lld: error: relocation SIGNED_1 is out of range: 2460890123 is not in [-2147483648, 2147483647]; references __cpu_features.5
> lld: error: relocation SIGNED_1 is out of range: 2460890145 is not in [-2147483648, 2147483647]; references __cpu_features.6
> lld: error: relocation SIGNED_1 is out of range: 2460890203 is not in [-2147483648, 2147483647]; references __cpu_features.8
> lld: error: relocation SIGNED is out of range: 2460890229 is not in [-2147483648, 2147483647]; references _canary
> lld: error: relocation SIGNED is out of range: 2460890284 is not in [-2147483648, 2147483647]; references __cpu_features.11
> lld: error: relocation SIGNED is out of range: 2460890292 is not in [-2147483648, 2147483647]; references __cpu_features.8
> lld: error: relocation SIGNED_4 is out of range: 2460890323 is not in [-2147483648, 2147483647]; references __cpu_features.8
> lld: error: relocation SIGNED is out of range: 2460890329 is not in [-2147483648, 2147483647]; references __cpu_features.7
> ```

after enable asm, this error gone.

calvin2021y, 2023-02-07:
> What compiler/toolchain are you using and what version is it? Does updating your toolchain help? I would recommend trying to use the configure line I posted above: `CFLAGS='-Os' ./configure [...]

I test on all hosts, right now most of work.  but not ios and mingw x86 32. 

CA is related to #6063 and should be fixed.

closed 2023-02-07 by calvin2021y