wolfSSL/wolfssl #6024: [Bug]: (sha512.c:1309:29: error: 'sha512' undeclared)

llange, 2023-01-29:
### Version

tag `v5.5.4-stable` or `master`

### Description

Compiling OVMSv3 for ESP-IDF version 5+ (NOT using the component integration (https://github.com/espressif/esp-wolfssl/)).

It may be a not-often-used compiler path, but it seems that the commit 8bb6ff8dbbc6ea300998aae431af0a24c8ccc5b5 could have introduced an issue here:
```
static int InitSha384(wc_Sha384* sha384)
{
....
        esp_sha_hw_unlock(&(sha512->ctx));
....
```

the only variable defined at this point is `sha384` (which looks OK by the name of the function).

With the obvious patch it compiles this file OK:
```
@ wolfcrypt/src/sha512.c:1309 @ static int InitSha384(wc_Sha384* sha384)
     sha384->ctx.isfirstblock = 1;
     if(sha384->ctx.mode == ESP32_SHA_HW) {
         /* release hw */
-        esp_sha_hw_unlock(&(sha512->ctx));
+        esp_sha_hw_unlock(&(sha384->ctx));
     }
     /* always set mode as INIT
     *  whether using HW or SW is determined at first call of update()
```

What do you think ? Could it be a typo, or did I miss some configuration somewhere ?

### Relevant log output

```shell
[59/268] Building C object esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfssl/wolfcrypt/src/sha512.c.objFAILED: esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfssl/wolfcrypt/src/sha512.c.obj 
SRCS/ESP-IDF-TOOLS/5.0/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DWOLFSSL_USER_SETTINGS -ISRCS/OVMS-wt2/vehicle/OVMS.V3/build/config -ISRCS/OVMS-wt2/vehicle/OVMS.V3/components/wolfssl/port -ISRCS/OVMS-wt2/vehicle/OVMS.V3/components/wolfssl/wolfssl -ISRCS/ESP-IDF/5.0/components/newlib/platform_include -ISRCS/ESP-IDF/5.0/components/freertos/FreeRTOS-Kernel/include -ISRCS/ESP-IDF/5.0/components/freertos/esp_additions/include/freertos -ISRCS/ESP-IDF/5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -ISRCS/ESP-IDF/5.0/components/freertos/esp_additions/include -ISRCS/ESP-IDF/5.0/components/esp_hw_support/include -ISRCS/ESP-IDF/5.0/components/esp_hw_support/include/soc -ISRCS/ESP-IDF/5.0/components/esp_hw_support/include/soc/esp32 -ISRCS/ESP-IDF/5.0/components/esp_hw_support/port/esp32/. -ISRCS/ESP-IDF/5.0/components/esp_hw_support/port/esp32/private_include -ISRCS/ESP-IDF/5.0/components/heap/include -ISRCS/ESP-IDF/5.0/components/log/include -ISRCS/ESP-IDF/5.0/components/soc/include -ISRCS/ESP-IDF/5.0/components/soc/esp32/. -ISRCS/ESP-IDF/5.0/components/soc/esp32/include -ISRCS/ESP-IDF/5.0/components/hal/esp32/include -ISRCS/ESP-IDF/5.0/components/hal/include -ISRCS/ESP-IDF/5.0/components/hal/platform_port/include -ISRCS/ESP-IDF/5.0/components/esp_rom/include -ISRCS/ESP-IDF/5.0/components/esp_rom/include/esp32 -ISRCS/ESP-IDF/5.0/components/esp_rom/esp32 -ISRCS/ESP-IDF/5.0/components/esp_common/include -ISRCS/ESP-IDF/5.0/components/esp_system/include -ISRCS/ESP-IDF/5.0/components/esp_system/port/soc -ISRCS/ESP-IDF/5.0/components/esp_system/port/include/private -ISRCS/ESP-IDF/5.0/components/xtensa/include -ISRCS/ESP-IDF/5.0/components/xtensa/esp32/include -ISRCS/ESP-IDF/5.0/components/esp_timer/include -ISRCS/ESP-IDF/5.0/components/lwip/include -ISRCS/ESP-IDF/5.0/components/lwip/include/apps -ISRCS/ESP-IDF/5.0/components/lwip/include/apps/sntp -ISRCS/ESP-IDF/5.0/components/lwip/lwip/src/include -ISRCS/ESP-IDF/5.0/components/lwip/port/esp32/include -ISRCS/ESP-IDF/5.0/components/lwip/port/esp32/include/arch -mlongcalls -Wno-frame-address  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Os -freorder-blocks -fmacro-prefix-map=SRCS/OVMS-wt2/vehicle/OVMS.V3=. -fmacro-prefix-map=SRCS/ESP-IDF/5.0=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -Wno-cpp -Wno-char-subscripts -MD -MT esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfssl/wolfcrypt/src/sha512.c.obj -MF esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfssl/wolfcrypt/src/sha512.c.obj.d -o esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfssl/wolfcrypt/src/sha512.c.obj -c SRCS/OVMS-wt2/vehicle/OVMS.V3/components/wolfssl/wolfssl/wolfcrypt/src/sha512.c
SRCS/OVMS-wt2/vehicle/OVMS.V3/components/wolfssl/wolfssl/wolfcrypt/src/sha512.c: In function 'InitSha384':
SRCS/OVMS-wt2/vehicle/OVMS.V3/components/wolfssl/wolfssl/wolfcrypt/src/sha512.c:1309:29: error: 'sha512' undeclared (first use in this function); did you mean 'Sha512'?
 1309 |         esp_sha_hw_unlock(&(sha512->ctx));
      |                             ^~~~~~
      |                             Sha512
```

gojimmypi, 2023-01-29:
Hi @llange and that you for pointing this out. An excellent find! Thank you. I believe you are correct.

It appears the reason this was never noticed before is that the [default user_settings.h](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/user_settings.h) does [not enable SHA384](https://github.com/wolfSSL/wolfssl/blob/6e9d956a492c0b4df04384d5d1fd0e9fabeb297b/IDE/Espressif/ESP-IDF/user_settings.h#L58).

llange, 2023-01-29:
Hi @gojimmypi , thanks for the confirmation !
Should I create the PR or do you want to go ahead with it ?

gojimmypi, 2023-01-29:
Thanks for the offer. A contributor agreement is typically needed to merge a PR, so I'll take care of it. 

Let me know if you find anything else.

llange, 2023-01-31:
Thank you @gojimmypi for the merged PR.

closed 2023-01-31 by llange