wolfSSL/wolfssl #6025: Various uninitialized variables and typos

gojimmypi, 2023-01-29:
### Version

Latest master branch

### Description

[AlizTheHax0r on Twitter](https://twitter.com/AlizTheHax0r/status/1617925118986190850?s=20&t=vLzCK73zsrvODfUgTCrfzA) pointed out several items that need to be reviewed and cleaned up: 

[1.](https://twitter.com/AlizTheHax0r/status/1618195363701354499?s=20&t=vLzCK73zsrvODfUgTCrfzA)  The symbol redefinition errors - one for `THREAD_RETURN` and one for `INFINITE`. I bodged these by removing the offending definitions from `Espressif/ESP32/ESP32-SSH-Server/main/ssh_server.h` (lines 58 and 66).

[2.](https://twitter.com/AlizTheHax0r/status/1618196040691372033?s=20&t=vLzCK73zsrvODfUgTCrfzA) The uninitialised var is in wolfssh itself - in SendKexInit, `privateKey` and `privateKeyCount` on 7836 and 7837 in `wolfssh/src/internal.c`

[3.](https://twitter.com/AlizTheHax0r/status/1618271592005660673?s=20&t=vLzCK73zsrvODfUgTCrfzA) The 'output' and 'idx' in SendKexDhReply (internal.c:8477) and 'authSz' in SendUserAuthFailure on 11191

[4.](https://twitter.com/AlizTheHax0r/status/1618293206504050689?s=20&t=vLzCK73zsrvODfUgTCrfzA) Found a typo - in CMakeLists.txt:56, is this open-bracket supposed to be there?

[5.](https://twitter.com/AlizTheHax0r/status/1618293206504050689?s=20&t=vLzCK73zsrvODfUgTCrfzA)  Is that the correct -D option or should it be LINUX

[6.](https://twitter.com/AlizTheHax0r/status/1618271592005660673?s=20&t=vLzCK73zsrvODfUgTCrfzA) [uninitialized] 'output' and 'idx' in SendKexDhReply (internal.c:8477) and 'authSz' in SendUserAuthFailure on 11191

randomdude, 2023-01-29:
Hi, thanks for opening this for me! I hope I'm not being pedantic (I don't contribute to OSS projects often so please forgive any missteps on my part!), but I did notice a couple other minor things:

/wolfssh/src/internal.c around 5578 contains the line "    (void)tmpIdx;", I'm guessing it's introduced during a refactor since that variable 'tmpIdx' isn't in scope.

I got some 'unused variable' errors due to tmpIdx in DoKexDhReply - I ended up enclosing the tmpIdx declaration with "#ifndef WOLFSSH_NO_RSA". 

At the risk of being pedantic, once you fix issues 4 and 5 (the CMakeLists issue), thre are comments in my_config.h around 60-63 which you might want to update as well to match.

Finally - I think the tilde in ```if(EXISTS "(~/my_private_config.h")``` doesn't actually expand. I had some difficulty getting it working, and https://gitlab.kitware.com/cmake/cmake/-/issues/21729 states that it doesn't expand at all. 

Hope this is helpful! I'd share my fixes but I haven't emailed about the contributor agreement yet so I don't want to do so in case it causes more problems than it solves.

gojimmypi, 2023-01-29:
@randomdude thank _you_ for taking the time to point out these issues! Yes, your observations and suggestions are all appreciated. 

That's one of the cool benefits of open source: lots of eyes on the code to catch various potential problems.

Thanks again for taking my SSH to UART project for a test drive! I'm so glad it is working for you. Can you tell me more about your project using it as mentioned in your [original tweet](https://twitter.com/AlizTheHax0r/status/1617853885208735745?s=20&t=9cyyu0wq5Oe94fijbbUHmg)?

randomdude, 2023-01-30:
Sure! It's nothing unusual, really. It's just for my 'home network'. I've got a reasonably complex network setup, probably about six switches running with a whole host of vlans, mostly for traffic isolation (I'm a security researcher by day, so I like to keep my home network fairly not-insecure). Since it's a home setup, all the switches are of different brands and are quite low-end, so managing them is a little bit annoying - logging into a variety of poor web interfaces got tiring quickly, so I wrote a load of code that takes a yaml file specifying switch configuration, and applies it to the switches. It works quite well, but occasionally I'll make a mistake with my vlan'ing, and the switch will drop off the network - so I'm trying to move toward having them configurable via a proper out-of-band channel, which is where RS232 comes in.

The switch right now is I'm putting in is 'DIN rail' mounted, so I got a little din-rail rs232-to-wifi module on [aliexpress](https://www.aliexpress.com/item/1005004999249478.html), but it turned out to be close to unusable because of shoddy firmware (it's impossible to change the password, for example, and it uses no encryption other than at the 'wifi layer'). I cracked it open, though, and it's just an ESP32.. that's when this project came along and saved the day :)

Honestly though, I might end up putting little rs232-to-wifi boards in more devices going forward. I've got an audio amplifier that has RS232, it'd be neat if I could wifi it up instead.

Sorry, got a little bit rambly there :) I should really start a blog so I can talk about all this stuff..

gojimmypi, 2023-01-30:
@randomdude that's cool you've found good use for the SSH to UART server. Thanks for sharing the details. Just so that I understand: you reflashed it onto the din-rail rs232-to-wifi module you found on Aliexpress?! That's just excellent! I'd love to read a blog about that. That idea is so cool, perhaps I'll include it on the README page. Having full RS-232 and not just TTL level UART communication would open up that project to a much wider audience.

As a security researcher, perhaps you can appreciate my [port sniffing blog](https://gojimmypi.github.io/Espressif-ESP32-WiFi-Port-Sniffing-DUT/). I'm certain it is nothing nearly elaborate as your setup, but I find it quite helpful. Recently I bought Netgear EN104 dumb hubs on [ebay](https://www.ebay.com/itm/115663953993). It is surprisingly difficult to find hubs that are not elaborate managed switches. Although my Edgerouter-X mentioned in the blog is easy to port span / mirror, I was working on another device (dd-wrt router) that simply would not cooperate to allow me to monitor wired ethernet packets with Wireshark. 

Thanks again for sharing all your observations & ideas.

randomdude, 2023-01-31:
Yep that's correct, reflashed the module I got from AliExpress :) I'll blog about it "sometime" (by which I mean I'll add it to the long list of things I want to do but may never get around to :) )

Neat setup! I hadn't thought about hubs being difficult to acquire, but it makes sense when I think about it. I haven't had call to do anything that elaborate yet. Usually my response to 'just would not co-operate' hardware is to use a USB NIC - I found some really cheap, but didn't realise they were only 100mbit, so I just use them for testing and that kind of thing now.

gojimmypi, 2024-06-21:
I'm closing this lingering issue as I believe most have been addressed, but I've not yet confirmed them.

I've moved this to my Mothball section to not forget: https://github.com/wolfSSL/wolfssl/issues/6234

closed 2024-06-21 by gojimmypi