wolfSSL/wolfssl #6036: wolfssl error: ssl_accept error-313

sandilya761, 2023-01-31:
### Version

5.5.4

### Description

error in following commands:

./examples/server/server -d -l ECDHE-ECDSA-AES256-SHA -c ./certs/server-ecc.pem -k ./certs/ecc-key.pem
./examples/client/client -A ./certs/server-ecc.pem

Following is the error:

[image]

[image]

embhorn, 2023-01-31:
Hello @sandilya761 

Please try loading the CA cert in the client, not the server's cert:

```
./examples/client/client -A ./certs/ca-ecc-cert.pem
```

Thanks,
@embhorn - wolfSSL Support

embhorn, 2023-01-31:
Also, `ECDHE-ECDSA-AES256-SHA` is a TLS v1.0 cipher suite…  You would need `--enable-oldtls` to support it.
Please understand this is a very OLD cipher suite, and we actively discourage using these in new projects.

sandilya761, 2023-02-01:
Thanks @embhorn. It's working.

closed 2023-02-01 by sandilya761