wolfSSL/wolfssl #6044: [Bug]: TLSX_UseSRTP_GetSize returns incorrect value

TargoDove, 2023-02-02:
### Contact Details

targodove@yahoo.com

### Version

5.5.4

### Description

In "wolfssl/src/tls.c" in function "TLSX_UseSRTP_GetSize":

The returned value is (OPAQUE16_LEN + (srtp->profileCount * OPAQUE16_LEN) + 1).

However, going through function "TLSX_UseSRTP_Write" it appears that the correct formula is:
(OPAQUE16_LEN +  ((srtp->profileCount/2) * (srtp->profileCount * OPAQUE16_LEN)) + 1)

### Reproduction steps

To reproduce this, ensure that srtp->profileCount is greater than 3.

e.g. 
- if you enable a number of profiles such that "srtp->ids" has a value of "0b110000110", then "srtp->profileCount" will be equal to 4.
- If you then call "TLSX_UseSRTP_GetSize", it will return a value of "11" (2 +(4*2)+1).
- If however you call "TLSX_UseSRTP_Write", it will produce an offset of "19" (2 + ((4/2) * (4*2)) + 1).

### Relevant log output

```shell
I got the error "ERROR: DTLS: err = -328, malformed buffer input error" which is what led me to discovering this bug
```

embhorn, 2023-02-02:
Hi @TargoDove 

Thanks for sharing this. I tested your change and saw a failure in the unit test:

```
starting dtls srtp suite tests
trying server command line[3271]: SuiteTest -u -d -l ECDHE-RSA-AES256-GCM-SHA384 --srtp -2 -p 0 
Using SRTP Profile(s): SRTP_AES128_CM_SHA1_80
trying client command line[3272]: SuiteTest -u -x --srtp -l ECDHE-RSA-AES256-GCM-SHA384 -2 -p 47132 
Using SRTP Profile(s): SRTP_AES128_CM_SHA1_80
SSL_accept error -328, malformed buffer input error
wolfSSL error: SSL_accept failed
```

I am requesting a review of this issue by our engineers.

Thanks,
@embhorn - wolfSSL Support

ejohnstown, 2023-02-02:
Per RFC 5764, the function TLSX_UseSRTP_GetSize() is producing the correct size. I believe the function TSLX_UseSRTP_Write() is misbehaving. Write should be taking the bitfield `ids` and only processing `profileCount` of them. It looks like for each profileCount, it is writing all the profiles set.

Given `ids` equal to 0b00000110, and `profileCount` of 2, Write should output:

    00 02 00 02 00 03 00

For your example, Write should output:

    00 04 00 02 00 03 00 08 00 09 00

It outputs:

    00 04 00 02 00 03 00 08 00 09 00 02 00 03 00 08 00 09 00

(It is only producing two sets of four because the loop over profileCount is counting by 2.)

I'm submitting a fix for this.

closed 2023-02-03 by ejohnstown