wolfSSL/wolfssl #6058: Linking Issues using static library

g0dzcsgo, 2023-02-05:
### Version

Latest

### Description

Hello, I am trying to build my project that uses wolfssl to no avail. Since I am getting some strange linker errors.

I downloaded the latest version of wolfssl, and then built it. I then included the library in my project aswell as the include files such as "ssl.h" etc. And added these to the directories tab under Include Directories and Library Directories.

I also added wolfssl.lib into my input for the linker.

But I am still getting linker errors such as:

1>LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library
1>wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol __imp__mktime64
1>wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol __imp__gmtime64
1>wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol __imp_fgets
1>wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol __imp_rewind
1>wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol __imp__stricmp
1>wolfssl.lib(ssl.obj) : error LNK2001: unresolved external symbol __imp_strncat
1>wolfssl.lib(wc_port.obj) : error LNK2001: unresolved external symbol __imp__stat64i32
1>wolfssl.lib(asn.obj) : error LNK2001: unresolved external symbol __imp_strtok_s
1>wolfssl.lib(asn.obj) : error LNK2001: unresolved external symbol __imp__time64

Any help regarding to fix this would be greatly appreciated!

Thanks, kind regards.

gojimmypi, 2023-02-05:
Hi @g0dzcsgo can you please add a few more details about your exact build environment / platform / versions? Given the reference to MSVCRT that sounds like a Visual Studio project with the Microsoft Visual C Runtime? If so, was the [Using wolfSSL with Visual Studio](https://www.wolfssl.com/docs/visual-studio/) helpful?

If you are building on *Nix, have your tried [these instructions](https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html#building-on-nix)? 

Any additional details with be very helpful in diagnosing the problem here. Thank you.

gojimmypi, 2023-02-14:
Hello @g0dzcsgo I'm following up to see if I can help you with the linker issue.

If you are working on a private project that cannot be discussed here, you can reach out to me directly: jim@wolfssl.com or the general support desk:  support@wolfssl.com

g0dzcsgo, 2023-02-14:
> Hello @g0dzcsgo I'm following up to see if I can help you with the linker issue.
> 
> If you are working on a private project that cannot be discussed here, you can reach out to me directly: [jim@wolfssl.com](mailto:jim@wolfssl.com) or the general support desk: [support@wolfssl.com](mailto:support@wolfssl.com)

Hey, I solved the issue was just me being stupid :)

gojimmypi, 2023-02-14:
That's excellent! Thanks for following up. 

May I ask exactly what the problem & resolution was? If you've encountered it, others likely will too. 

Perhaps there's something I can update in the docs to help avoid this.

current, 2023-02-14:
> Hey, I solved the issue was just me being stupid :)

how did you end up fixing it?

closed 2023-02-15 by gojimmypi

Berthalamew, 2024-02-28:
Just following up, @g0dzcsgodo you remember how you resolved this?

Berthalamew, 2024-02-28:
I think I've resolved the issue with this, 
it seems as if the project files for wolfssl cmake generated for Visual Studio had the "Runtime Library" option set to use "Multi-Threaded DLL" rather than "Multi-Threaded".
Hope this helps anyone who encounters the same issue.