wolfSSL/wolfssl #6142: [Bug]: TLSX_SetResponse ifdef-Preprocessor-directives not sufficient

Flole998, 2023-02-28:
### Version

5.5.4

### Description

If none of the several modules that use TLSX_SetResponse are enabled an error is shown at compile-time. The `#ifdef` preprocessor-directive is not sufficient and additional checks are needed there. For example `HAVE_ALPN`, `HAVE_SNI`, `HAVE_TRUSTED_CA` are guarding the usages, if none of those are defined the function ends up being defined but not used.

### Reproduction steps

Try to compile the example using the PR from https://github.com/RIOT-OS/RIOT/pull/19302

### Relevant log output

```shell
wolfssl/src/tls.c:1411:13: error: ‘TLSX_SetResponse’ defined but not used [-Werror=unused-function]
 1411 | static void TLSX_SetResponse(WOLFSSL* ssl, TLSX_Type type)
      |             ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

Flole998, 2023-02-28:
It's one of the tests that's failing, the one in `examples/dtls-wolfssl`. The build settings can be found in it's Makefile.

Flole998, 2023-02-28:
So it looks like TLS 1.3 is enabled (as it's required for DTLS) but none of the extensions mentioned above. That's enough to cause the declaration to appear, but there are no usages in that case.

kareem-wolfssl, 2023-02-28:
@Flole998 Thanks for the clarification.  Please give this patch a try.

```
diff --git a/src/tls.c b/src/tls.c
index 5265a9a67..d5a9f7f56 100644
--- a/src/tls.c
+++ b/src/tls.c
@@ -1412,7 +1412,13 @@ int TLSX_HandleUnsupportedExtension(WOLFSSL* ssl)

 #endif

-#if !defined(NO_WOLFSSL_SERVER) || defined(WOLFSSL_TLS13)
+#if defined(WOLFSSL_TLS13) || (!defined(NO_WOLFSSL_SERVER) \
+    && (defined(WOLFSSL_SRTP) || defined(HAVE_SESSION_TICKET) \
+    || defined(HAVE_SECURE_RENEGOTIATION) || defined(HAVE_SERVER_RENEGOTIATION_INFO) \
+    || defined(HAVE_SUPPORTED_CURVES) || defined(HAVE_CERTIFICATE_STATUS_REQUEST_V2) \
+    || defined(HAVE_CERTIFICATE_STATUS_REQUEST) || defined(HAVE_TRUNCATED_HMAC) \
+    || defined(HAVE_MAX_FRAGMENT) || defined(HAVE_TRUSTED_CA) || defined(HAVE_SNI) \
+    || defined(HAVE_ALPN)))
 /** Mark an extension to be sent back to the client. */
 static void TLSX_SetResponse(WOLFSSL* ssl, TLSX_Type type)
 {
```

Flole998, 2023-02-28:
Thanks, that one worked!

kareem-wolfssl, 2023-02-28:
Great, will work on getting that fix upstreamed then.

PR #6144 (2023-02-28): Fix for PIC32 crypto HW and unused `TLSX_SetResponse`
# Description

* Avoid building TLSX_SetResponse unless it is being used.
* Initialize PIC32 UPDPTR register in all cases.

Fixes zd# 15741
Fixes #6142

# Testing

Test suite and customer tests

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2023-02-28 by dgarske