wolfSSL/wolfssl #6143: Zephyr module broken

donusry, 2023-02-28:
### Version

master

### Description

After #6043, wolfSSL is no longer following the required folder structure for Zephyr modules.

Second bullet of https://docs.zephyrproject.org/3.1.0/develop/modules.html#module-repositories

donusry, 2023-03-02:
Hey, everyone. I wanted to come back and update this issue with some more info.

Kconfig symbols such as `CONFIG_WOLFSSL` and `CONFIG_WOLFSSL_BUILTIN` are not visible using the current module since the modules Kconfig could not be found. I have resolved this issue on a local branch by moving the `zephyr` folder out of `RTOS` and back into the root.

Also, when building any of the samples the Zephyr headers cause errors because they are not found. For instance `<kernel.h>` is now `<zephyr/kernel.h>`.

dgarske, 2023-03-02:
Hi @donusry ,

Thank you for raising this issue. It sounds like we'll need to move it back to where it was. @jpbland1 will you revert the zephyr changes in PR 6043? 

In the future we can put other RTOS projects in there, but for Zephyr we'll have to keep it out.

Thanks,
David Garske, wolfSSL

jpbland1, 2023-03-03:
Ah I see, that folder isn't copied into a zephyr build system like in the case of nuttx, wolfssl itself is so that needs to be in the conforming folder

donusry, 2023-03-03:
Should I open another issue for the sample applications failing to build?

jpbland1, 2023-03-03:
we've merged the fix, please pull it and try one more time and if the sample application still fails to  build then yes please open an issue for it, thanks for bringing this to our attention

closed 2023-03-03 by jpbland1

dgarske, 2023-04-03:
Fixed in https://github.com/wolfSSL/wolfssl/pull/6251