wolfSSL/wolfssl #6173: [Bug]: BIO_get_mem_data used in functions to get general BIO data

bgermann, 2023-03-08:
### Version

5.5.4

### Description

Some public functions that take a WOLFSSL_BIO input parameter operate on it with wolfSSL_BIO_get_mem_data, which is only valid when the WOLFSSL_BIO is memory-based BIO. I have seen it in at least wolfSSL_d2i_PKCS8_PKEY_bio and wolfSSL_PKCS7_verify. They should also be able to read from other BIO methods.

anhu, 2023-03-08:
Hi @bgermann ,

Can you be a bit more specific please?  What other BIO methods where you hoping to see supported? 

Warm regards, Anthony

bgermann, 2023-03-08:
e.g., file based.

anhu, 2023-03-08:
Hi , 

Thank you for your prompt response. I believe this would be a new feature request.  As such, could you please send a description of what you would like to see supported to facts@wolfssl.com along with your location?  The wolfSSL business director for your location will then promptly get in touch with you.  

Warm regards, Anthony

closed 2023-03-08 by anhu

bgermann, 2023-03-08:
The openssl compatibility layer builds on these functions. As they should be compatible with OpenSSL and they are not, it is actually a bug.

anhu, 2023-03-08:
Thank you for sharing your perspective.  

We might have a difference of opinion in this case.  For our OpenSSL compatibility layer, we consider missing APIs to be feature requests.  Please get in touch with us via facts@wolfssl.com to register your desire for those APIs. 

Warm regards, Anthony

bgermann, 2023-03-08:
It is also not a missing API but incompatible one but sure, nobody can expect you to work for free.