wolfSSL/wolfssl #6205: [Bug]: Invalid CSR generated by ESP32s3

PaulMartinsen, 2023-03-18:
### Version

Jim/ED25519_SHA2_fix

### Description

Library creates an invalid certificate signing request on the ESP32s3 when hardware acceleration is enabled. A valid signing request is generated with hardware acceleration disabled, or the code running on a Windows target. Previously I've seen the same problem on the ESP32 and even with hardware acceleration disabled. Have tested on a recent master branch and also [ED25519_SHA2_fix](https://github.com/gojimmypi/wolfssl/commit/06592dc719d171d4bb7fc0493aea70beefbabb9c). I used @gojimmypi 's branch as he has a lot of fixes for the Espressif implementation of the library. 

Output from the signing command when supplied with a good CSR (ESP32s3 without h/w acceleration):
```
$ openssl.exe x509 -req -in "Input\BLS-001 - ESP32s3 - no HW acceleration - new key - signing request.csr" -days 10   -CA "CertificateAuthority-4096-PublicKey.crt" -CAkey "CertificateAuthority-4096-PrivateKey.pem" -passin pass:secret -CAcreateserial   -extfile "CertificateExtensions.txt" -extensions LumosDevice   -out Output\BLS-001-Device-Certificate.crt -text
Certificate request self-signature ok
subject=C = NZ, L = Hamilton, O = "Blue Leaf Software, Ltd", CN = Lumos, serialNumber = BLS-001
```

Output from the signing command when supplied with a bad CSR (ESP32s3 with h/w acceleration):
```
$ openssl.exe x509 -req -in "Input\BLS-001 - ESP32s3 - HW acceleration - New private key - Jim ED25519_SHA2_fix -signing request.csr" -days 10   -CA "CertificateAuthority-4096-PublicKey.crt" -CAkey "CertificateAuthority-4096-PrivateKey.pem" -passin pass:secret -CAcreateserial   -extfile "CertificateExtensions.txt" -extensions LumosDevice   -out Output\BLS-001-Device-Certificate.crt -text
Certificate request self-signature did not match the contents
545B0000:error:0200008A:rsa routines:RSA_padding_check_PKCS1_type_1:invalid padding:crypto\rsa\rsa_pk1.c:75:
545B0000:error:02000072:rsa routines:rsa_ossl_public_decrypt:padding check failed:crypto\rsa\rsa_ossl.c:599:
545B0000:error:1C880004:Provider routines:rsa_verify:RSA lib:providers\implementations\signature\rsa_sig.c:774:
545B0000:error:06880006:asn1 encoding routines:ASN1_item_verify_ctx:EVP lib:crypto\asn1\a_verify.c:217:
```

If I take the private key that's generated on the device I can create a CSR as below (in Windows), but that produces the same error message as above when I try to sign the CSR as above suggesting the problem might be with the private key. 
`openssl req -new -key "Input\BLS-001 - ESP32s3 - HW acceleration - New private key - Jim ED25519_SHA2_fix.pem" -nodes -out Output\BLS-001-req.csr`

Code used to generate a private key and signing request on the device:
```
bool CCredentialStore::GenerateCertificate(const char* pchSerial, const char* pAddress)
{
  // Generate a new private device key. Using a RSA key because the ESP32 has hardware
  // acceleration for that. 
  // https://www.wolfssl.com/documentation/manuals/wolfssl/chapter07.html#rsa-key-generation
  // https://www.wolfssl.com/documentation/manuals/wolfssl/group__Random.html#function-wc_initrng
  // https://www.wolfssl.com/documentation/manuals/wolfssl/group__RSA.html#function-wc_initrsakey
  // https://www.wolfssl.com/documentation/manuals/wolfssl/group__RSA.html#function-wc_makersakey
  
  // todo: check if this is using the Espressif RNG. If it is, make sure the radio
  // is on. The radio is used as an entropy source to make sure that true random
  // numbers are generated. If it isn't using the Espressif RNG, maybe it should 
  // be since that should generate True Random Numbers. 
  WC_RNG RandomNumberGenerator; 
  int nResult = wc_InitRng(&RandomNumberGenerator);
  if (0 != nResult)
  {
    ReportError("InitRng", nResult);
    return false;
  }

  unique_ptr<RsaKey> pNewKey(new RsaKey());
  nResult = wc_InitRsaKey(pNewKey.get(), nullptr);
  if (0 != nResult)
  {
    ReportError("InitRsaKey", nResult);
    return false;
  }

  nResult = wc_MakeRsaKey(pNewKey.get(), PrivateKeySize, 65537, &RandomNumberGenerator);
  if (0 != nResult)
  {
    ReportError("MakeRsaKey", nResult);
    return false;
  }
  
  const size_t szPrivateKeyBuffer = 2048;
  auto puPrivateKeyDer = make_unique<uint8_t[]>(szPrivateKeyBuffer);
  nResult = wc_RsaKeyToDer(pNewKey.get(), puPrivateKeyDer.get(), szPrivateKeyBuffer);
  if (nResult > 0)
  {
    WriteCertificate(puPrivateKeyDer.get(), nResult, PRIVATEKEY_TYPE);
  }
  else
  {
    cerr << "Error: wc_RsaKeyToDer\n";
  }

  // Generate a certificate signing request. 
  // https://www.wolfssl.com/documentation/manuals/wolfssl/chapter07.html#certificate-signing-request-csr-generation
  // https://www.wolfssl.com/documentation/manuals/wolfssl/asn__public_8h.html#function-wc_makecertreq
  // https://www.wolfssl.com/documentation/manuals/wolfssl/asn__public_8h.html#function-wc_signcert
  unique_ptr<Cert> pRequest(new Cert());
  wc_InitCert(pRequest.get());
  pRequest->daysValid = 30;

#if IS_WINDOWS
  strncpy_s(pRequest->subject.country, "NZ", sizeof(pRequest->subject.country));
  strncpy_s(pRequest->subject.org, "Blue Leaf Software, Ltd", sizeof(pRequest->subject.org));
  strncpy_s(pRequest->subject.locality, "Hamilton", sizeof(pRequest->subject.locality));
  strncpy_s(pRequest->subject.commonName, "Lumos", sizeof(pRequest->subject.commonName));
#else
  strncpy(pRequest->subject.country, "NZ", sizeof(pRequest->subject.country));
  strncpy(pRequest->subject.org, "Blue Leaf Software, Ltd", sizeof(pRequest->subject.org));
  strncpy(pRequest->subject.locality, "Hamilton", sizeof(pRequest->subject.locality));
  strncpy(pRequest->subject.commonName, "Lumos", sizeof(pRequest->subject.commonName));
#endif
  CFixedStringPrint SerialDev(pRequest->subject.serialDev, sizeof(pRequest->subject.serialDev));
  SerialDev << pchSerial;

  const size_t szCertificateRequestBuffer = 4096;
  auto puCertificateRequest = make_unique<uint8_t[]>(szCertificateRequestBuffer);
  int nRequestSize = wc_MakeCertReq(pRequest.get(), puCertificateRequest.get(), szCertificateRequestBuffer, pNewKey.get(), nullptr);
  if (nRequestSize <= 0)
  {
    ReportError("MakeCertReq", nRequestSize);
    return false;
  }

  cout << "SignCert\n";
  pRequest->sigType = CTC_SHA256wRSA;
  int nCertificateSize = wc_SignCert(pRequest->bodySz, pRequest->sigType, puCertificateRequest.get(), szCertificateRequestBuffer, pNewKey.get(), nullptr, &RandomNumberGenerator);
  if (nCertificateSize <= 0)
  {
    ReportError("SignCert", nCertificateSize);
    return false;
  }

  WriteCertificate(puCertificateRequest.get(), nCertificateSize, CERTREQ_TYPE);
  
  cout << "\ndone.\n";

  return true;
}

void CCredentialStore::WriteCertificate(const uint8_t *pDERCertificate, size_t szCertificate, int nType)
{
  const size_t szTextRequestBuffer = 4096;
  auto puTextRequest = make_unique<uint8_t[]>(szTextRequestBuffer);
  memset(puTextRequest.get(), 0, szTextRequestBuffer);
  int nPemSize = wc_DerToPem(pDERCertificate, szCertificate, puTextRequest.get(), szTextRequestBuffer, nType);
  if (nPemSize <= 0 || nPemSize == szTextRequestBuffer)
  {
    ReportError("DerToPem", nPemSize);
    return;
  }

  // make sure null terminated. 
  puTextRequest.get()[nPemSize - 1] = 0;

  cout << reinterpret_cast<char*>(puTextRequest.get()) << endl << endl;
}

```

Also attached,

dgarske, 2023-03-18:
@Anhu will you review this or assign to another engineer? Jim is traveling next week.

anhu, 2023-03-20:
Hi @PaulMartinsen , 

Thank you very much for all the details.  I have read your report very carefully and reproduced what you have seen on windows on my linux machine.  Her are my steps (for the benefit of future readers): 

```
#create root cert
$ openssl req -x509 -newkey rsa:2048 -keyout key.pem -out self_signed_req.pem -nodes
<enter dummy data>
# try to create the cert
$ openssl x509 -req -in BLS-001\ -\ ESP32s3\ -\ HW\ acceleration\ -\ New\ private\ key\ -\ Jim\ ED25519_SHA2_fix\ -signing\ request.csr -CA self_signed_req.pem  -CAkey key.pem -CAcreateserial -out certificate.pem -text 
```

Interestingly, when I use openssl to generate a CSR with the same private key, the same thing happens: 
<enter dummy data>

```
openssl req -new -key "BLS-001 - ESP32s3 - HW acceleration - New private key - Jim ED25519_SHA2_fix.pem" -nodes -out req.csr
<enter dummy data>
openssl x509 -req -in req.csr -CA self_signed_req.pem  -CAkey key.pem -CAcreateserial -out certificate.pem -text

```
This would make me agree with you that there is a problem with the private key.  However, I find it strange that the problem would manifest itself in the usage of the public key that is based on the private key and not usage of the private key itself.  

I will first do my best to better understand what kind of problem OpenSSL is encountering and report my findings here. However, please note that I do not have the ESP hardware. As such, once I report my findings, I will need to assign to another developer who has the relevant hardware. 

Warm regards, Anthony

PaulMartinsen, 2023-03-20:
Hi @anhu , thanks very much for your analysis. A great idea to use the private key to sign a CSR locally. 

My understanding is the ESP hardware accelerates low level operations (e.g., Z = X * Y (mod M) ; Z = X^Y mod M), SHA and AES operations. I don't know enough about how the private key's are generated to know where the padding is being done though. I do have TLSv1.3 connections working on the ESP32s3 using the hardware acceleration and @gojimmypi has fixed some issues in SHA (that's the branch I mentioned above). 

Is there a function in the library to load a PEM private key? I could upload a private key onto the device then do the CSR on that to confirm the private key is the issue if that were useful. I couldn't find a library function for that though. Another possibility would be to use the Wolf library to make a CSR using the bad private key. Would we expect the problem to be detected by `wc_MakeCertReq` or is that not possible? 

Have a great day
Paul.

anhu, 2023-03-20:
Hi @PaulMartinsen , 
The API is `wc_RsaPrivateKeyDecode()` . It takes as input a buffer containing a DER encoded key.  If your key is PEM encoded, you will also need `wc_KeyPemToDer()`.  There are examples of how to use these here: https://github.com/wolfSSL/wolfssl-examples/tree/master/certgen

PaulMartinsen, 2023-03-21:
Thanks @anhu . 

I modified the code sample from above to load the private key I'd previously made with OpenSSL and loaded onto the ESP32s3 instead of calling `wc_MakeRsaKey`, copied below. Then the rest of the code created the certificate signing request, with the following output:

```
$ openssl.exe x509 -req -in "Input\Sign existing certificate - ESP32s3.csr" -days 10   -CA "CertificateAuthority-4096-PublicKey.crt" -CAkey "CertificateAuthority-4096-PrivateKey.pem" -passin pass:secret -CAcreateserial   -extfile "CertificateExtensions.txt" -extensions LumosDevice   -out Output\test-Device-Certificate.crt -text
Certificate request self-signature ok
subject=C = NZ, L = Hamilton, O = "Blue Leaf Software, Ltd", CN = Lumos, favouriteDrink = BLS-001
```

So that doesn't have the self-signature miss-match I see when using `wc_MakeRsaKey` to create a new private key, which again suggests the issue lies with the private key itself, somehow. One weird behaviour is that `favouriteDrink` instead of `serialNumber` appears in the subject distinguished name. Though the value (`BLS-001`) is what I loaded for `subject.serialDev`. That could just be  a distraction though. 

```
#if 1
  ifstream file("/content/Prvt+PubKy.crt");
  file.seekg(0, ios::beg);
  ostringstream Buffer;
  Buffer << file.rdbuf();
  string strContent = Buffer.str();
  
  unsigned char abyBuffer[1250];
  int nWritten = wc_KeyPemToDer((const unsigned char*)(strContent.c_str()), strContent.length(), abyBuffer, sizeof(abyBuffer), nullptr);
  if (nWritten < 0)
  {
    ReportError("wc_KeyPemToDer", nWritten);
    return false; 
  }
  
  word32 nStartOfKey = 0; 
  nResult = wc_RsaPrivateKeyDecode(abyBuffer, &nStartOfKey, pNewKey.get(), nWritten);
  if (0 != nResult)
  {
    ReportError("wc_RsaPrivateKeyDecode", nResult);
    return false;
  }
#else
  nResult = wc_MakeRsaKey(pNewKey.get(), PrivateKeySize, 65537, &RandomNumberGenerator);
  if (0 != nResult)
  {
    ReportError("MakeRsaKey", nResult);
    return false;
  }
#endif
```

anhu, 2023-03-21:
Hi @PaulMartinsen , 

Wow.  Thank you so much for your prompt reaction time and hard work!! It seems quite clear to me now that when the private key is generated, it has the public components in it.  Then, when the CSR is generated, the public components are just being copied into the CSR.  Once the public key is then actually used to verify the CSR, the padding problem is discovered.

As you said in your original post, this only happens when ESP32s3 acceleration is enabled.  Please forgive my ignorance, but can you please let me know which flag/macro turns on ESP32s3 acceleration? 

Warm regards, Anthony

anhu, 2023-03-21:
@PaulMartinsen , 

Is this an urgent issue for you? Is temporarily disabling hardware acceleration a viable **temporary** solution?  

Warm regards, Anthony

anhu, 2023-03-21:
Hi @PaulMartinsen , 

The next sensible step would be to turn off hardware acceleration and see the differences in the csr that are generated.  Can you try that?  I do not have the correct hardware so I can't try that. 

Note that @gojimmypi will be helping here to setup an environment to reproduce what you are seeing.  However, due to other activities he will have limited time to work on this. I'm assigning this to him, but will continue to monitor and help out on this issue.

anhu, 2023-03-21:
Oh, the unaccelerated version is in the zip file. Let me look into that.

PaulMartinsen, 2023-03-21:
Hi @anhu 

This isn't a blocking problem for me at the moment as I can load externally generated keys onto the device. But I'm keenly interestedin, and happy to help with, its resolution :). 

I add the following pre-processor symbols to `user_settings.h` to turn off hardware acceleration:
```
#if 1
// Undocumented. Disable hardware acceleration of cryptography functions including RSA
#define NO_ESP32WROOM32_CRYPT

// Undocumented. Disable hardware acceleration of RSA primitives?
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI 

// Undocumented. Disable hardware acceleration of SHA (e.g., SHA-256, SHA-512)
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_HASH

// Undocumented. Disable hardware acceleration of AES
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_AES

#endif
```

`BLS-001 - ESP32s3 - no HW acceleration - new key - signing request.csr` in [Keys.and.certificates.zip](https://github.com/wolfSSL/wolfssl/files/11007188/Keys.and.certificates.zip) was generated with hardware acceleration off. 

BTW, I just tried `openssl rsa -in "BLS-001 - ESP32s3 - HW acceleration - New private key - Jim ED25519_SHA2_fix.pem" -text -pubout` to see if any errors but it printed private key and public key things without reporting any errors. 

I'm not quite sure what you mean by comparing the CSR with and without hardware acceleration. I'm generating a separate private key each time. Since you saw the same problem with OpenSSL, I'm not sure if reusing the hardware acceleration generated private key in a non-hardware accelerated CSR will tell us anything new? Did I missunderstand? 

Thanks for your help with this. Much appreciated. 

BTW, I'm in New Zealand. We are about 12 hours behind GMT, so sometimes there might be delays in my responses :)

anhu, 2023-03-21:
Hi @PaulMartinsen , 

> I'm not sure if reusing the hardware acceleration generated private key in a non-hardware accelerated CSR will tell us anything new?

So I wanted to confirm that the format of the padding was conforming to this: 
```
    /*
     * The format is
     * 00 || 01 || PS || 00 || D
     * PS - padding string, at least 8 bytes of FF
     * D  - data.
     */
```
And it is!!  I saw a 0, about 200 FF bytes, and a 0 and then some other bytes after that. 

I need to talk to my colleagues.  Please stay tuned. 

Warm regards, Anthony
...and indeed in openssl's `RSA_padding_check_PKCS1_type_1

anhu, 2023-03-21:
Sorry, I've made a rookie mistake.  I thought this was a problem with padding, but actually this indicates that there are problems somewhere else in the cryptographic procedure.  Basically, the verify operation is failing. To get a better idea of the root cause we would need to override the RNG with a  fake RNG that gives deterministic data (ie a buffer of incremented byte values) and see what the difference is between hardware acceleration and no hardware acceleration. 

Warm regards, Anthony

PaulMartinsen, 2023-03-22:
FWIW, I generated a 512 bit key and CSR on the ESP32s3 (previous keys have been 2048 bits). But I still see the signature miss-match error.

gojimmypi, 2023-03-22:
For those following along, I looked at the [docs](https://www.wolfssl.com/documentation/manuals/wolfssl/chapter07.html#rsa-key-generation) and the [csr example](https://github.com/wolfSSL/wolfssl-examples/blob/master/certgen/csr_example.c) and created a [VisualGDB example & README](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/VisualGDB) for a [sample ESP32 app](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/main/station_example_main.c) that creates a RSA Key cert:

```
    RsaKey genKey;
    RNG    rng;

    wc_InitRng(&rng);
    wc_InitRsaKey(&genKey, 0);

    ret = wc_MakeRsaKey(&genKey, 1024, 65537, &rng);
    if (ret != 0) {
        ESP_LOGI(TAG, "wc_MakeRsaKey error %d", ret);
    }

    byte der[4096];
    int  derSz = wc_RsaKeyToDer(&genKey, der, sizeof(der));
    if (derSz < 0) {
        ESP_LOGI(TAG, "wc_RsaKeyToDer error %d", ret);
    }

    byte pem[4096];
    int  pemSz = wc_DerToPem(der, derSz, pem, sizeof(pem),
                      PRIVATEKEY_TYPE);
    if (pemSz < 0) {
        ESP_LOGI(TAG, "wc_DerToPem error %d", ret);
        /* pemSz contains error */
        ;
    }

    ESP_LOGI(TAG, "%s", pem);

```

I manually copied the console output and saved to a file, for example `./input/gojimmypi_private_rsa_generated_key_hw.txt` for example:

```
-----BEGIN RSA PRIVATE KEY-----
MIICXAIBAAKBgQDh1MnHdI/gGE7qG+PXGZ2U8pkW27T0QXEtBupI7//b+sOl2Q0B
bh/IUqI61Ys5MiDi9jr01r0nxOiMqnAfzx+DhEpu8PFfbElokg3nVLXcuKLBQRdz
kcyl/DbSeatvl0behZY6fMmUqpklWqZ74Zr2gQbiFJIyIu9QDLcdKPY1gQIDAQAB
AoGATdTjki13JLncAM2J8elvKRWPc5RXQlOHqQgYGPk2Sl+brH4pAFQu+gCYzwQo
DpdbRD2uxhF4cctohop7SEs/RZOveoDJyzK7+vIsHkvtSK2idQcQtnzyINDPM61s
8yaniDVJjySek19bBk3bMsOAyidmcdMAdYuwY7NzE5LtgAECQQD4iLEo6rxguW4P
OKXotWGk8Hfq/XL0nwgzGOku2NEHCtqLo5ZEYa6Yq52y/492OCppHhbqxSMSXK5x
Ee29uesBAkEA6J2BKhKVQdDnL1itwzuUmhfJLWpmMVcIh4y6qypiC2pZrKckIiBh
rcyY6BGL24pbS7XG5Ytzo/uf5fy3ktjKgQJBANvhK0N08XZtd0a2KmpIVnvzV1Sq
KJLtQ/bY+iw9HCkr3BAgiWg0N6DQbDOZz9WaeUMIoOPRx2aOYrKcVwY8vwECQD8X
EQRHPAI41asqwx4zl69fiinuNL9nGqIhEjQAb80m0CGax5sV60hHOKWgtK0bzn0v
VT3G39t3ELxBGc0TTYECQH4tQToR7UMrI1YVjE0mDDBiZOSLL4keOKfZ85JPqnF0
Q/zCMEmj8dls1tagi2a8ALuwvUPByVCALWMZ3JUYdck=
-----END RSA PRIVATE KEY-----
```

Adjusting the default [user_settings.h](https://github.com/gojimmypi/wolfssl/blob/86751b279cd70dd86dc7f0b8f4545e2a128f4f4d/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/components/wolfssl/include/user_settings.h#L138) both enabling and disabling this section for either no hardware or use hardware acceleration:

```
#define NO_ESP32WROOM32_CRYPT
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_HASH
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_AES
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI
```

In each case, with a hardware-generated cert or not, both files could be successfully signed. So it _appears_ this is not a hardware acceleration issue, rather a possible incompatible settings. 

Thanks @PaulMartinsen for the tips on using OpenSSL: 

### Create request:
```bash
SRC=gojimmypi_private_rsa_generated_key_hw.txt
CSR_OUT=$SRC-CertificateRequest.csr
DEVICE_ID=4243
openssl req -new -key ./input/$SRC \
   -nodes \
   -out ./output/$CSR_OUT \
   -subj "/C=NZ/ST=Waikato/L=Hamilton/O=Blue Leaf Software Ltd/CN=Lumos-%DeviceId%/serialNumber=$DEVICE_ID" \
   -addext "subjectAltName = URI:dev:ops:59430-Lumos-$DEVICE_ID" \
   -addext "extendedKeyUsage = serverAuth, clientAuth"
```

### Sign the request:
```
openssl x509 -req -in "./output/$CSR_OUT" -days 10   \
    -CA "private/CertificateAuthority-4096-PublicKey.crt" \
    -CAkey "private/CertificateAuthority-4096-PrivateKey.pem" \
    -passin pass:secret -CAcreateserial  \
    -extfile "private/CertificateExtensions.txt" -extensions LumosDevice  \
     -out ./output/gojimmypi-Device-Certificate_hw.crt -text
```

### Settings

A side note: I'm not sure about the impact of having a [relative include path](https://github.com/gojimmypi/wolfssl/blob/ba3f6d0f0a386f68402e1264e7d1ce216c00e8f8/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/components/wolfssl/include/user_settings.h#L10) to a separate header file from the wolfSSL `user_settings.h`. Intuitively I think it would be best to keep all the settings in one `user_settings.h` file. I suppose this is a trivial pursuit question of the compiler and linker would always work this out for all wolfSSL files referencing the `user_settings` (itself also from a relative directory in a `CMakeLists.txt` Espressif component). In theory it should work, but I recommend keeping everything in a single `user_serttings.h`.

In the coming days I'll wrap up this example and save it over on the [wolfssl-examples/ESP32](https://github.com/wolfSSL/wolfssl-examples/tree/master/ESP32). I'll also add some code to make a cert request from the ESP32. 

I'll take a closer look at the original settings to see if there was some sort of conflict that could be detected and flagged with a warning.

PaulMartinsen, 2023-03-22:
That's a useful [test case](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/VisualGDB) @gojimmypi . A couple of suggestions:
* It could be simplified by dropping the `-extfile` and `-extensions` arguments ( I'm adding custom policies to the certificates when signing the CSR, which I don't believe is common, has been very complicated and not relevant here). 
* Please use the number `32473`, the registered [private enterprise number for documentation/examples](https://datatracker.ietf.org/doc/html/rfc5612), instead of `59430` in the `subjAltName` content. 

It looks like key size plays a role in this problem. Consider two processes:
1. Generate a private key on ESP32s3 &rarr; create CSR with OpenSSL &rarr; sign CSR with OpenSSL
2. Generate a private key on ESP32s3 &rarr; create CSR on ESP32s3 &rarr; sign CSR with OpenSSL

OpenSSL here refers to `OpenSSL 3.0.7 1 Nov 2022 (Library: OpenSSL 3.0.7 1 Nov 2022)` running on Windows. The result for each process using typical key sizes is summarized below. The keys posted in the original issue were all 2048-bit. 
| Key size [bits] | Process 1 | Process 2|
| --------------- | ---------- | ---------- | 
| 512 | ok | signature mismatch |
| 1024 | ok | ok |
| 2048 | signature mismatch | signature mismatch |

Unfortunately, relative paths are a necessary side effect of multiple targets, build systems and finite resources for me. But I agree, simpler is better!

PaulMartinsen, 2023-03-22:
Follow up observation for Process 2 with 1024 bit keys. OpenSSL reports the subject in the signed CSR as:
 `subject=C = NZ, L = Hamilton, O = "Blue Leaf Software, Ltd", CN = Lumos, favouriteDrink = BLS-001`. 
That last field should be `serialNumber = BLS-001`. Set on the ESP32s3 using: 
`strncpy(pRequest->subject.serialDev, "BLS-001", sizeof(pRequest->subject.serialDev));`. 

I noticed `ASN_FAVOURITE_DRINK` and `ASN_SERIAL_NUMBER` have the same value in `asn.h` (0x05), though I'm not sure if that's a related or different problem.

gojimmypi, 2023-03-23:
@PaulMartinsen  I've been able to duplicate your issue on the ESP32-S3, even with fresh code and what I believe to be a good `user_settings/h` file.

Shown below is a table of RSA key bit length sizes, along with whether the keys were generated with HW or SW.  This latest version of code is using [wolfSSL to create](https://github.com/gojimmypi/wolfssl/blob/724035504bd8dedb2e187280bba447119187345d/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/main/station_example_main.c#L285) and [sign](https://github.com/gojimmypi/wolfssl/blob/724035504bd8dedb2e187280bba447119187345d/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/main/station_example_main.c#L311) the [generated CSR](https://github.com/gojimmypi/wolfssl/blob/724035504bd8dedb2e187280bba447119187345d/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/main/station_example_main.c#L358).

The links in this table are to `cert_peek` output files in my [./output](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output) directory.

| Key size [bits] |  SW | HW |
| --------------- | ---------- | ---------- | 
| 512 | [ok](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output/gojimmypi-ESP32-S3_SW_512_cert_peek.txt) | [signature mismatch](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output/gojimmypi-ESP32-S3_HW_512_cert_peek.txt) |
| 1024 | [ok](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output/gojimmypi-ESP32-S3_SW_1024_cert_peek.txt) | [ok](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output/gojimmypi-ESP32-S3_HW_1024_cert_peek.txt) |
| 2048 | [ok](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output/gojimmypi-ESP32-S3_SW_2048_cert_peek.txt) | [signature mismatch](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output/gojimmypi-ESP32-S3_HW_2048_cert_peek.txt) |

The `cert_peek.sh` [script](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/cert_peek.sh) looks like this:

```
#!/bin/bash
if [ "$1" == "HW" ] || [ "$1" == "SW" ]; then
    echo HW/SW: $1
else
    echo "Not a valid HW/SW value: $1"
    exit 1
fi

if [ "$2" == "512" ] || [ "$2" == "1024" ] || [ "$2" == "2048" ]; then
    echo RSA size: $2
else
    echo "No a valid RSA size 512/1024/2048 value: $2"
    exit 1
fi

echo ""
echo "Files:"
echo ""

ls ./output/gojimmypi-ESP32-S3_"$1"_"$2"* -al
echo ""
openssl rsa -in "./output/gojimmypi-ESP32-S3_"$1"_"$2"_private_key.txt" -noout -text

echo ""
echo "Request:"
echo ""

openssl req -in "./output/gojimmypi-ESP32-S3_"$1"_"$2"_request.crt" -noout -text

echo ""
echo "CA sign:"
echo ""

openssl x509 -req -in "./output/gojimmypi-ESP32-S3_"$1"_"$2"_request.crt" \
        -days 10                                                          \
        -CA "private/CertificateAuthority-4096-PublicKey.crt"             \
        -CAkey "private/CertificateAuthority-4096-PrivateKey.pem"         \
        -passin pass:secret -CAcreateserial                               \
        -extfile "private/CertificateExtensions.txt"                      \
        -extensions LumosDevice                                           \
        -out ./output/gojimmypi-Device-Certificate_"$1"_"$2".crt -text
```

Note the naming convention is `[username]-[Espressif chipset]_[HW or SW]_[RSA bit length]_[file content]` for files that `cert_peek.sh` will process.

The [code](https://github.com/gojimmypi/wolfssl/blob/724035504bd8dedb2e187280bba447119187345d/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/main/station_example_main.c#L191) has a `#define KEY_SIZE` for values of `512`, `1024` or `2048`.  HW/SW is still defined in the `user_settings.h` by leaving or commenting out the [crypto defines](https://github.com/gojimmypi/wolfssl/blob/724035504bd8dedb2e187280bba447119187345d/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/components/wolfssl/include/user_settings.h#L136):

```
/* Default is HW enabled unless turned off.
** Uncomment these lines for SW: */
/*
#define NO_ESP32WROOM32_CRYPT
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_HASH
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_AES
#define NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI
*/
```

Now to figure out _why_ only HW 512 and 2048 bit lengths have OpenSSL complain about signature mismatches, with otherwise no errors or indications of and sort of problems.

gojimmypi, 2023-03-24:
I have an [interim solution](https://github.com/gojimmypi/wolfssl/commit/4d29f68d8db6943cd9dfa39675c7333a9a5a74ed) in place for this on my [ED25519_SHA2_fix](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix) branch. The current wolfSSL master branch will need to completely turn off one of the hardware encryption features: `NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI` if this problem is encountered.

I've confirmed this is problematic on Xtensa architectures: ESP32 and ESP32-S3 with wolfSSL hardware encryption turned on..

As it turns out, the cause of the certificate signature failure is in the [wolfcrypt/tfc.c](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/tfm.c) for certain RSA bit lengths when Espressif Hardware Acceleration is turned on. There are 2 different problems.

The first is when doing RSA 512 bit: the [fp_exptmod( )](https://github.com/wolfSSL/wolfssl/blob/e0e590f126bd7d947ce3dfbea5ea1405bb646374/wolfcrypt/src/tfm.c#L2816) returns an incorrect result when calling the HW `esp_mp_exptmod()`.

The other one is for RSA 1024 bit: the [mp_mulmod( )](https://github.com/wolfSSL/wolfssl/blob/e0e590f126bd7d947ce3dfbea5ea1405bb646374/wolfcrypt/src/tfm.c#L4310) returns a incorrect result. Note this one is related to the `#define ESP_RSA_MULM_BITS 2000` setting in the [default Espressif user_settings.h](https://github.com/wolfSSL/wolfssl/blob/e0e590f126bd7d947ce3dfbea5ea1405bb646374/IDE/Espressif/ESP-IDF/user_settings.h#L96) file. The HW `esp_mp_mulmod()` is only called when `fp_count_bits(a)` and `fp_count_bits (b)` are both greater than `ESP_RSA_MULM_BITS` (2000), otherwise it falls back to software calculation. I'll need to so some more homework on this one.

Curiously, the [wolfssl_test](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test) passes all of the tests. Clearly there's a missing test that I'll need to add. As this is not a SHA test, I've not yet decided if I create a new issue, or just rename https://github.com/wolfSSL/wolfssl/issues/5989.

For now, I've added compile-time macro gating that detects and falls back to software calculations for both of the problematic Espressif HW calcs. This requires that the RSA bit length be defined in the `user_settings.h`:

```
#define WOLFSSL_RSA_KEY_SIZE 1024
```

Currently if this is not defined, and the TFM `USE_FAST_MATH` is enabled, there will be a compile-time warning. 

As a side note: For experimenting with various settings that get assigned with `./configure`, see https://github.com/wolfSSL/wolfssl/pull/5876. I've added this tool to my WIP branch. It helped me discover that the commandline options for certs turn on these:

```
#undef  WOLFSSL_KEY_GEN
#define WOLFSSL_KEY_GEN

#undef  WOLFSSL_CERT_REQ
#define WOLFSSL_CERT_REQ

#undef  WOLFSSL_CERT_GEN
#define WOLFSSL_CERT_GEN

#undef  WOLFSSL_CERT_EXT
#define WOLFSSL_CERT_EXT
```
but these are apparently enabled by default:

```
#undef  WOLFSSL_SYS_CA_CERTS
#define WOLFSSL_SYS_CA_CERTS

#undef  WOLFSSL_ASN_TEMPLATE
#define WOLFSSL_ASN_TEMPLATE
```

Note that I've also added `#define WOLFSSL_SYS_CA_CERTS` and `#define WOLFSSL_ASN_TEMPLATE` to the `user_settings.h`. See the [user_settings.h](https://github.com/gojimmypi/wolfssl/blob/fa37eaa0ee6bf816d9c0506a4aff22df5889ff42/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/components/wolfssl/include/user_settings.h#L167) file used during testing.

As noted above, this is tested with the the [cert_peek.sh script](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/cert_peek.sh) that has since been updated to also accept a 3rd parameter for target chip set. (ESP32 or ESP32-C3 or ESP32-S3)

To test with existing files, there are 6 new files with a `_fixed` suffix in my [./output directory](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_server_6205/output).

I'll need more time to dig into `esp_mp_exptmod()` and determine the actual root cause.

gojimmypi, 2023-04-25:
Hi @PaulMartinsen a brief update: I'm still working on a better solution than just disabling the known problematic key lengths as noted in [#6286](https://github.com/wolfSSL/wolfssl/pull/6286#issuecomment-1513410833).

I believe I have found the root cause. As the problem is not reproducible, even on a fixed binary when running the same code after fresh reboot, more testing is in order before I can confirm.

PaulMartinsen, 2023-04-27:
Thanks for persisting with this @gojimmypi . I've been using 1024-bit keys for now, which are okay for testing, but this is going to be another important bug-fix for the library! Appreciate your hard work on this.

gojimmypi, 2023-04-30:
Hi @PaulMartinsen - thank you! Indeed I've learned a lot during this exercise! At this point, I believe I have found the root cause of this issue. I have an initial fix with on my [ED25519_SHA2_fix branch](https://github.com/gojimmypi/wolfssl/blob/42e38b249c2e633202d48c62b5e7467fadfbce10/wolfcrypt/src/port/Espressif/esp32_mp.c#L505).

The problem seems to be a simple [sign issue as noted in #6359](https://github.com/wolfSSL/wolfssl/issues/6359#issuecomment-1529053181), specifically related to the `WOLFSSL_SP_INT_NEGATIVE` setting. (used in `esp32_mp.c`, not in `tfm.c`). This certainly seems like a reasonable explanation as to why the problem was not reproducible. 

I need to clean up my code and perform more testing, but I wanted you to have this information as soon as possible. It's curious that this problem does not manifest itself in other key sizes, such as the 1024-bit keys that you are using. I do need to revisit the operand pointer issue originally noted in https://github.com/wolfSSL/wolfssl/issues/6359 and convince myself there's no problem there.

gojimmypi, 2023-04-30:
In addition to the sign issue in [esp_mp_mul](https://github.com/wolfSSL/wolfssl/blob/52a20fc2b54a90c4fe3cce868af2d98e86377e09/wolfcrypt/src/port/Espressif/esp32_mp.c#L331) described above and shown here:

[image]

It appears there's another problem in [esp_mp_exptmod](https://github.com/wolfSSL/wolfssl/blob/52a20fc2b54a90c4fe3cce868af2d98e86377e09/wolfcrypt/src/port/Espressif/esp32_mp.c#L744) where although the answer seems the same, the `Y.used` property is different as shown here:

[image]

When this condition is encountered, the CSR signing process fails.

PaulMartinsen, 2023-05-07:
That's very interesting @gojimmypi . Mind boggling that the 1024-bit keys seem to work okay (I've made a few 1024-bit ones since we first found this issue, still with no problems). So I guess wrong math can give the right answer sometimes!

Thanks again for your persistence with this, rather thorny, problem. Much appreciated!

gojimmypi, 2023-05-25:
Hi @PaulMartinsen just an update on this topic: I've certainly not forgotten this issue even though there's be no recent update here. In fact, it continues to have my undivided attention. I've found a variety of peripheral issues and improvements that I'm working on. (e.g. see [ESP32 Errata](https://www.espressif.com/sites/default/files/documentation/esp32_errata_en.pdf)) 

The end result should be quite nice. I'm also adding an entire testing infrastructure that will greatly ease the development and validation of the new ESP32-C3/C6 RISC-V Architecture, as soon as I can wrap up the ESP32/S3 Xtensa. This new feature has certainly be quite helpful to confirm proper operation using the [wolfcrypt tests](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test).

Thanks very much for your patience. Cheers

PaulMartinsen, 2023-09-20:
Fantastic to see this one resolved. Thanks to all and particularly @gojimmypi for all your hard work on it. Much appreciated!