wolfSSL/wolfssl #6218: wolfSSL_dtls_export does not export TX/RX NONCE?

nunojpg, 2023-03-22:
### Version

master

### Description

I'm trying dtls_export with a local build with -enable-dtls13 --enable-sessionexport=nopeer.

For a DTLSv1.3 TLS_AES_128_GCM_SHA256 when calling wolfSSL_dtls_export I get 334 bytes.

The export **does not** change as more datagrams are sent or received.

I wonder what is the impact of NONCE window/counters as this session is restored from this dump and communication continued.

julek-wolfssl, 2023-03-22:
Hi @nunojpg,

unfortunately exporting DTLS 1.3 sessions is not yet supported. If you would like to submit a feature request, please write to us at support@wolfssl.com.

Sincerely
Juliusz

nunojpg, 2023-03-22:
Yes, I see it works for DTLS 1.2 great. I have edited the subject to reflect the actual issue now that you have explained. 

Thanks

closed 2023-03-22 by nunojpg