wolfSSL/wolfssl #6258: TLS 1.3 not usable for me

joeftiger, 2023-04-03:
### Version

5.6.0 (#979707380c677dfa65e3ba48f19e149773a4a32d)

### Description

I am trying to use TLS 1.3 for [my small playground server](https://github.com/joeftiger/wolfssl-playground).
CLion (IDE) keeps reporting `Use of undeclared identifier 'wolfTLSv1_3_client_method'` and trying to compile and run the server results in a `SIGSEGV` crash immediately.

I've tried compiling wolfSSL with both `make`:
```sh
./autogen.sh
./configure --enable-tls13
make
make check
sudo make install
```
and `cmake`:
```sh
mkdir build
cd build
cmake .. -DWOLFSSL_TLS13=yes
cmake --build .
sudo cmake --install .
```
but TLS 1.3 is just not available for me to use. Is there something else I must do to enable it? I went through lots of official docs now, but am unable to fix it.

embhorn, 2023-04-03:
Hi @joeftiger 

Thanks for contacting wolfSSL Support. You'll need to include the configuration header before any other wolfSSL includes:

```
#ifndef WOLFSSL_USER_SETTINGS
    #include <wolfssl/options.h>
#endif
#include <wolfssl/wolfcrypt/settings.h>

#include <wolfssl/ssl.h>
```

Let us know if that resolves the issue.

Thanks,
@embhorn - wolfSSL Support

embhorn, 2023-04-04:
Hi @joeftiger 

Excellent! If you haven't already, I recommend checking out our examples repository:
https://github.com/wolfSSL/wolfssl-examples

I'll go ahead and mark this issue as resolved.

Cheers,
Eric

closed 2023-04-04 by embhorn