wolfSSL/wolfssl #6360: [Bug]: wolfssl_server.ino doesn't even compile for ESP32 boards

BojanJurca, 2023-05-01:
### Contact Details

bojan.jurca@gmail.com

### Version

5.5.4

### Description

https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/user_settings.h

### Reproduction steps

1. install wolfssl do Arduino
2. cnage user_settings.h
3. compile

I managed to get rid of some compile-time errors, like #error and missing #include files, but than many others occur ...

### Relevant log output

```shell
Just for beginning:

In file included from C:\Users\uporabnik\Documents\Arduino\libraries\wolfssl\src/wolfssl/wolfcrypt/settings.h:13,
                 from C:\Users\uporabnik\Documents\Arduino\libraries\wolfssl\src/wolfssl/ssl.h:33,
                 from C:\Users\uporabnik\Documents\WolfSSL\WolfSSL.ino:24:
C:\Users\uporabnik\Documents\Arduino\libraries\wolfssl\src/wolfssl/wolfcrypt/wolf_crypt_settings.h:802:14: fatal error: FreeRTOS.h: No such file or directory
     #include "FreeRTOS.h"
              ^~~~~~~~~~~~
compilation terminated.

exit status 1
```

gojimmypi, 2023-05-01:
Hello @BojanJurca and thank you for taking a look at wolfSSL for the Arduino and pointing out this problem.

I've been able to reproduce the error you are seeing. In my case, the first thing I did was try the [default Espressif user_settings.h](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/user_settings.h). This is *not* the file to use for Arduino. 

I followed the instructions on the [wolfSSL with Arduino](https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO) page. Although mentioned in the README there, I could not find a suitable `user_settings.h` in the [/examples/configs/](https://github.com/wolfSSL/wolfssl/tree/master/examples/configs) directory. I ended up using the [user_settings.h from #2380](https://github.com/wolfSSL/wolfssl/issues/2380#issuecomment-519268912)

The setup instructions for Arduino appear to a be a bit dated. I'll be working on tiding that up. For now, please be aware that in order to get things working, I needed to manually copy some source files from `wolfssl` into a newly created `wolfssl/src` directory:

[image]

Using the Arduino environment for the ESP32, although possible, presents itself with a variety of challenges. Have you considered using the [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/)? See the [Espressif examples](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF). Note that some of the examples no longer require a wolfSSL install process. For example, the [wolfssl_benchmark](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_benchmark) is ready to go immediately upon cloning. I still need to update the TLS client and server examples.

You'll notice that much of the work I do uses the [VisualGDB](https://visualgdb.com/) extension for Visual Studio - basically a nice IDE on top of the Espressif ESP-IDF. This is a quite productive environment for me, but purely optional. I also use the Tigard JTAG for single-step debugging. (see [my notes](https://gojimmypi.github.io/Tigard-JTAG-SingleStep-Debugging-ESP32/)). Of course, there are many other environments and editors, such as VSCode, and more.

If you definitely do not want to use the Espressif ESP-IDF, then perhaps you'd be interested in the [Visual Micro Arduino Extension for Visual Studio](https://www.visualmicro.com/).

In any case, the Arduino examples definitely need to be fixed up a bit. Thanks again for bringing this to our attention.

gojimmypi, 2024-02-16:
Hello @BojanJurca !

Good news! There's  an Official Arduino wolfSSL coming soon to the integrated Arduino Library Manager!!

The first release from the new  [Arduino-wolfSSL repository](https://github.com/wolfSSL/Arduino-wolfSSL) can be found here:

https://github.com/wolfSSL/Arduino-wolfSSL/releases/tag/5.6.6-Arduino.1

I believe this release fixes all the problems encountered in the 3rd party library that had been published during the absence of any other wolfSSL Arduino library.

Details of the new library publishing issues can be found in https://github.com/wolfSSL/Arduino-wolfSSL/pull/1. TL;DR: The problems were extra files that cannot currently be excluded from the Arduino build process.

Since this is a `change` and not an `add` to the Arduino repository directory, human-review is needed. See https://github.com/arduino/library-registry/pull/4056 and the related https://github.com/arduino/library-registry/pull/4059. Once those are merged I expect the proper wolfSSL library to be available in the IDE.

Moving forward, we'll have full support and more boards tested. See also https://github.com/wolfSSL/wolfssl/pull/7177 specifically related to the ESP32 [Arduino Examples](https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO/sketches).

I hope you'll give this new library a try and let me know how it goes. 

Cheers

BojanJurca, 2024-03-15:
Thank you very much.

I tried the Arduino wolfssl_server example with my ESP32 board. It compiles now with some minor modifications (file is attached), but it doesn't really work. My ESP32 often crashes, even if there is no client trying to connect (please see console output) or when curl -k https://10.18.1.111/ command is executed.

[ino.zip](https://github.com/wolfSSL/wolfssl/files/14620539/ino.zip)

Console output:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

wolfSSL TLS Server Example Startup.
show_memory() not implemented for this platform
Waiting for time to be set...
Waiting for time to be set...
Waiting for time to be set...
Waiting for time to be set...
Waiting for time to be set...
Connecting to **********. Connected!
********************************************************
      wolfSSL Example Server IP = 10.18.1.111
********************************************************
Setup network complete.
No WOLFSSL_USER_SETTINGS_ID found.
wolfSSL Debugging is Off! (enable with DEBUG_WOLFSSL)
WARNING: Unknown or no TLS session cache setting.
Successfully called wolfSSL_Init
Here we go!
Initializing certificates...
show_memory() not implemented for this platform
Initializing certificates...
Success: use certificate: server_cert_der_2048
Success: use private key buffer: server_key_der_2048
Completed Arduino setup()
Begin Server... (waiting for remote client to connect)
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3f4085e4  PS      : 0x00060830  A0      : 0x8014630c  A1      : 0x3ffd56a0  
A2      : 0xffffffff  A3      : 0x3ffdce7c  A4      : 0x00000009  A5      : 0x00000002  
A6      : 0x40107ad0  A7      : 0x00000008  A8      : 0x8014aefc  A9      : 0x3f4085e4  
A10     : 0x3ffdce64  A11     : 0x3ffdce7c  A12     : 0x00000009  A13     : 0x00000002  
A14     : 0x000000aa  A15     : 0x000000fc  SAR     : 0x00000010  EXCCAUSE: 0x00000002  
EXCVADDR: 0x3f4085e4  LBEG    : 0x400896c4  LEND    : 0x400896cf  LCOUNT  : 0x00000000  

Backtrace: 0x3f4085e1:0x3ffd56a0 |<-CORRUPTED

ELF file SHA256: 1b076292deed92a0

Rebooting...

```

gojimmypi, 2024-03-21:
Hello @BojanJurca and thank you so much for taking the wolfSSL on the Arduino for a test  drive!

> It compiles now with some minor modifications (file is attached),

Ah yes, let's address each item separately:

- ## my_private_config.h

[image]

Yes, the required-but-missing `my_private_config.h` is perhaps not the best idea. Do you have any suggestions? I'm thinking instead of the having the `MY_PRIVATE_CONFIG` enabled, to instead have default-named WiFi SSID and passwords that at least successfully compiles. At startup time, I'd give a conspicuous warning if not changed:

```
*******************************************************************************
******************************** ALERT ****************************************
*******************************************************************************
Unless your WiFi is actually called "MySSID" with a password of "MyPassword", 
please enter your own WiFi name and password. See [docs].
```

What do you think; Any recommendations?

- ## Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

I've seen this when the _exact_ model number of the ESP32 was not selected in the Arduino IDE and appropriate configuration parameters not set properly. For instance, the difference between the `DOIT ESP32 DEVKIT V1` and the Espressif `ESP32 Dev Module`:

[image]

Although Arduino does a wonderful job on their own Arduino boards, needing to manually know and assign parameters on these other boards has been a bit more challenging.

The `InstructionFetchError` is also caused by a stack overflow at runtime. See the latest [Arduino default user_settings.h](https://github.com/wolfSSL/wolfssl/blob/master/examples/configs/user_settings_arduino.h) which is copied to the Arduino library in [src/user_settings.h](https://github.com/wolfSSL/Arduino-wolfSSL/blob/main/src/user_settings.h). 

I've been able to get both the client and server examples to work in as little as 32KB.

Were you able to eventually resolve the error and get the code to run?

- ## Default TLS ports

As for the default ports: `11111` vs `443`:

[image]

I believe that _should_ work with cURL, although I've only tested with wolfSSL clients & servers. I'll be taking a closer look at this in the coming days to confirm it works properly.  Certainly the `Guru Meditation Error: Core  0` might explain that not working.

- ## Defaults for WiFi SSID and Password for config without `my_private_config.h`
 
The SSID and password is simply an oversight on my part & will be corrected in the next version. I appreciate you pointing this one out:

[image]

- ## Latest code

Please see https://github.com/wolfSSL/Arduino-wolfSSL/pull/6 for the latest version of wolfSSL for Arduino.

Thanks again for trying this out. I'd like to make sure this is all working for you. Let me know how I can help.

gojimmypi, 2024-06-21:
I believe the Arduino examples are considerable more robust at this point.

Please open a new issue if any additional problems are encountered.

See also:

https://www.wolfssl.com/getting-started-with-wolfssl-on-arduino/

https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO

https://github.com/wolfSSL/arduino-wolfSSL

https://www.arduino.cc/reference/en/libraries/wolfssl/

closed 2024-06-21 by gojimmypi