wolfSSL/wolfssl #6365: [Bug]: No connection binding with the connection-id-extension

Conradowatz, 2023-05-02:
### Contact Details

conrad@owatz.de

### Version

5.5.0

### Description

WolfSSL partially supports the connection identifier extension (https://www.rfc-editor.org/rfc/rfc9146.html).
However, the extension is currently not usable as intended, as a a connection endpoint cannot be recognized purely by connection id. Currently wolfSSL identifies connecting peers purely by their adress and port.
The goal of the extension would be to enable an endpoint to change its ip adress and or port whithout the need of a new handshake.

### Reproduction steps

In user_settings.h, enable DTLS and the extension:
```
#define WOLFSSL_DTLS
#define WOLFSSL_DTLS_CID
```

1. Connect to the wolfSSL example server using DTLS and with connection id enabled (`--cid XX`) and do a handshake
2. On the client: reopen your udp connection to change source port
3. try sending messages to the server again
4. the wolfSSL server will report a unknown peer and discard the messages

### Relevant log output

```shell
wolfSSL Entering EmbedReceiveFrom()
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
    Ignored packet from invalid peer
wolfSSL error occurred, error = -323
```

rizlik, 2023-05-08:
Hey @Conradowatz,

Thanks for reaching out.
Yes, you are right, Connection ID support is partial, you can find more detail in the PR that added the support https://github.com/wolfSSL/wolfssl/pull/5453 . 

Complete support is in the feature request stage at this point. 
The current status may still be useful if the other endpoint is the one that wants to differentiate using the ConnectionID.

Thanks,
Marco

trainman419, 2024-10-23:
I'm interested in this support as well. Has there been any progress on this?

julek-wolfssl, 2025-01-15:
Hi @trainman419 @Conradowatz 
we recently added API to better support CIDs. Please see the API added in https://github.com/wolfSSL/wolfssl/pull/8224. For example to parse the CID you can use [wolfSSL_dtls_cid_parse](https://www.wolfssl.com/documentation/manuals/wolfssl/ssl_8h.html#function-wolfssl_dtls_cid_parse). See the example in https://github.com/wolfSSL/wolfssl-examples/pull/472 how  to use these new API.
Juliusz

closed 2025-01-15 by julek-wolfssl