wolfSSL/wolfssl #6384: Wolfssl accepts digital certificates with incorrect expiry date format

llmlla, 2023-05-05:
When I used wolfssl for digital certificate validation, I found that wolfssl accepted a digital certificate with the wrong expiry date format.
The version I used is wolfssl 4.2.0.
Here are  certificates I used.
[leaf.txt](https://github.com/wolfSSL/wolfssl/files/11404907/leaf.txt)
[root.txt](https://github.com/wolfSSL/wolfssl/files/11404908/root.txt)

kareem-wolfssl, 2023-05-05:
Hi @llmlla ,

4.2.0 is extremely outdated and no longer supported, please upgrade to 5.6.0, our latest release, and retry.
Your root certs appear to have valid date formats, we do support both UTC and Generalized time.  With their expiry dates set far into the future (9999/6566), I do expect them to validate.  Your leaf cert appears to have an invalid notBefore date, and I expect it to fail.

Thanks,
Kareem

llmlla, 2023-05-07:
Thanks for the suggestion, I will try the latest version.

closed 2025-12-18 by kareem-wolfssl