wolfSSL/wolfssl #6386: Native STM32WL Support

maxgerhardt, 2023-05-08:
### Version

5.6.0

### Description

Currently, as per

https://github.com/wolfSSL/wolfssl/blob/62a4329f8eadff2b2b3459c4677d9a80f4a7f150/wolfssl/wolfcrypt/settings.h#L1361-L1366

WolfSSL supports a lot of STM32 series microcontrollers, but not STM32WL natively, per this.

When using STM32CubeMX to generate a project for the STM32WL(55JCI7, for example), and the wolfSSL.ICUBE component is activate, one just gets the error

```cpp
../../wolfSSL/wolfSSL.I-CUBE-wolfSSL_conf.h:200:6: warning: #warning Please define a hardware platform! [-Wcpp]
..
../../Middlewares/Third_Party/wolfSSL_wolfSSL_wolfSSL/wolfssl/wolfssl/wolfcrypt/settings.h:1367:22: fatal error: stm32f4xx_hal.h: No such file or directory
```
since it seems to default to STM32F4xx then.

maxgerhardt, 2023-05-08:
FYI, I've been playing around with this for a few hours now, and it seems it can be added fairly easily. Major added configuration options in the `wolfSSL.I-CUBE-wolfSSL_conf.h` are just
```cpp
#elif defined(STM32WL55xx)
	#define HAL_CONSOLE_UART huart2
	#define STM32_HAL_V2
	#define WOLFSSL_STM32WL
    #define WOLFSSL_STM32_PKA
	#undef  NO_STM32_CRYPTO
```
(just before "Please define a hardware platform..") and in `settings.h` adding some `|| defined(WOLFSSL_STM32WL)` in the above shown block, then ofc
```cpp
    #ifdef WOLFSSL_STM32_CUBEMX
		#elif defined(WOLFSSL_STM32WL)
			#include "stm32wlxx_hal.h"
...
    #else
		#elif defined(WOLFSSL_STM32WL)
			#include "stm32wlxx.h"
        #endif
```
with a minor adjustment in `stm32.c` with respect to the used crypto instance
```cpp
  hcryp->Instance = /*CRYP*/ AES;
```
and that gives me something that compiles and I think reasonable uses the STM32WL's hardware with regards to the Public Key Accelerator (PKA), AES accelerator and Hardware RNG. Still testing of course.

lealem47, 2023-05-08:
Hi @maxgerhardt,

Thanks for the suggested improvements to our STM32 support. 

I'll track this issue so that the next STM32Cube Pack release contains support for at least the STM32WL55. Let me know how your testing goes.

Thanks.

dgarske, 2023-11-01:
Hi @maxgerhardt ,

I've pushed fixes to support the STM32WLXX to https://github.com/wolfSSL/wolfssl/pull/6937

Thanks,
David Garske, wolfSSL

PR #6937 (2023-11-01): Support for the STM32WL55 and PKA improvements for ECC parameters
# Description

* Support for the STM32WL55
* PKA improvements for ECC parameters to support custom curves
* Fix for `WOLF_CONF_MATH` options `6=Single Precision C all small` and `7=Single Precision C all big`.

Fixes #6386 and Fixes #6396

# Testing

Tested on STM32WL55 and STM32U585.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
commit 33e12e35 (2023-11-01): Support for the STM32WL55 and PKA improvements for ECC parameters. Fixes #6386 and Fixes #6396.
closed 2023-11-01 by lealem47