wolfSSL/wolfssl #6397: [Bug]: Testbench doesn't respect NO_ECC_SECP

maxgerhardt, 2023-05-11:
### Contact Details

maximilian.gerhardt@rub.de

### Version

870f7cc95b1061b0f829d15315c66b6b6823eb99

### Description

WolfSSL and NIST-P curves seem to glued together inseparably. I really don't want the NIST P-x / SECP curve in my application. WolfSSL has the `NO_ECC_SECP` macro which is supposed to disable these curves.

However, doing so makes the selftests check **still** attempt to do tests on the NIST curves -- subsequently failing on it, although WolfSSL would be prefectly operateble on the chosen curves.

The testbench should  `NO_ECC_SECP`.

### Reproduction steps

1. `./configure --enable-brainpool --enable-ecccustcurves --disable-curve25519 --disable-ed25519  --enable-crypttests-libs --enable-crypttests`
2. `make`
3. `./wolfcrypt/test/testwolfcrypt`
4. `EXTRA_CFLAGS="-DNO_ECC_SECP" ./configure --enable-brainpool --enable-ecccustcurves --disable-curve25519 --disable-ed25519  --enable-crypttests-libs --enable-crypttests`
5. `make`
6. `./wolfcrypt/test/testwolfcrypt`

### Relevant log output

```shell
With first config:

ECC      test passed!
```
with second config:

```
ecc_test_vector 28 failed!: -140
ECC      test failed!
 error L=140
 [fiducial line numbers: 7580 23060 33425 45580]
Exiting main with return code: -1
```
```

dgarske, 2023-05-11:
Hi @maxgerhardt ,

Yeah you have hit a use-case we haven't tested. We typically always have the standard NIST prime curves enabled. I understand wanting just the brain pool curves enabled. I think at one time this worked and getting it working should be easy. I'll have another engineer review effort.

Thanks,
David Garske, wolfSSL

lealem47, 2023-05-12:
Hi @maxgerhardt,

Thank you for this report. This issue should be resolved with https://github.com/wolfSSL/wolfssl/pull/6402

lealem47, 2023-05-15:
@maxgerhardt Could you test with https://github.com/wolfSSL/wolfssl/pull/6411?

lealem47, 2023-07-05:
Hi @maxgerhardt,

Since we never heard back I will go ahead and close out this issue. If anything else come up please do not hesitate to open a new issue anytime. We are always happy to help out in any way we can.

Until next time all the best and thank you for reaching out.

closed 2023-07-05 by lealem47