wolfSSL/wolfssl #6408: Compatibility issues with other TLS libraries

SmallTown123, 2023-05-13:
### Version

5.5.1 and below

### Description

I think there are two compatibility issues in the TLS 1.2 implementation:

(1) When processing the **SessionID** field in the ClientHello message, the wolfssl server side seems to respond normally only to **SessionID** of length **32** at present.

However, RFC5246 specifies the SessionID length as `opaque SessionID<0..32>`.

There may be compatibility issues with other TLS library clients.

(2) It seems that wolfssl server has **support upper limit 150 for Cipher Suites** in ClientHello.

In general, this upper number is sufficient, but is there a potential compatibility issue with this implementation? 

For example, a client with an individual TLS implementation library will send all the cipher suites it supports, and the number happens to be greater than 150.

dgarske, 2023-05-15:
@julek-wolfssl can you look into this report?

SmallTown123, 2023-05-15:
Ok, I understand. Thank you for your reply! @julek-wolfssl

dgarske, 2023-06-28:
Hi @SmallTown123 ,

Have you had a chance to try the fixes in PR #6415 to resolve the shorter session ID issue? You feedback would be appreciated.

Thanks,
David Garske, wolfSSL

closed 2023-06-28 by SparkiDev

julek-wolfssl, 2023-06-29:
Hi @SmallTown123,

the PR fixing this has been merged.

Juliusz