wolfSSL/wolfssl #6409: [Bug]: STM32 RTC integration code is uncompilable

maxgerhardt, 2023-05-15:
### Contact Details

maximilian.gerhardt@rub.de

### Version

5.6.0-stable

### Description

If a STM32 project is activating the RTC module (which defines `HAL_RTC_MODULE_ENABLED`), the following code gets auto-activated so that `XTIME()` maps to a function returning the time

https://github.com/wolfSSL/wolfssl/blob/c17cff72654ec46b1be6d3c03869bd4bd91f240a/wolfssl/wolfcrypt/wc_port.h#L965-L969

however, the expected return value is **not** `time_t*`, as can be seen by the usage of that macro in `wc_Time()`

https://github.com/wolfSSL/wolfssl/blob/c17cff72654ec46b1be6d3c03869bd4bd91f240a/wolfcrypt/src/asn.c#L13991-L13997

the return value of XTIME() should be the same as that of `wc_Time`, aka, `time_t`.

### Reproduction steps

1. Create STM32 project through STM32CubeMX with activated RTC support
2. Add WolfSSL
3. Make sure `NO_ASN_TIME` is **not** defined
4. Try to compile

### Relevant log output

```shell
lib/wolfssl-5.6.0-stable/wolfcrypt/src/wc_port.c:3024:8: error: conflicting types for 'stm32_hal_time'
 3024 | time_t stm32_hal_time(time_t *t1)
      |        ^~~~~~~~~~~~~~
In file included from lib/wolfssl-5.6.0-stable/wolfssl/wolfcrypt/types.h:35,
                 from lib/wolfssl-5.6.0-stable/wolfcrypt/src/wc_port.c:28:
lib/wolfssl-5.6.0-stable/wolfssl/wolfcrypt/wc_port.h:949:27: note: previous declaration of 'stm32_hal_time' was here
  949 |     WOLFSSL_LOCAL time_t* stm32_hal_time(time_t* t1);
      |                           ^~~~~~~~~~~~~~
```

lealem47, 2023-05-15:
Hi @maxgerhardt,

The second commit in this PR https://github.com/wolfSSL/wolfssl/pull/6411 should address this bug.

maxgerhardt, 2023-05-17:
https://github.com/wolfSSL/wolfssl/pull/6411/commits/1c3fa3882189697b0fcfab0e5f83191fdd6ff4b0 looks good, I applied the same bugfix and it made it compile.

closed 2023-05-17 by maxgerhardt