wolfSSL/wolfssl #6419: The problem of compiling a project using wolfssl

murz1ch, 2023-05-17:
wolfssl.lib(ssl.obj) : error LNK2001: unresolved external wc_PRF character.
wolfssl.lib(tls.obj) : error LNK2001: unresolved external character wc_PRF_TLS.
wolfssl.lib(tls13.obj) : error LNK2001: unresolved external character wc_Tls13_HKDF_Extract.
wolfssl.lib(tls13.obj) : error LNK2001: unresolved external character wc_Tls13_HKDF_Expand_Label.
I dont know what is it, please help

PS. I'm using VS2022

embhorn, 2023-05-17:
Hello @murz1ch 

Thanks for contacting wolfSSL Support. Typically these issues are a result of not including the configuration header (or the correct header) in the application. Please be sure that the application includes `wolfssl/wolfcrypt/settings.h`

```
#include <wolfssl/wolfcrypt/settings.h> /* before any other wolfSSL headers */
#include <wolfssl/ssl.h>
```

Thanks,
@embhorn - wolfSSL Support

embhorn, 2023-05-31:
Hi @murz1ch 

I hope this helped to resolve the issue you were seeing. I'll go ahead and mark this as resolved, but please feel free to reopen if you are still having issues.

Thanks,
@embhorn

closed 2023-05-31 by embhorn