wolfSSL/wolfssl #6469: [Bug]: build failure since May 17th (error in wolfssl/openssl/aes.h)

wtarreau, 2023-06-03:
### Version

master

### Description

I used to see 5.6.0 build fine, but running configure on latest master (37aada03) resulted in errors about wolfssl/openssl/aes.h.

I ran a `git bisect` which stopped on this commit:
```
checking for wolfssl/openssl/aes.h... no
configure: error: Error including wolfssl/openssl/aes.h. Possible circular dependency introduced or missing include.
dd9edfee24fd9f51eb1977d210199e1e56868651 is the first bad commit
commit dd9edfee24fd9f51eb1977d210199e1e56868651
Author: Juliusz Sosinowicz <juliusz@wolfssl.com>
Date:   Wed May 17 16:22:55 2023 +0200

    Implement atomic operations interface
```

I don't see the relation quite well, to be honest, except, maybe if some of the changed #ifdef had a side effect impact.

The simplest I tried that gives me this error (on latest master) is:
```
$ ./configure --enable-haproxy --enable-quic --prefix=/opt/wolfssl-5.6.0/
```
And it ends like this:
```
config.status: executing libtool commands
config.status: executing wolfssl/wolfcrypt/async.h commands
config.status: executing wolfssl/wolfcrypt/fips.h commands
config.status: executing wolfssl/wolfcrypt/port/cavium/cavium_nitrox.h commands
config.status: executing wolfssl/wolfcrypt/port/intel/quickassist.h commands
config.status: executing wolfssl/wolfcrypt/port/intel/quickassist_mem.h commands
configure: ---
configure: Running make clean...
configure: error: Header file inconsistency detected -- error including wolfssl/openssl/aes.h.
make: *** [config.status] Error 1
configure: ---
configure: Generating user options header...
checking for wolfssl/openssl/aes.h... cat: confdefs.h: No such file or directory
no
configure: error: Error including wolfssl/openssl/aes.h. Possible circular dependency introduced or missing include.
```
It initially failed with the extra options `--enable-sp, --enable-intelasm, --enable-rwlock`  which I dropped above.

Given how noticeable it should be, I suspect it can have something to do with the build environment. It's a CentOS 7 x86_64 running on a dual-core Atom C2350. I had to build latest automake (1.16.5), autoconf (2.71) and libtool (2.4.7) otherwise wolfssl fails to build saying the ones from this distro are too old. gcc is 4.8.5, and make is 3.82.

I'm a bit confused by seeing configure run make and am wondering if this couldn't be related to some operations ordering there (though I have no idea what).

### Reproduction steps

./configure --enable-haproxy --enable-quic --prefix=/opt/wolfssl-5.6.0/

### Relevant log output

```shell
config.status: executing libtool commands
config.status: executing wolfssl/wolfcrypt/async.h commands
config.status: executing wolfssl/wolfcrypt/fips.h commands
config.status: executing wolfssl/wolfcrypt/port/cavium/cavium_nitrox.h commands
config.status: executing wolfssl/wolfcrypt/port/intel/quickassist.h commands
config.status: executing wolfssl/wolfcrypt/port/intel/quickassist_mem.h commands
configure: ---
configure: Running make clean...
configure: error: Header file inconsistency detected -- error including wolfssl/openssl/aes.h.
make: *** [config.status] Error 1
configure: ---
configure: Generating user options header...
checking for wolfssl/openssl/aes.h... cat: confdefs.h: No such file or directory
no
configure: error: Error including wolfssl/openssl/aes.h. Possible circular dependency introduced or missing include.
```
```

kareem-wolfssl, 2023-06-05:
Hi @wtarreau ,

I can't reproduce, I see no issues with --enable-haproxy --enable-quic on the latest master.
Please try adding `--disable-makeclean` in your configure line, we have seen issues with our forced make clean in the past with older toolchains.

Given the commit you've bisected to, it's possible your toolchain is too old to support this atomic type, and this is indirectly leading to the issue you're seeing.  Please try disabling our atomic support by adding `CFLAGS='-DWOLFSSL_NO_ATOMICS'` to your ./configure line and let me know if it helps.

wlallemand, 2023-11-03:
@wtarreau Did you `autoreconf -i` after pulling? I had some issues multiple times when forgetting to do it.

Not sure this issue is still relevant, but just in case it is.

wtarreau, 2023-11-03:
I don't remember about this one, and I've since built it successfully quite a few times, so either there was indeed something related to autoreconf as you suggest, or it was since fixed. In both cases there's no point in leaving this issue open, I'm closing it. Thanks!

closed 2023-11-03 by wtarreau