wolfSSL/wolfssl #6474: wolfssl-config does not support --cflags-only-I, uses old-style pkg-config script

mcr, 2023-06-05:
### Version

wolfSSL Release 5.6.0 (from master)

### Description

the installed wolfssl-config script does not support the --cflags-only-I, and some other options that pkg-config provides.
The script is generated by configure.ac, and it's unclear how to update it.
It probably should install a wolfssl.pc file instead.

kareem-wolfssl, 2023-06-05:
Hi @mcr ,

As you've already found, we do install a wolfssl.pc file in your prefix under pkgconfig, by default this is in /usr/local/lib/pkgconfig/wolfssl.pc.
Did you have any further questions I can assist with?

mcr, 2023-06-07:
It looks like the wolfssl.pc does not include any -DOPENSSL_EXTRA, etc. to reflect the options that were configured, so the application doesn't know it should include them.   I have hacked around this while I figure out how to update the wolfssl.pc that is created.
But, it may also be that declared openssl version is never at least 1.1.1, so many other libraries will fail, as they need at least 1.1.x to run these days.

mcr, 2023-06-07:
Oh, and one needs to have: 

   -I${includedir}/wolfssl

in the CFLAGS in order for includes like openssl/opensslv.h to work.

mcr, 2023-06-07:
I'm building an application (ruby-openssl) that expects to include only openssl headers.
So ssl.h really needs to do the above.   Otherwise, it's not really much of an openssl emulation.

kareem-wolfssl, 2023-06-08:
You can use the flag `EXTERNAL_OPTS_OPENVPN` to have us include options.h in our settings.h (which is included by ssl.h).  We don't generally recommend or support this flag, it is only meant for cases where you are unable to modify the source code.
You will need to define this flag while building ruby-openssl, and not while building wolfSSL itself.

mcr, 2023-06-30:
https://github.com/mcr/ruby-openssl/tree/ruby-wolfssl-hacked is my very hacked up copy of ruby-openssl.  
(Unfortunately, I have patches that I need in that branch, which I can't get upstreamed, which is why it's so behind master)
I use defined(WOLFSSL_TYPES_DEFINED) all over to remove code that depends upon features that wolfssl does not seem to have.  I'd rather test for those features explicitely, but I was in a rush.

I use the following wolfssl.pc, inserted into /usr/lib/x86_64-linux-gnu/pkgconfig/openssl.pc (on a test build VM, of course):

````
prefix=/sandel/3rd/wolfssl
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: wolfssl
Description: wolfssl C library.
Version: 5.3.0
Libs: -L${libdir} -lwolfssl
Cflags: -I${includedir} -I${includedir}/wolfssl -DHAVE_EX_DATA -DOPENSSL_EXTRA -DOPENSSL_ALL -DOPENSSL_NO_NEXTPROTONEG -fPIC -DWC_NO_HARDEN -Werror=implicit-function-declaration -Wno-discarded-qualifiers
````

I think that there is a bug in X509_get_default_XXX, as it returns NULL, and I think it should never do that.
https://github.com/mcr/ruby-openssl/commit/79cff7987451f65a6ce65fed266dde195574bbd7

My reason for using wolfssl is that there are patches for it to operated against the WiseKey VaultIC.
In the end, the effort was a failure, as I wound up having to remove so much code that I really couldn't do much, *AND* it turns out the the VaultIC patches are rather not-standalone, and require more support in main() than I was ready to hack in.
(The goal being to run unchanged, or only very mildly changed ruby code)

I think that the .pc file installed ought to have all the right things in based upon the settings from ./configure.
I would submit a patch to do that, but I didn't figure out how the wolfssl.pc file was even created yet.

kareem-wolfssl, 2025-12-25:
Hi @mcr, 

My sincere apologies for my delayed response on this.

X509_get_default_XXX are currently returning NULL because they are stub functions.  I'll work on a patch to have them return an empty string.

The .pc file is generated from the wolfssl.pc.in file in support/: https://github.com/wolfSSL/wolfssl/blob/master/support/wolfssl.pc.in
The build settings/CFLAGS go in options.h (when using configure) or user_settings.h (when using user_settings.h) and need to be included in the application rather than the .pc file.