wolfSSL/wolfssl #6479: [Bug]: Missing headers when trying to integrate static library from Xcode project

wolf81, 2023-06-06:
### Contact Details

info+wolfssl@wolftrail.net

### Version

5.6.0

### Description

Trying to integrate the build of the Xcode project gives some issues. It seems several files are missing from the `include` directory in the Xcode project, namely:

- `wolfssl/wolfcrypt/sp_int.h`
- `wolfssl/openssl/**`

The missing files can be copied over from the root project, afterwards the app using WolfSSL will build without errors.

lealem47, 2023-07-05:
Hi @wolf81,

Since we never heard back I will go ahead and close out this issue. If anything else come up please do not hesitate to open a new issue anytime. We are always happy to help out in any way we can.

Until next time all the best and thank you for reaching out.

wolfgang-ironsoftware, 2023-07-05:
Hi @lealem47 that's ok. I haven't had the time to verify. Right now I have some other priorities, for the coming months. Will probably revisit WolfSSL integration in a couple of months.