wolfSSL/wolfssl #6497: [Bug]: SSIZE_T defined check fix (compile time error for header consumers)

mitchcapper, 2023-06-13:
### Contact Details

see gh profile

### Version

master

### Description

https://github.com/wolfSSL/wolfssl/blob/9ffa9faecda87a7c0ce7521c83996c65d4e86943/wolfssl/wolfcrypt/settings.h#L1880-L1883

uses the HAVE_SSIZE_T gating but the configure script doesn't actually check for it.  Of course it doesn't matter at compile time for the library itself, but when consumers include the headers it will throw an error if it is defined by something.

It was added in commit: https://github.com/wolfSSL/wolfssl/commit/fe3099b08ed23146b9098e3504b725ce63bafc13 a few months ago.

### Reproduction steps

have ssize_t defined in a common include file and then 
`#include <wolfssl/wolfcrypt/settings.h>`
will cause a problem

### Relevant log output

```shell
will depend on how ssize_t is defined but:
`wolfssl\x64-windows-static-release\include\wolfssl\wolfcrypt\settings.h(1882,25): error C2628: 'SSIZE_T' followed by 'long' is illegal (did you forget a ';'?)`
```

embhorn, 2023-06-13:
Hi @mitchcapper 

Thanks for the detailed report and the suggested fix PR. I've requested a review of this issue by the author of the offending commit.

mitchcapper, 2023-06-14:
@douzzer WFM able to fully build wget2 without my patch for this.

douzzer, 2023-06-14:
@mitchcapper awesome, thanks!

closed 2023-12-18 by douzzer