wolfSSL/wolfssl #6514: [Bug]: wolf ssl's error to string functions broken for certain core errors when OPENSSL_EXTRA is used

mitchcapper, 2023-06-15:
### Contact Details

gh

### Version

master

### Description

wolfSSL_ERR_reason_error_string (and by extension wolfSSL_ERR_error_string / wolfSSL_ERR_error_string_n)  can easily break with OPENSSL_EXTRA defined.  The first thing the function does is flip the value to negative  with that defined (lineno/context link won't work as internal.c is too massive for GH).

This is particularly sad as values we set like: WOLFSSL_ERROR_WANT_READ  are positive at least these are also:
```
    WOLFSSL_ERROR_WANT_READ        =  2,
    WOLFSSL_ERROR_WANT_WRITE       =  3,
    WOLFSSL_ERROR_WANT_CONNECT     =  7,
    WOLFSSL_ERROR_WANT_ACCEPT      =  8,
    WOLFSSL_ERROR_SYSCALL          =  5,
    WOLFSSL_ERROR_WANT_X509_LOOKUP = 83,
    WOLFSSL_ERROR_ZERO_RETURN      =  6,
    WOLFSSL_ERROR_SSL              = 85,
```
As this is a primary error flow mechanic per:
https://www.wolfssl.com/doxygen/group__Debug.html
calling [wolfSSL_get_error](https://www.wolfssl.com/doxygen/group__Debug.html#gaafd5671d443fa684913ba5955a4eb591)
`After wolfSSL_get_error is called and returns the unique error code, wolfSSL_ERR_error_string may be called to get a human-readable error string`

Ironically if you actually have the original error code like WANT_READ (-323) and pass that directly to `wolfSSL_ERR_error_string` it would work.

### Reproduction steps

Looking at wolfSSL_get_error  ( ssl.c also too big for gh context links)  the bottom 3 values are all positive, 
[image]
and the code that flips the positive value before switching on those error codes
[image]

embhorn, 2023-06-15:
Hi Mitch,

My colleague is going to review this issue.

Thanks,
@embhorn

douzzer, 2023-06-15:
@mitchcapper the error codes are a mess once the compatibility layer is in the picture.  In this case I _think_ the routine at issue is only used in contexts where there aren't outright namespace collisions, since the OpenSSL codes are all under 100 and the native wolfSSL codes are all >=100.

In any case, see what you think of #6517.  There's more work to be done but I think this answers your immediate concern.

PR #6517 (2023-06-21): 20230615-opensslextra-wolfSSL_ERR_reason_error_string
in `wolfSSL_ERR_reason_error_string()`, unconditionally convert positive error codes to negative, and in `switch()`, match against negated values of OpenSSL positive codes like `WOLFSSL_ERROR_WANT_READ`.  fixes #6514

tested with `wolfssl-multi-test.sh ... super-quick-check`
commit 970663f8 (2023-06-21): in wolfSSL_ERR_reason_error_string(), unconditionally convert positive error codes to negative, and in switch(), match against negated values of OpenSSL positive codes like WOLFSSL_ERROR_WANT_READ.  fixes #6514
closed 2023-06-21 by dgarske