wolfSSL/wolfssl #6524: [Bug]: arm64 for ios stack overflow

calvin2021y, 2023-06-20:
### Version

v5.6.2-stable

### Description

use same script build for 5.6.0 work fine, after upgrade to 5.6.2, get this error.

### Reproduction steps

```sh
configure --enable-shared=no --enable-harden --enable-filesystem=no --enable-pwdbased=no --enable-ip-alt-name --enable-sni --enable-alpn --enable-truncatedhmac --enable-earlydata --enable-tlsv10=no --enable-oldtls=yes --enable-tlsv12=yes --enable-tls13 --enable-rsa --enable-psk-one-id --enable-session-ticket --enable-savesession --enable-sessioncerts --enable-rng --enable-aescbc=yes --enable-aescfb=no --enable-aesccm=no --enable-aesctr=no --enable-aesctr=no --enable-maxfragment=yes --enable-blake2=no --enable-blake2s=no --enable-hkdf=no --enable-sys-ca-certs=no --enable-examples=no --enable-crypttests=no --enable-singlethreaded=no --enable-asynccrypt=no --enable-asyncthreads=no --enable-sha384 --enable-asm=yes --enable-sp=small,asm --enable-intelasm --enable-aesni --enable-bigcache --enable-curl --enable-curve25519=yes --enable-ed25519=yes --enable-crl=no --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-hrrcookie=no
```

calvin2021y, 2023-06-20:
try build with ` --enable-smallstack`, not crashed anymore.

but CURL throw this error and not able to finish ssl handshake: ` CA signer not available for verification`

anhu, 2023-06-20:
Hi Calvin, 

Thank you for letting us know what you are seeing. 

I'd like to reproduce what you are seeing but you only show a wolfSSL configure step.  For example do you do `make all` and then `sudo make install`?  Are you using the head of the master branch? 

How do you configure and build curl?  How do you execute curl.  Please let me know enough details to reproduce your step. 

Warm regards, Anthony

calvin2021y, 2023-06-20:
Yes, I use `make install`.  I am use wolfSSL for years,  and old version work fine for ios arm64.

I use a 32 * PAGE_SIZE stack with user context,  after upgrade `5.6.2`,  32 page or 128 page, none of them work without ` --enable-smallstack`.

after use `--enable-smallstack`,  I get `CA signer not available for verification`.

calvin2021y, 2023-06-20:
test `master` not work too.

anhu, 2023-06-20:
Hi @calvin2021y , 
I took a stab at reproducing.  Here are my steps: 

```
cd wolfssl
git clean -xdf
./autogen.sh
./configure --enable-shared=no --enable-harden --enable-filesystem=no --enable-pwdbased=no --enable-ip-alt-name --enable-sni --enable-alpn --enable-truncatedhmac --enable-earlydata --enable-tlsv10=no --enable-oldtls=yes --enable-tlsv12=yes --enable-tls13 --enable-rsa --enable-psk-one-id --enable-session-ticket --enable-savesession --enable-sessioncerts --enable-rng --enable-aescbc=yes --enable-aescfb=no --enable-aesccm=no --enable-aesctr=no --enable-aesctr=no --enable-maxfragment=yes --enable-blake2=no --enable-blake2s=no --enable-hkdf=no --enable-sys-ca-certs=no --enable-examples=no --enable-crypttests=no --enable-singlethreaded=no --enable-asynccrypt=no --enable-asyncthreads=no --enable-sha384 --enable-asm=yes --enable-sp=small,asm --enable-intelasm --enable-aesni --enable-bigcache --enable-curl --enable-curve25519=yes --enable-ed25519=yes --enable-crl=no --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-hrrcookie=no
make all 
sudo make install
cd ../curl
./configure --with-wolfssl
make all 
sudo make install 
sudo ldconfig
/usr/local/bin/curl  https://www.wolfssl.com --output wolfssl.txt
```
This was successful.  Now, I did no expect to see what you got because I'm on x86_64 and running ubuntu.  That said, if I was on arm64 running ios, would these steps reproduce what you are seeing? 

Warm regards, Anthony

calvin2021y, 2023-06-20:
I am build my app for  android/ios/linux/windows, I test all work even include ios X86 emulator.

just ios arm64 crashed.

anhu, 2023-06-20:
Hi @calvin2021y , 
please remove ` --enable-intelasm --enable-aesni` and try again. 

If that fails, I would like to request exact reproduction steps. 

Warm regards, Anthony

calvin2021y, 2023-06-20:
I post wrong options in op, the ios options is

```sh
--enable-shared=no --enable-harden --enable-filesystem=no --enable-pwdbased=no --enable-ip-alt-name --enable-sni --enable-alpn --enable-truncatedhmac --enable-earlydata --enable-tlsv10=no --enable-oldtls=yes --enable-tlsv12=yes --enable-tls13 --enable-rsa --enable-psk-one-id --enable-session-ticket --enable-savesession --enable-sessioncerts --enable-rng --enable-aescbc=yes --enable-aescfb=no --enable-aesccm=no --enable-aesctr=no --enable-aesctr=no --enable-maxfragment=yes --enable-blake2=no --enable-blake2s=no --enable-hkdf=no --enable-sys-ca-certs=no --enable-examples=no --enable-crypttests=no --enable-singlethreaded=no --enable-asynccrypt=no --enable-asyncthreads=no --enable-sha384 --enable-asm=no --enable-sp=small --enable-bigcache --enable-curl --enable-curve25519=yes --enable-ed25519=yes --enable-crl=no --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-hrrcookie=no --host=arm-apple-darwin
```

calvin2021y, 2023-06-20:
I use `CURLOPT_SSL_CTX_FUNCTION` with curl,  

```c

CURLcode curl_sslctx_function(CURL *curl, void *sslctx, void *parm)
{
	CURLcode rv = CURLE_ABORTED_BY_CALLBACK;
    int line = 0;
	(void)curl;
	(void)parm;
    switch(1) default : {
        int e = wolfSSL_CTX_load_verify_buffer(sslctx, (const unsigned char *) pem_ptr, pem_len, SSL_FILETYPE_PEM);
        if ( e != SSL_SUCCESS ) {
            line = 4;
    		break;
        }
        rv = CURLE_OK;
    }
	return rv;
}
```

pem_ptr and pen_len is the pem file load from `https://curl.se/ca/cacert-2023-05-30.pem`

On IOS with `--enable-smallstack`  i keep get `Problem with the SSL CA cert (path? access rights?)`,  what can cause this?

calvin2021y, 2023-06-20:
I get `Problem with the SSL CA cert (path? access rights?)` for request with domain name.

try with  https://1.1.1.1/ has no problem.

anhu, 2023-06-20:
Since I cannot reproduce this I can try taking a guess. 
Can you try changing: 
--enable-sp=small

To: 
--enable-sp 
 
(remove "=small") 

Warm regards, Anthony

anhu, 2023-06-20:
In cURL the following appears: 

```
  case CURLE_SSL_CACERT_BADFILE:
    return "Problem with the SSL CA cert (path? access rights?)";
```

There is a problem occurring while parsing your CA file. 

Instead of a crash, I think the same problem is manifesting itself while loading the CA cert.

calvin2021y, 2023-06-20:
change `--enable-sp=small`  into `--enable-sp` not work.

I just verify `wolfSSL_CTX_load_verify_buffer` is return `SSL_SUCCESS` from `CURLOPT_SSL_CTX_FUNCTION`

the same CA `https://curl.se/ca/cacert-2023-05-30.pem` is used for all platform, ios X86 emulator work fine.

please see the trace logs,  crash from `DoTls13HandShakeMsg`, `DoTls13HandShakeMsgType`, `ProcessPeerCerts`, `ProcessPeerCertParse`, `ParseCertRelative`,  `ConfirmSignature`,  `wc_RsaSSL_VerifyInline`, `RsaPrivateDecryptEx`, `wc_RsaFunction_ex`.

 `https://1.1.1.1/` work, maybe they are not use RSA in cert.

`--enable-smallstack` maybe just avoid crash but cert auth still faluire.

anhu, 2023-06-20:
https://1.1.1.1  uses ecc key for the server and intermediate certificate and the root certificate is RSA.. 

Have you tried  `--enable-sp-math-all`?

calvin2021y, 2023-06-20:
test with `--enable-smallstack`

`https://8.8.8.8` get this error, `https://1.1.1.1` work fine.

error from here:
```diff
diff --git a/lib/vtls/wolfssl.c b/lib/vtls/wolfssl.c
index 2928728..6938cb2 100644
--- a/lib/vtls/wolfssl.c
+++ b/lib/vtls/wolfssl.c
@@ -788,7 +788,7 @@ wolfssl_connect_step2(struct Curl_cfilter *cf, struct Curl_easy *data)
     else if(ASN_NO_SIGNER_E == detail) {
       if(conn_config->verifypeer) {
         failf(data, " CA signer not available for verification");
-        return CURLE_SSL_CACERT_BADFILE;
+        return CURLE_BAD_CONTENT_ENCODING;
       }
       else {
         /* Just continue with a warning if no strict certificate
```

with this patch I get `Unrecognized or bad HTTP Content or Transfer-Encoding`.

test with `--enable-sp-math-all` &&  `--enable-smallstack` get the same results.

test with `--enable-sp-math-all` and without `--enable-smallstack` get the same crash logs.

anhu, 2023-06-20:
Is the commonality that the server certificates all have RSA public keys when you get the error?

calvin2021y, 2023-06-20:
Yes, RSA & PKCS #1 SHA-384.
RSA & PKCS #1 SHA-256.

calvin2021y, 2023-06-20:
there is a crash logs with IOS arm64 without `--enable-smallstack`, this is a clear sign some bugs in the RSA Function

anhu, 2023-06-20:
Yes.  the crash happens in the RSA function, but it could be due to corruption somewhere else. 

Lets try different math implementations. 
Can replace `--enable-sp-math-all` with `--enable-fastmath` ? 
It that does not work, can you replace it with `--enable-heapmath` ? 

Also, if you have valgrind on your system, can you please try running it with that?

dgarske, 2023-06-20:
@bigbrett can you take a look at this?

calvin2021y, 2023-06-21:
> Yes. the crash happens in the RSA function, but it could be due to corruption somewhere else.
> 
> Lets try different math implementations. Can replace `--enable-sp-math-all` with `--enable-fastmath` ? It that does not work, can you replace it with `--enable-heapmath` ?

replace `--enable-sp-math-all` with `--enable-fastmath` OR `--enable-heapmath`, without `--enable-smallstack` not crashed anymore.

still get the RSA problem.

> Also, if you have valgrind on your system, can you please try running it with that?

I am build with clang sanitizer, the crash log is report by sanitizer.  (not sure how to make3 valgrind work with ios)

calvin2021y, 2023-06-21:
I switch back to openssl with CURL, work fine.  let me know what I can do to fix this problem.

calvin2021y, 2023-06-21:
confirm the ssl error exists for android arm32,  aarch64 work fine.

calvin2021y, 2023-06-21:
I find arm64 ios crashed problem first.  use `--enable-fastmath` fix it.

then I find arm64 ios can not finish RAS cert handshake,   and late I find android arm32 has the same problem.

anhu, 2023-06-21:
Hi Calvin, 

it seems all this back and forth is not very productive.  Perhaps it would be best if we got on a zoom call where you could do a screen share and give me details of your platform (hardware, OS version, etc) and the exact steps you use to build and run the application. Would you be willing to do such a call?  If so, please send a link to your favourite meeting infrastructure (zoom, MS Teams, etc.)  to me at anthony@wolfssl.com .  Please note that I'm in the Eastern Time Zone (same as New York) .  My Friday this week is almost all open so that would be the best day to meet. 

Warm

calvin2021y, 2023-06-22:
Thanks very much for the offer, I am much busy this week. I will drop you a email to check when you are available next week.

calvin2021y, 2023-06-23:
This is the wireshark report for 8.8.8.8 error on ios arm64 https://gist.github.com/calvin2021y/613e7680601c99865d3237b4638991b1

anhu, 2025-11-17:
Hi Calvin, 

I'm going to guess that similar to me, you have lost track of this issue. As such, as a matter of housekeeping, I'm going to close this ticket. Please send re-open it if you are interested in pursuing this again.

Warm regards, Anthony