wolfSSL/wolfssl #6553: c# wrapper for WolfCrypt

trampster, 2023-06-28:
### Version

5.6.3

### Description

We need to be able to use WolfCrypt from c#.

We are happy to write our own c# wrapper and have experience doing this. However the current Aes code makes doing this difficult. Let me explain:

WolfCrypt requires you to allocate the Aes struct and pass it to the wc_AesInit and the other AES functions. In c# this means we have to manually define a matching Aes struct in c#. This presents the following problems:
- The WolfCrypt Aes struct layout is heavily dependent on the configuration with many #ifdefs in the struct definition. You have to know which are enabled in order to match it, and this has to be done manually, can't just reference the options.h header.
- The WolfCrypt Aes struct uses inline arrays, which until c# 12 (not yet released) are not supported (you can work around this using a blog struct in place of the array)

However when looking at the API, it doesn't seem that the client actually needs to look inside the struct. This means that WolfCrypt AES could be made c# friendly by doing ether of the following:

- Expose a Aes* ws_CreateAes() method that allocates the Aes struct and returns it, this way the c# won't need to define the struct it can just use a pointer, would also need to create a ws_FreeAes() that would clean it up.
- Expose a int ws_SizeOfAes(), which would return the size of the Aes struct and would allow c# to allocate the correct amount of memory to pass the the Aes methods, again it wouldn't actually need to define the struct in this case.

We have currently worked around this by creating a c wrapper lib for WolfCrypt that does nothing except implement a sizeOfAes() method, and have testing this works for us.

dgarske, 2023-06-28:
Hi @trampster ,

I understand the issue you are facing. I would recommend following the pattern we did for ECC with `wc_ecc_key_new` and `wc_ecc_key_free`. I started a template for wolfCrypt C# wrappers here: https://github.com/wolfSSL/wolfssl/pull/3166

You are welcome to build on that and put up a pull request and we can set you up as a contributor.

Can you tell us more about your project and use case? If you'd like to discuss in a private forum please email support at wolfssl dot com.

Thanks,
David Garske, wolfSSL

trampster, 2023-07-11:
We have completed our c# wrapper, however at this stage we are unlikely to continue persuing wolfcrypt as a solution. As such I'll close this now.

closed 2023-07-11 by trampster