wolfSSL/wolfssl #6555: How to set client ciphersuites TLS_CHACHA20_POLY1305_SHA256?

yangyangliuli, 2023-06-29:
### Version

WolfSSL commercial v5.5.3

### Description

I want to use TLS_CHACHA20_POLY1305_SHA25 to connect to DTLS1.3, but fail. Error occurs as following：I have defined macro
HAVE_CHACHA HAVE_POLY1305 in user_setting.h, is there others macro need to define? Thank you in advance.
wolfssl/src/tls13.c,2635: wolfSSL Entering DecryptTls13

wolfssl/src/tls13.c,2563: MAC did not match

wolfssl/src/internal.c,19377: Decrypt failed

wolfssl/src/internal.c,18870: DTLS: Ignoring failed decryption

julek-wolfssl, 2023-07-04:
Hi @yangyangliuli,

I am able to perform a DTLS 1.3 connection using the following commands:

```
./examples/server/server -v 4 -u -l TLS_CHACHA20_POLY1305_SHA256
./examples/client/client -v 4 -u -l TLS_CHACHA20_POLY1305_SHA256
```

with wolfSSL 5.5.3 compiled using:

```
./configure --enable-dtls --enable-dtls13
```

Could you share more about how you are building wolfSSL? Can you also share steps how you reproduce the issue?

I also checked the connection using WireShark:

[image]

Juliusz

yangyangliuli, 2023-07-05:
Hi Juliusz,
I tried to build connection between wolfssl server and client example using TLS_CHACHA20_POLY1305_SHA256, it is no problem. 
Because wolfssl as our project sub lib, so we only used user_setting.h and wolfssl.mk to build lib. So I don't know what's the difference between wolfssl client example. Our project and wolfssl client example both connect the same wolfssl server example. But our project can't parse DecryptTls13 correctly.
user_setting.h wolfssl.mk, our project log, server qcap log as attachment.

[chacha.zip](https://github.com/wolfSSL/wolfssl/files/11955530/chacha.zip)

yangyangliuli, 2023-07-05:
Our project use tls1.3 to connect wolfssl server example successfully, So I guess may be some error in DTLS1.3, or dtls1.3 may be need to another define?

yangyangliuli, 2023-07-05:
using udp.port == 8036 to filter wireshark log.

yangyangliuli, 2023-07-06:
Our project doesn't have the following extension
[image]
How can I open this?

julek-wolfssl, 2023-07-07:
@yangyangliuli can you explain how do you setup your testing? Are you using wolfSSL as both the client and the server? From your logs, it appears that the wolfSSL client can not decrypt the encrypted handshake packets of the server.

yangyangliuli, 2023-07-11:
Hi，
No，I am not using wolfssl example client to build the connection. I am using wolfssl of our project to build connection with wolfssl server example. But our project wolfssl's version is v5.5.3, wolfssl server example's version is 5.6.0.
Yes, from the log to see, our project wolfssl can't decrypt the package from the server, and occurs "Mac did not match". Do you know what can affect Mac mismatch？
Brs,
Yang

yangyangliuli, 2023-07-11:
Hi，
There are two projects(A and B) using same lib wolfssl v5.5.3. The configuration of A and B wolfssl is exactly the same. A can work well, the CPU of A project uses the little endian store. B can't wok well, CPU of Project B uses the big endian store.I don't know what can go wrong when parsing.

yangyangliuli, 2023-07-11:
And B with build TLS1.3 connection successfully. Only DTLS1.3 fail.

julek-wolfssl, 2023-07-11:
Am I understanding correctly that one of the peers (project B) runs on a big endian platform?

yangyangliuli, 2023-07-12:
Yes, but project B can build a TLS1.3 with TLS_CHACHA20_POLY1305_SHA256 connection, only DTLS1.3 fail. So it might not be a big endian problem.

julek-wolfssl, 2023-07-12:
That is very important information. I suspect that it can have an impact as DTLS 1.3 builds on top of TLS 1.3 and there is some separate logic to handle the unreliability of the transport medium. I will attempt to recreate your scenario with a big endian peer. Thank you for you report.

yangyangliuli, 2023-07-20:
If you need my help, or have any progress, please contact me in time. Looking forward to your progress.

yangyangliuli, 2023-07-25:
Hi,

I have updated three files according to your commit, 
src/dtls13. c,
wolfssl/wolfcrypt/misc. h
wolfcrypt/src/misc. c

I can't build pass if it's the same as your changes, because 5.5.3 version doesn't have WC_MISC_STATIC, it can build pass after chaganed to WC_STATIC.
But DTLS1.3 still can't connect to server successfully using TLS_CHACHA20_POLY1305_SHA256, error is same as before.
Patch and my project debug log as attachment.

julek-wolfssl, 2023-07-26:
@yangyangliuli Please make sure to apply this fix to both sides of the connection. If this still doesn't work, please place a `printf` statement inside `Dtls13GetRnMask` to show the value of counter (like `printf("counter: %u", counter)`) on both sides of the connection and check if they print the same values.

julek-wolfssl, 2023-07-26:
Reopening while @yangyangliuli still has issues running on a BE system.

yangyangliuli, 2023-07-28:
Hi，
Well done. You are amazing.
The problem has been resolved, I didn't change server code last time. Now it works fine after modifying server code. This patch will be added to the latest wolfssl, right? Many thanks for your support!

julek-wolfssl, 2023-07-28:
Thank you for confirming the fix works for you. https://github.com/wolfSSL/wolfssl/pull/6641 has been merged and will be present in the next release. Thank you again for the report!

closed 2023-07-28 by julek-wolfssl