wolfSSL/wolfssl #6568: How to use ECH for both server/client

joeftiger, 2023-07-02:
### Version

5.6.3 (#3b3c175af0e993ffaae251871421e206cc41963f)

### Description

I am interested in getting to run Encrypted Client Hello in a simple echo-server constellation.
While I know of [your blog post](https://www.wolfssl.com/encrypted-client-hello-ech-now-supported-wolfssl/) and I went through the given example, it is only valid for some preconfigured server of Cloudflare supporting ECH.
There is no documentation on how to use it on the wolfSSL web page, so my question is the following:

How do you set up ECH for a simple TLS server and client?

I am developing a Remote Attestation extension in TLS for my master's thesis and ECH would be used to hide the attestation data / request from the client.

jpbland1, 2023-07-04:
Hey Joe,

When I implemented ECH I made a test for the server side, I can't link to it because github says the file is too big but its in `tests/api.c` and the test is called `test_wolfSSL_Tls13_ECH`. It's a total pain looking at that huge file so I'm going to make a standalone example but it's at least something in the meantime.

Best Wishes,
John Bland

jpbland1, 2023-07-07:
Hey Joe,

My ech examples just got merged, you can get them from https://github.com/wolfSSL/wolfssl-examples in the tls folder they are `server-ech-local.c` and `client-ech-local.c`. You need to start the server first and it will print out the base64 encoded ech-configs. then just pass that as the first command line argument to the client and it will use them to connect with ech.

Best Wishes,
John Bland

closed 2023-07-07 by jpbland1

joeftiger, 2023-07-08:
Thanks a lot!