wolfSSL/wolfssl #6608: [Bug]:  aes_gcm_asm.asm doesn't support Windows x86-64 calling convention

ilka1999, 2023-07-13:
### Contact Details

ilka1999@mail.ru

### Version

5.6.3

### Description

[according](https://learn.microsoft.com/en-us/cpp/build/x64-software-conventions?view=msvc-170#register-volatility-and-preservation) to MS convention, registers XMM6-15 are nonvolatile. 

see my [previous](https://github.com/wolfSSL/wolfssl/issues/5190) report for details please

ilka1999, 2023-07-13:
for example, clang can use XMM6-15 registers for initialization constants

SparkiDev, 2023-07-14:
Hi @ilka1999,

I see the issue and have prepared a fix for this: https://github.com/wolfSSL/wolfssl/pull/6617

Please try out the PR code and let me know if this works for you.

Thanks,
Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

SparkiDev, 2023-07-19:
Hi @ilka1999

It has been some time since I posted the PR.
I was wondering if you had had a chance to try it out?

Sean

ilka1999, 2023-07-19:
Hi Sean

I'm a native code expert, not a wolfSSL user.
My friend who can test wolfSSL is offline right now (on vacation) and should be available in a week.
I'll ask him.

Ilias

SparkiDev, 2023-07-19:
Thanks @ilka1999.

Sean

SparkiDev, 2023-07-19:
Hi @ilka1999

I've updated the pull request based on the fixes required from your observations.
Note that the functions use stack for temporary storage.

Thanks,
Sean

closed 2023-07-31 by JacobBarthelmeh

SparkiDev, 2023-07-31:
Hi @ilka1999

The PR that fixes the issues you raised has been committed.
Let us know if there are any more problems like this!

Thanks,
Sean

PR #6617 (2023-07-31): AES-GCM x86_64 MSVC ASM: XMM6-15 are non-volatile
# Description

Put XMM6-15, when used, on the stack at start of function and restore at end of function.

Fixes #6608

# Testing

Standard

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation