wolfSSL/wolfssl #6631: [Bug]: some structures and functions that are compatible with openssl are not defined

shove70, 2023-07-18:
### Version

master,  commit: 67d35ea48

### Description

```c
#include <openssl/ssl.h>

int main()
{
    SSL_CTX* context = SSL_CTX_new(TLS_client_method());
}
```

compile:
```
gcc -I/usr/local/include/wolfssl -L/usr/local/lib -lwolfssl -o test001 test001.c
```

output:
```
test001.c:5:5: error: use of undeclared identifier 'SSL_CTX'
    SSL_CTX* context = SSL_CTX_new(TLS_client_method());
    ^
test001.c:5:14: error: use of undeclared identifier 'context'
    SSL_CTX* context = SSL_CTX_new(TLS_client_method());
             ^
test001.c:5:24: error: call to undeclared function 'SSL_CTX_new'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    SSL_CTX* context = SSL_CTX_new(TLS_client_method());
                       ^
test001.c:5:36: error: call to undeclared function 'TLS_client_method'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    SSL_CTX* context = SSL_CTX_new(TLS_client_method());
                                   ^
test001.c:5:36: note: did you mean 'wolfTLS_client_method'?
/usr/local/include/wolfssl/ssl.h:841:33: note: 'wolfTLS_client_method' declared here
    WOLFSSL_API WOLFSSL_METHOD *wolfTLS_client_method(void);
```

build wolfssl:
```
./configure --enable-opensslextra
```

gcc version: 
```
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: x86_64-apple-darwin22.5.0

and

gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-4)
```

shove70, 2023-07-18:
this OK:

```c
int main()
{
    WOLFSSL_CTX* context = wolfSSL_CTX_new(wolfTLS_client_method());
}
```

dgarske, 2023-07-18:
Hi @shove70 ,

Please make sure:

1) Include `#include <wolfssl/options.h>` before `#include <openssl/ssl.h>`. The options.h is generated when wolfssl ./configure is run and has all the build options / defines used at build time.

2) Add `/usr/local/include/wolfssl/wolfssl`, so the wolfssl openssl compatibility headers are used. https://github.com/wolfSSL/wolfssl/tree/master/wolfssl/openssl

Our library uses headers and macros like `#define SSL_CTX_new(method)             wolfSSL_CTX_new((WOLFSSL_METHOD*)(method))` to setup the compatibility. This also requires the build options like `OPENSSL_EXTRA` be defined to expose this.

Thanks,
David Garske, wolfSSL

closed 2023-07-19 by shove70