wolfSSL/wolfssl #6678: [Bug]: wolfSSL_connect returns bad error code

4e4o, 2023-08-07:
### Version

master

### Description

Maybe process goes faster if i open issue.
Exactly the same issue as https://github.com/wolfSSL/wolfssl/issues/5325 but with wolfSSL_connect_TLSv13.
Pull request: https://github.com/wolfSSL/wolfssl/pull/6677

### Reproduction steps

Reproduced with this https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-nonblocking.c example.

kareem-wolfssl, 2023-08-07:
Hi @4e4o ,

Thanks for the report, and for your PR.  We do require a signed contributor agreement to accept third-party contributions, and we don't file agreements for minor changes.
Please let me know if you have any further contributions planned.  If you do, we can file a contributor agreement.  Otherwise, we will treat your PR as a bug report.

Thanks,
Kareem

JacobBarthelmeh, 2023-08-09:
Closing in favor of PR6692

closed 2023-12-05 by dgarske