wolfSSL/wolfssl #6684: [Bug]: wolfSSL_CTX_set1_groups declared twice in headers, one time without #ifdef

laurenzfg, 2023-08-08:
### Contact Details

[myfirstname].grote@rwth-aachen.de

### Version

5.6.3

### Description

Target + Build Environment: 64bit Linux

./configure --enable-nullcipher --enable-psk --enable-opensslcoexist --enable-debug --enable-tls13 --enable-ecccustcurves  --enable-brainpool --enable-curve25519 --enable-ed25519 --enable-curve448 --enable-ed448 CFLAGS="-DWOLFSSL_STATIC_RSA -DWOLFSSL_STATIC_DH -DKEEP_PEER_CERT -DHAVE_SECRET_CALLBACK -DHAVE_SUPPORTED_CURVES"

**My generated `ssl.h` looks like this:** https://gist.github.com/laurenzfg/1dc1c7a5fded206fe67b56798f0b9598

Hey,

thanks for your fast support on issue #6680. I have found a similar issue with regards to the function `wolfSSL_CTX_set1_groups_list`. It's sibling `wolfSSL_CTX_set1_groups` is correctly hidden behind a `OPENSSL_EXTRA` guard. `wolfSSL_CTX_set1_groups_list` is not. But in `ssl.c`, `wolfSSL_CTX_set1_groups_list` is only defined if `OPENSSL_EXTRA` is defined.
This way, I can compile but not link against the function `wolfSSL_CTX_set1_groups_list` in my environment.
I believe this inconsistency is a bug.

kareem-wolfssl, 2023-08-09:
Hi @laurenzfg ,

Please confirm whether this PR fixes the issue for you: https://github.com/wolfSSL/wolfssl/pull/6688

Thanks,
Kareem

laurenzfg, 2023-08-09:
Hi @kareem-wolfssl 

thx for preparing a fix. This seems to indeed clarify the inconsistency and makes my build fail during the compile stage as expected.

It would be wonderful if one could set the supported suites using the curve name strings without OPENSSL_EXTRA, though. As far is I understand Wolf's API, I have to use the name enum values if OPENSSL_EXTRA is not enabled.

PR #6688 (2023-08-10): Update set1_groups/_list prototype conditions to match the functions' conditions.  Fixes #6684.
# Description

Fixes #6684

# Testing

Built + unit tests

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
commit 26fcdbf3 (2023-08-10): Merge pull request #6688 from kareem-wolfssl/gh6684

Update set1_groups/_list prototype conditions to match the functions' conditions.  Fixes #6684.
commit 9f577dc4 (2023-08-10): Update set1_groups/_list prototype conditions to match the functions' conditions.  Fixes #6684.
closed 2023-08-10 by JacobBarthelmeh