wolfSSL/wolfssl #6689: DECLARE_CURVE_SPECS(ECC_CURVE_FIELD_COUNT) makes a NULL spec_ints array

dientc, 2023-08-09:
### Version

5.6.3

### Description

This issue happens while we try to verify a ECC signature and `wc_ecc_verify_hash_ex `function is invoked. Inside this function, `DECLARE_CURVE_SPECS(ECC_CURVE_FIELD_COUNT)` is used to make a spec_ints array.

However, in debugging we've found that marco `DECLARE_CURVE_SPECS `does not work well, `spec_ints ` is `NULL`. See  attached image. This leads to further crashing.

After having some investigation, we see if we replace `ECC_CURVE_FIELD_COUNT `by a hardcode (i.e. 5 ) or using a define, then it works well. 

I suspect if passing an `enum ` value as a parameter to a macro can cause this problem. I tried some experiment to test a same scenario in other source with my compiler, it still works.

C Compiler: Tasking TriCore

[image]

kareem-wolfssl, 2023-08-10:
Hi @dientc ,

I've created a test PR here which converts this enum into a series of defines, please try it and confirm if it works for you: https://github.com/wolfSSL/wolfssl/pull/6694/files
However, as you've said, I don't believe this variable being an enum is the core issue here, rather I think you're running into an issue with the DECLARE_CURVE_SPECS macro.  We do have multiple definitions of this macro, so please share your build settings (./configure line or user_settings.h) so I can confirm which you are using.

dientc, 2023-08-10:
Hi @kareem-wolfssl ,

1. I confirm your PR working well to me. As I said, if we use defines instead of `enum`, it can work in my case.
2. Using a trick to add `#error` to print log while compiling, I am quite sure that I am using correct `DECLARE_CURVE_SPECS` macro in many. See below:
 
[image]

I attach my user_settings.h. As  you can see there, `WOLFSSL_SP_MATH_ALL` is not defined. However, that define is set in `settings.h Line: 2063` as default.

[user_settings.h.log](https://github.com/wolfSSL/wolfssl/files/12311925/user_settings.h.log)

Thank you.

PR #6694 (2023-08-11): Change enum ecc_curve_load_mask into defines
# Description

Fixes #6689

# Testing

Built in tests