wolfSSL/wolfssl #6691: PKCS11 store/retrieve GP data

embetrix, 2023-08-09:
### Version

v5.6.3

### Description

we think about using Wolfssl as PKCS11 backend, I'm wondering if there is support to store/retrieve General Purpose data (Not Crypto Key material) withing Wolfssl PKCS11 Interface ?

Normally it should be supported by the PCKS11 Standard : 

```
$echo "some secret data!" > data.bin

$ export PKCS11_MODULE_PATH="/usr/lib/softhsm/libsofthsm2.so"

$ pkcs11-tool --module $PKCS11_MODULE_PATH --login --write-object data.bin --type data --id 6987 --label 'data-6987'
Using slot 0 with a present token (0x45e68566)
Logging in to "SoftToken".
Please enter User PIN: 
Created Data Object:
Data object 2
  label:          'data-6987'
  application:    ''
  app_id:         <empty>
  flags:           modifiable

$ pkcs11-tool --module $PKCS11_MODULE_PATH --login --read-object --type data --label 'data-6987'
Using slot 0 with a present token (0x45e68566)
Logging in to "SoftToken".
Please enter User PIN: 
some secret data!
```

SparkiDev, 2023-08-09:
Hi @embetrix,

We do not have this feature at this time.
If it is something important to you please let us know and we can discuss how we can have this feature added.

Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

SparkiDev, 2023-09-04:
Hi @embetrix,

Is there any update for this Issue?

Thanks,
Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

SparkiDev, 2023-10-02:
Hi @embetrix.

Please raise another issue if you have further questions.

Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

closed 2023-10-02 by SparkiDev