wolfSSL/wolfssl #6728: [Bug]: Possible buffer overrun in wc_PKCS7_DecodeAuthEnvelopedData

markhermeling, 2023-08-24:
### Contact Details

mark@hermeling.org

### Version

master

### Description

There is a possible buffer overrun in wc_PKCS7_DecodeAuthEnvelopedData, discovered by CodeSonar during by static analysis. It is a very deep bug, but the path seems plausible. 

SARIF file of the warning path is attached, you can open this in VS Code using the SARIF plug-in and it will explain the path.

The root-cause is that it is possible for 

int GetLength_ex(const byte* input, word32* inOutIdx, int* len, word32 maxIdx, int check)

to return 0 for length, while not returning error.

This function is called on line 12055 in pkcs7.c. This 0 valued is then used in the call to wc_PKCS7_AddDataToStream, which can lead to a malloc of 0 size in GrowStream, which creates a buffer of size 0, which is then used in a XMEMCOPY on line 221.

It is quite a long and complex path, but it seems feasible. 
[2260110.sarif.zip](https://github.com/wolfSSL/wolfssl/files/12432616/2260110.sarif.zip)

Not sure if it makes more sense for GetLength to return error if length is 0, or if there should be an check for length 0 before the malloc in GrowStream.

### Reproduction steps

./configure --enable-all

anhu, 2023-08-24:
Hi @markhermeling , 

Thank you for this report.  My name is Anthony Hu and I will be looking into this. It would be great if you could open this as a ticket in our ZenDesk support address: support@wolfssl.com. 

Warm regards, Anthony

markhermeling, 2023-08-25:
I ran the analysis on https://github.com/markhermeling/wolfssl/tree/feature/merge-master-2023-08-17, which is a merge from wolfssl master

anhu, 2023-08-25:
Closing in favour of continue the conversation on Zendesk.

closed 2023-08-25 by anhu