wolfSSL/wolfssl #6758: [Bug]: Certificate chain is 0 when using non-blocking OCSP and wolfSSL_connect is called more than once.

Dor-kpf, 2023-09-07:
### Version

5.5.4

### Description

### Target
Linux x86

### Configuration
compiled WolfSSL with the WOLFSSL_NONBLOCK_OCSP flag.

### Version
5.5.4

### Bug description
Hello!
We compile WolfSSL with `WOLFSSL_NONBLOCK_OCSP` flag to achieve connection with non-blocking OCSP.
The problem is that when WolfSSL calls our verify callback (set by `wolfSSL_set_verify` function) we have the certificate chain with count equal to 0.
We see this problem when we are calling `wolfSSL_connect()` the second time. We saw that the call hierarchy is like this:
```
wolfSSL_connect()
    ...
    DoCertificate()
        ssl->session->chain->count = 0;
        ProcessPeerCerts()
            switch(ssl->options.asyncState)
                case TLS_ASYNC_BEGIN
                    AddSessionCertToChain(&ssl->session->chain, input + args->idx, certSz)
                        ssl->session->chain->count++;
                …
                case TLS_ASYNC_VERIFY
                    CheckCertOCSP_ex()
                        ocspCallback()
                case TLS_ASYNC_FINALIZE
                    DoVerifyCallback()
                        verifyCallback()
               …
```

As we understand, each time we call `wolfSSL_connect()`, `DoCertificate()` is called which resets the chain count to 0. Then `ProcessPeerCerts()` is called.
Inside `ProcessPeerCerts()` we have a state machine. At first, the state is `TLS_ASYNC_BEGIN` and we increment the count size (**this is the only place that the chain count is incremented**).
Once we reach the `TLS_ASYNC_VERIFY` state, WolfSSL calls our OCSP callback (set by `wolfSSL_SetOCSP_Cb` function). Our OCSP callback may not complete the first time, so we return `WOLFSSL_CBIO_ERR_WANT_READ`.
Then, `ProcessPeerCerts()` does not proceed to the `TLS_ASYNC_FINALIZE` state.
The next time we call `wolfSSL_connect()`, `DoCertificate()` is called, **which resets the chain count to 0, again**. `ProcessPeerCerts()` is then called again and this time the `ssl->options.asyncState` is preserved from the last call to `wolfSSL_connect()` (which is `TLS_ASYNC_VERIFY`).
WolfSSL calls our OCSP callback again. This time our OCSP callback is finished successfully.
This transition the state to `TLS_ASYNC_FINALIZE` which calls our verifyCallback (set by `wolfSSL_set_verify` function).
Here we already see that the chain count is set to 0, even though the certificate chain is containing a certificate (calling `wolfSSL_get_chain_count` returns 0 inside our verifyCallback function).
We found a workaround for this:
```
    int count = 0;
    for(; count < 9; count ++) // 9 == MAX_CHAIN_DEPTH
    {
        if(wolfSSL_get_chain_length(chain, count) == 0)
        {
            break;
        }
    }
```
We would like to know if our assumptions about this problem are correct.
And how do you suggest dealing with this.
Thank you!

closed 2023-09-13 by jpbland1

jpbland1, 2023-09-13:
Hey @Dor-kpf,

Thanks for bringing this to our attention, I've merged the fix for the chain count reset on async retry. Let me know if you have have any issues or discover another bug.

Best Wishes,
John Bland

Dor-kpf, 2023-09-13:
Thanks for the quick handling of the bug!
In which WolfSSL version this bug fix will be available?

jpbland1, 2023-09-13:
Hey @Dor-kpf

These changes have been merged into the current master branch and will be included in the next release. If you are using a commercial version of wolfSSL and can't pull the latest changes please contact your sales rep or if you purchased support use zendesk, we can provide a patch for these changes.

Best Wishes,
John Bland