wolfSSL/wolfssl #6760: [Bug]: Memory leak in ssl->rng

razvancrainea, 2023-09-08:
### Contact Details

razvan@opensips.org

### Version

5.4.0

### Description

We are compiling the library static, as it follows:
```
 ./configure \
	--enable-all --disable-fpecc -disable-aligndata --disable-examples \
	--enable-writedup --enable-tlsv10 --disable-shared --enable-static \
	--disable-jni --disable-crl-monitor \
	C_EXTRA_FLAGS="-fPIC" \
	CFLAGS="-DWOLFSSL_STATIC_RSA -DWOLFSSL_DEBUG_MEMORY"
```
We are using our own memory allocators, which are keeping track of the memory usage, and we have identified a memory leak of the `ssl->rng` memory allocated in [InitSSL](https://github.com/wolfSSL/wolfssl/blob/v5.4.0-stable/src/internal.c#L6723)  function, along with an equal number of chunks allocated from [_InitRng](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/random.c#L1633). The number of fragments/chunks allocated in these two places are 100x more than the number of actual `WOLFSSL ssl` contexts, and they are never going down - this confirms 100% this is a leak.

We have tried to troubleshoot this ourselves, and observed that the release of the `ssl->rng` structure [depends](https://github.com/wolfSSL/wolfssl/blob/v5.4.0-stable/src/internal.c#L7574) on the `ssl->options.weOwnRng` field, which is properly set to `1` when the memory is [allocated](https://github.com/wolfSSL/wolfssl/blob/v5.4.0-stable/src/internal.c#L6729). However, when the memory is released, its value is `0` - this is acutally the bug; We have checked the code and the weOwnRng is only reset in [one](https://github.com/wolfSSL/wolfssl/blob/v5.4.0-stable/src/internal.c#L7574) place, when `ssl->rng` is also set to `NULL`. However, according to our logs, when `FreeSSL` and `SSL_ResourceFree` functions are called, our `ssl->options.weOwnRng` is set `0`, but `ssl->rng` is present as well (not `NULL`). We're also not using `SINGLE_THREADED`, thus we have our own rng (which we've seen allocated in our memory logs).

Bottom line, we do not understand who overwrites the `weOwnRng` field without resetting or freeing the `ssl->rng` structure. I was suspecting this as a compiler issue (related to bitfield usage of `weOwnRng`), but I am not sure how to prove (and hopefully fix) it.

In the meantime, we have changed [this](https://github.com/wolfSSL/wolfssl/blob/v5.4.0-stable/src/internal.c#L7210) code as it follows:
```
-    if (ssl->options.weOwnRng) {
+    if (ssl->options.weOwnRng || (ssl->rng
+#ifdef SINGLE_THREADED
+        && ssl->rng != ssl->ctx->rng
+#endif
+       )) {
```
This fixes our leak, but I don't think is the correct approach, as I still don't understand how `weOwnRng` gets changed. Any help would be really appreciated.

### Reproduction steps

 ```
 ./configure \
	--enable-all --disable-fpecc -disable-aligndata --disable-examples \
	--enable-writedup --enable-tlsv10 --disable-shared --enable-static \
	--disable-jni --disable-crl-monitor \
	C_EXTRA_FLAGS="-fPIC" \
	CFLAGS="-DWOLFSSL_STATIC_RSA -DWOLFSSL_DEBUG_MEMORY"
```

Run a few hundreds connections. Run a memory dump and check the logs - they will show a huge number of `ssl->rng` and `ssl->rng->drbg` chunks unalocated. Check the logs below. Note that not all of the fragments get leaked, only some of them.

### Relevant log output

```shell
BAD:
  169585| 2023-09-07 15:33:30.452|       src/internal.c   6763: ctx=0x7f3240d335c8 ssl=0x7f3243b974e0 rng=0x7f3243bcf408 weOwnRng=1 v=TLSv1: allocated
  169588| 2023-09-07 15:33:33.536|       src/internal.c   7776: ctx=0x7f3240d335c8 ssl=0x7f3243b974e0 rng=0x7f3243bcf408 weOwnRng=0 v=TLSv1: releasing
  169588| 2023-09-07 15:33:33.536|       src/internal.c   7233: ctx=0x7f3240d335c8 ssl=0x7f3243b974e0 rng=0x7f3243bcf408 weOwnRng=0 v=TLSv1: freeing forcefully

GOOD:
  169578| 2023-09-07 15:03:48.641|       src/internal.c   6763: ctx=0x7f3240d335c8 ssl=0x7f32432334c0 rng=0x7f32473e8380 own=1 v=TLSv1: allocated                                                  
  169588| 2023-09-07 15:04:12.422|       src/internal.c   7776: ctx=0x7f3240d335c8 ssl=0x7f32432334c0 rng=0x7f32473e8380 own=1 v=TLSv1: releasing                                                  
  169588| 2023-09-07 15:04:12.422|       src/internal.c   7235: ctx=0x7f3240d335c8 ssl=0x7f32432334c0 rng=0x7f32473e8380 own=1 v=TLSv1: freeing OK
```

embhorn, 2023-09-08:
HI @razvancrainea,

It looks like the issue you are referring to was fixed here:
https://github.com/wolfSSL/wolfssl/pull/5468

Could you please update to the latest release and test?

Thanks,
@embhorn - wolfSSL Support

razvancrainea, 2023-09-08:
@embhorn thank you for your response! I tested against v5.6.3-stable too, and I got the same behavior/leak. I see that this commit got into 5.5.0, so I am pretty sure we used it. Also, we are not using `wolfSSL_clear` either, as I understand the initial bug was related to.

embhorn, 2023-09-08:
I see that you are using the write duplicate feature (`--enable-writedup`). Are you freeing both SSL objects after the connection terminates? Here is an example:
https://github.com/wolfSSL/wolfssl-examples/blob/eb4a33690f899f89aea498cc330468735014a2e5/tls/client-tls-writedup.c#L219-L222

razvancrainea, 2023-09-12:
@embhorn yes, I am sure we are releasing both SSL objects - we are actually not leaking any `WOLFSSL` object, just the `rng` and its associated `rng->drbg` fields. For example, this is one of our analysis, based on our memory manager (notice we compiled with `-DWOLFSSL_DEBUG_MEMORY`, so we can have wolfssl func and line):

```
38945 x [func _InitRng, line 843]
38943 x [func InitSSL, line 6723]
3855 x [func TLSX_New, line 1285]
1645 x [func TLSX_SupportedCurve_New, line 3764]
956 x [func wolfSSL_write_dup, line 811]
956 x [func wolfSSL_new, line 658]
956 x [func wolfSSL_NewSession, line 19555]
956 x [func TLSX_UseSecureRenegotiation, line 5038]
956 x [func SetKeys, line 2380]
956 x [func SetKeys, line 2369]
956 x [func InitSSL, line 6691]
956 x [func InitSSL, line 6678]
956 x [func InitSSL, line 6665]
956 x [func InitSSL, line 6653]
956 x [func DupSSL, line 736]
329 x [func TLSX_SNI_New, line 1820]
329 x [func TLSX_SNI_New, line 1804]
329 x [func TLSX_PointFormat_New, line 3782]
8 x [func AllocDer, line 20087]
6 x [func ParseCert, line 18618]
4 x [func ParseCert, line 18633]
4 x [func MakeSigner, line 19846]
2 x [func wolfSSLv23_client_method_ex, line 11866]
2 x [func wolfSSL_CertManagerNew_ex, line 4143]
2 x [func wolfSSL_CTX_new_ex, line 505]
2 x [func wolfSSL_CTX_new_ex, line 487]
2 x [func InitSSL_Ctx, line 2287]
2 x [func InitSSL_Ctx, line 2279]
2 x [func AddTrustedPeer, line 4984]
2 x [func AddTrustedPeer, line 4959]
```

We have 956 ongoing TLS connection which match precisely the `wolfSSL_new`, `wolfSSL_write_dup`, `InitSSL` and `DupSSL` fragments. Its just the two fields that go out of the charts: InitSSL, line 6723 and _InitRng, line 843, with 38943 fragments. And again, ignoring the `weOwnRng` field and releasing the `rng` structure if it exists (`!= NULL`) fixes the leak.

To me it looks like somebody/something overwrites that field, but I really don't understand how.

embhorn, 2023-09-12:
Hi @razvancrainea 

I reviewed this with the team. There is not an obvious place where the RNG is being allocated and not freed.

I'd like to get a reproducer for this. If I added a loop to the previously shared example, do you think that will reproduce this issue?

Thanks,

razvancrainea, 2023-09-13:
Hi, @embhorn !

That is correct, I did a bit of review myself and there is no obvious place where RNG would not be released. But I think that is just a consequence of the bug itself, that is, the fact that the `ssl->options.weOwnRng` field somehow gets set to `0` by some sort of invalid operation without the `ssl->rng` field being freed. I did add logs everywhere where the `weOwnRng` field is being set (there are only 3 spots), and they do not hit - that makes me think the reset of the field is not explicit, but rather comes from a different bug/invalid memory access. My ideas are:
1. compiler error - as `weOwnRng` is a bit in a bitfield, perhaps some sort of operation is not properly generated in assembly/code, and the field gets overwritten along with a different other field - I am going to try to compile with a different compiler (i.e. clang) to rule this out.
2. compilation error - due to some bugs in the compilation (i.e. mishandled flags), the WOLFSSL options structure might not be aligned with the library's structure, therefore the `weOwnRng` bitfield might get overwritten when the lib is trying to set a different other bitfield.
3. programming error - a place where a memset or something similar would write more fields than expected - I don't have enough expertise to evaluate where this might happen
 
I don't think it will replicate, as I was not able to replicate it myself. It only reproduces in production, where we're checking the memory usage in 2h windows, and we see the above data. Do note that not all connections/WolfSSL structures leak in this manner, just some of them.
And we were not able to correlate the number that leaks with anything (I tried to group based on TLS version, connection patterns, etc. - nothing matches up).
I know this is a weird one, and I don't know how to get a grab on it. But we are willing to provide any information you need, and even add debugging patches, if that might lead to reveal the actual problem.

Thanks for helping us :)

embhorn, 2023-09-13:
Your idea for #2 is plausible. If the configuration include used to build the library is different from the header used with the application, then some of the fields would be skewed in the object. The application must include the config header before any other wolfSSL headers.
```
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
```

Could you set a memory watch breakpoint on the address of the `weOwnRng` field to see when it changes?

razvancrainea, 2023-09-13:
I can confirm the headers are included in the correct order.
Regarding setting a memory watch, I did thought about this as well, but not sure if possible. As I said, this does not reproduce in a testing environment, only on live production traffic, so setting a watch breakpoint for ~900 concurrent connections, probably volatile as well, will probably not be plausible.
But thinking about this, do you think running under valgrind might reveal some interesting behavior? That we might be able to do.

embhorn, 2023-09-14:
Yes, certainly valgrind could help point to source of the issue. Also `fsanitize=leak` could be useful.

embhorn, 2023-09-14:
@razvancrainea - Is this open source? Could you point me to the code that sets up the SSL object?

If you'd rather keep it private, please send a message to support@wolfssl.com

razvancrainea, 2023-09-15:
@embhorn the bug is in [OpenSIPS](https://github.com/OpenSIPS/opensips) which is open source, so I can definitely point you the code :). We statically linking the [WolfSSL static library](https://github.com/OpenSIPS/opensips/blob/master/modules/tls_wolfssl/Makefile#L26) into our [tls_wolfssl](https://github.com/OpenSIPS/opensips/tree/master/modules/tls_wolfssl) module.

First thing that happens is the initialization of the connection, where we create the [`read_ssl`](https://github.com/OpenSIPS/opensips/blob/master/modules/tls_wolfssl/wolfssl_conn_ops.c#L197). Then we duplicate this object for write in multiple spots, depending on the operation:
-  when connecting ([asynchronously](https://github.com/OpenSIPS/opensips/blob/master/modules/tls_wolfssl/wolfssl_conn_ops.c#L386) or [synchronously](https://github.com/OpenSIPS/opensips/blob/master/modules/tls_wolfssl/wolfssl_conn_ops.c#L668)) to a remote TLS endpoint
- when [accepting](https://github.com/OpenSIPS/opensips/blob/master/modules/tls_wolfssl/wolfssl_conn_ops.c#L560) a new TLS connection

Finally, we release the two objects [here](https://github.com/OpenSIPS/opensips/blob/master/modules/tls_wolfssl/wolfssl_conn_ops.c#L298), when the associated TLS/TCP connection is released.

I can agree the code is not that easy to follow, especially due to the modularity of OpenSIPS, but please bear with me during this process. If you have any other questions, please let me know how I can assist.

dgarske, 2023-09-15:
@embhorn , A couple things. The CTX_new method is wolfSSLv23_method , so they could be using TLS v1.3. Also this is using the "either side" logic. Both of these are likely not tested with write_dup. Eric, can you try our write-dup with TLS v1.3 and also with the wolfSSLv23_method method? These example are hard coded for TLS v1.2, but I'm curious how well they work with TLS v1.3 and either side method: https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-writedup.c and https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/server-tls-writedup.c with TLS v1.3.
Also with TLS v1.3 resumption the session ticket can arrive after the connect completes, so I am not sure how that might affect the dup feature.

Thanks,
David Garske, wolfSSL

embhorn, 2023-09-15:
I made changes in the client and server writedup examples tro use TLS 1.3:
[tls13-writedup.patch](https://github.com/wolfSSL/wolfssl/files/12623404/tls13-writedup.patch)

I ran them with both valgrind
[valgrind_client.log](https://github.com/wolfSSL/wolfssl/files/12623405/valgrind_client.log)
[valgrind_server.log](https://github.com/wolfSSL/wolfssl/files/12623406/valgrind_server.log)

...and fsanitize=leak
[fsan_client.log](https://github.com/wolfSSL/wolfssl/files/12623416/fsan_client.log)
[fsan_server.log](https://github.com/wolfSSL/wolfssl/files/12623403/fsan_server.log)

razvancrainea, 2023-09-18:
FYI, in production we're currently using only TLS v1.0 - didn't realize this might be a helpful information until now.
Although, during our tests, I was testing TLS v1.3 as well, without no leaks as well, just as @embhorn tests show.

embhorn, 2023-09-18:
Yes, that was useful!
I modified the example to use TLS1.0:
[writedup_tls1_0.patch](https://github.com/wolfSSL/wolfssl/files/12652719/writedup_tls1_0.patch)

This produced fsanitize errors:
```
=================================================================
==66794==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 32 byte(s) in 1 object(s) allocated from:
    #0 0x7f794f3d1302 in __interceptor_malloc ../../../../src/libsanitizer/lsan/lsan_interceptors.cpp:75
    #1 0x7f794ef44628 in wolfSSL_Malloc wolfcrypt/src/memory.c:344
    #2 0x7f794f0074ef in ReinitSSL src/internal.c:6993
    #3 0x7f794f007c29 in InitSSL src/internal.c:7342
    #4 0x7f794f0795e7 in wolfSSL_write_dup src/ssl.c:1632
    #5 0x5620b821e538 in main (/home/eric/repos/wolfssl-examples/tls/client-tls-writedup+0x1538)

Indirect leak of 136 byte(s) in 1 object(s) allocated from:
    #0 0x7f794f3d1302 in __interceptor_malloc ../../../../src/libsanitizer/lsan/lsan_interceptors.cpp:75
    #1 0x7f794ef44628 in wolfSSL_Malloc wolfcrypt/src/memory.c:344
    #2 0x7f794ef068c0 in _InitRng wolfcrypt/src/random.c:1634
    #3 0x7f794ef06bb5 in wc_InitRng_ex wolfcrypt/src/random.c:1756
    #4 0x7f794f00757f in ReinitSSL src/internal.c:7003
    #5 0x7f794f007c29 in InitSSL src/internal.c:7342
    #6 0x7f794f0795e7 in wolfSSL_write_dup src/ssl.c:1632
    #7 0x5620b821e538 in main (/home/eric/repos/wolfssl-examples/tls/client-tls-writedup+0x1538)

SUMMARY: LeakSanitizer: 168 byte(s) leaked in 2 allocation(s).
```
[fsan_client_error.log](https://github.com/wolfSSL/wolfssl/files/12652718/fsan_client_error.log)

I am working on a fix.

embhorn, 2023-09-18:
Hi @razvancrainea 

Please test with the changes I am proposing in #6785

razvancrainea, 2023-09-19:
Sure, let us apply and look into this, thanks!

razvancrainea, 2023-09-19:
Sure, we will keep you updated with the progress. Thanks for looking into this!

razvancrainea, 2023-09-20:
@embhorn I can confirm this fixes our leak, thank you very much for fixing it!

razvancrainea, 2023-10-04:
@embhorn Unfortunately it looks like this patch breaks TLS v1.2 connections - after we applied the patch, everything was working all right, until we needed to connect to some TLS v1.2 endpoints. These endpoints would not establish, being closed just after the handshake.
Digging a bit into this, we started using the latest master branch for testing, and the problem was the same. Then, we ended up reverting the memory leak patch, and the connections were working as expected. So somehow the patch breaks TLS v1.2 connection. Is there anything related to the patch that you can spot on your side?
Let me know if you need further information.

Thank you!

embhorn, 2023-10-04:
Hi @razvancrainea 

That is strange. I can assure you that our CI tests exercised the changes against a TLS1.2 connection. Can you share the log when wolfSSL is configured with `--enable-debug` and the application calls:

```
        #ifdef DEBUG_WOLFSSL
            wolfSSL_Debugging_ON();
        #endif
```

Thanks.

razvancrainea, 2023-10-04:
Sure, attached is the log, hope this helps.
[wolfssl.log](https://github.com/wolfSSL/wolfssl/files/12805641/wolfssl.log)

embhorn, 2023-10-05:
Hi @razvancrainea 

The log only shows non-blocking errors, which should be handled by the application.

With the fix in #6785, your fix in https://github.com/OpenSIPS/opensips/commit/716f1d633d19e08165917423cb141c29c0f2b2df  should not be required.

razvancrainea, 2023-10-05:
@embhorn apologies, that was actually an attempt that was timing out during connect. Here is a new log: 
[wolfssl.log](https://github.com/wolfSSL/wolfssl/files/12819735/wolfssl.log)
I believe the relevant spot is
```
Oct  5 17:42:29 [1581635] WOLFSSL: wolfSSL Leaving BuildMessage, return -173
Oct  5 17:42:29 [1581635] WOLFSSL: wolfSSL error occurred, error = 173 line:22114 file:src/internal.c
Oct  5 17:42:29 [1581635] WOLFSSL: wolfSSL Leaving wolfSSL_write, return -320
```
And the error seems to come from [here](https://github.com/wolfSSL/wolfssl/blob/master/src/internal.c#L21873), where `ssl->rng` is `NULL`.

Regarding the OpenSIPS/opensips@716f1d6, that has been dropped and replaced by OpenSIPS/opensips@2548f47. However, on my test machine, I've reverted all patches and I am running with stock wolfssl master branch, which already contains the leak patch, so I don't have anything extra applied.

Thanks for your help!

embhorn, 2023-10-05:
@razvancrainea 

Thanks for the new log! I was able to reproduce the issue. Working on a fix now.

embhorn, 2023-10-05:
Hi @razvancrainea 

Please check out #6843

razvancrainea, 2023-10-06:
@embhorn I've just applied the new patch on latest master and works as expected. Thank you for the fix!

PR #6785 (2023-09-19): Fix writedup rng leak
# Description

In `DupSSL`, the already allocated `dup->rng` is only freed when `dup->options.weOwnRng` is true. But `dup->options.weOwnRng` is cleared when the options are copied from the original ssl:

```
    /* copy write parts over to dup writer */
    XMEMCPY(&dup->specs,   &ssl->specs,   sizeof(CipherSpecs));
    XMEMCPY(&dup->options, &ssl->options, sizeof(Options));
```

Fixes #6760 

# Testing

## wolfssl

```
./configure --enable-debug --enable-all --disable-fpecc -disable-aligndata --disable-examples --enable-writedup --enable-tlsv10 --disable-jni --disable-cr [... cut]
closed 2023-10-06 by JacobBarthelmeh